espressif/ESP8266_RTOS_SDK #407: Memory Leak on Interrupt Handler

filzek, 2018-12-05:
Hi @wujiangang 

We did face a issue where ISR causing memory leak (Master) if you use a routine that manipulate any kind of volatile variables it cause a memory leakage and it disrupt all the memory until 16KB being free. 

We try to find what kind of operations cause the leak but so far the cause is the ISR. Need to check.

I can provide private proprietary code to exam and find solution in the driver part.

donghengqaz, 2018-12-11:
ISR is always used by hardware driver, could you please tell which driver causes this problem? And could you supply the application for it?

filzek, 2018-12-18:
When manipulate char arrays with malloc instructions inside ISR, we try to find solution but is impossible to fix, didnt find where it creates the memory leak, but it does until 16kb and stop.

donghengqaz, 2018-12-28:
We have fixed some bugs recently, is your problem there ?

closed 2019-01-09 by FayeY