espressif/ESP8266_RTOS_SDK #408: How do i choose the right value for FS1_FLASH_ADDR, FS1_FLASH_SIZE.

nghiaquy1991, 2018-12-07:
## Environment

- Development Kit: ESP8266-12E (4 MB flash)
- IDF version (``git rev-parse --short HEAD`` to get the commit id.): 
    ba08d874f2216a0d4d577ae82f8b9534a0d20a6c
- Development Env:      [Arduino IDE|Make|Eclipse|other] Make
- Operating System:     [Windows|Ubuntu|MacOS] Ubuntu
- Power Supply:         [USB|external 5V|external 3.3V|Battery] USB

I am using spiffs for my application and I need to use OTA firmware, but i donot know How do i choose the right value for FS1_FLASH_ADDR, FS1_FLASH_SIZE.
I read chapper 4. Flash Map (doc: 20a-esp8266-sdk_getting_started_guide_en.pdf) and many docs from espressif, but it is not clear.
Please help me to determine the right value of that define? or give me the links, docs which can help me to have more information. 

Thanks,

MalcolmChen, 2018-12-23:
same issue here

nghiaquy1991, 2018-12-24:
I am still looking the answer.

FayeY, 2018-12-28:
Hi, this feature should be based on the partition, not operate the flash directly. We do have an [OTA example](https://github.com/espressif/ESP8266_RTOS_SDK/tree/master/examples/system) in the SDK, will it help? Thanks.

FayeY, 2019-01-09:
Closing due to lack of response, please feel free to reopen if this is still an issue. Thanks.

nghiaquy1991, 2019-01-10:
Hi FayeY,

On the [OTA example](https://github.com/espressif/ESP8266_RTOS_SDK/blob/master/examples/system/ota/main/ota_example_main.cl)
I did not see FS1_FLASH_ADDR, FS1_FLASH_SIZE define value?
In my board (ESP8266 12F), Flash size is 4MB (512 + 512).
Which value should I choose?

Thanks,