espressif/ESP8266_RTOS_SDK #410: make flash的问题

MinjunHuang, 2018-12-11:
环境：
ubuntu 14.04
python：2.7.14

问题：
在成功编译后，使用make flash，出现以下错误：
Flashing binaries to serial port /dev/ttyUSB0 (app at offset 0x10000)...
esptool.py v2.4.0
Traceback (most recent call last):
  File "/home/xxx/usr/esp/ESP8266_RTOS_SDK/components/esptool_py/esptool/esptool.py", line 3022, in <module>
    _main()
  File "/home/xxx/usr/esp/ESP8266_RTOS_SDK/components/esptool_py/esptool/esptool.py", line 3015, in _main
    main()
  File "/home/xxx/usr/esp/ESP8266_RTOS_SDK/components/esptool_py/esptool/esptool.py", line 2711, in main
    ser_list = sorted(ports.device for ports in list_ports.comports())
  File "/home/xxx/usr/esp/ESP8266_RTOS_SDK/components/esptool_py/esptool/esptool.py", line 2711, in <genexpr>
    ser_list = sorted(ports.device for ports in list_ports.comports())
AttributeError: 'tuple' object has no attribute 'device'
make: *** [flash] 错误 1

donghengqaz, 2018-12-11:
是不是依赖程序没装全啊？ 下面链接有需要安装什么。
https://github.com/espressif/ESP8266_RTOS_SDK/blob/master/docs/en/get-started/linux-setup.rst

guanglun, 2018-12-11:
应该是pyserial没装

MinjunHuang, 2018-12-15:
确实是没有安装pyserial的问题，不知道需要这个包，在错误提示中没有相关提示，建议可以增加一个requirement文件，需要的python包。
问题已经解决，多谢两位的支持，谢谢。

closed 2018-12-15 by MinjunHuang