espressif/ESP8266_RTOS_SDK #411: Error libcirom.a(lib_a-signal.o):(.irom0.literal+0x0): undefined reference to `_getpid_r'

nghiaquy1991, 2018-12-11:
## Environment

- Development Kit: NodeMCU v1.0: 4MB flash
- RTOS_SDK release/v2.x.x: commit f07915833fe22cf86b6c10f17a88500c162af9c4
- Development Env:      Make
- Operating System:     Ubuntu
- Power Supply:         USB

## Problem Description

Dear,
I am developing an application use RTOS_SDK.
But after add some functions, I can not make the app.
I ensure that size of code is smaller than 768KB as required in the doc 2a-es8266-sdk_getting_started_guide_en.pdf.

## Debug Logs
xtensa-lx106-elf-ar: creating .output/eagle/debug/lib/libuser.a
make[1]: Leaving directory `/mnt/Share/IIS/user'
xtensa-lx106-elf-gcc  -L/home/esp8266/Share/ESP8266_RTOS_SDK/lib -Wl,--gc-sections -nostdlib -T/home/esp8266/Share/ESP8266_RTOS_SDK/ld/eagle.app.v6.ld -Wl,--no-check-sections -u call_user_start -Wl,-static -Wl,--start-group -lcirom -lgcc -lhal -lphy -lpp -lnet80211 -lwpa -lcrypto -lmain -lfreertos -llwip -lespconn -lsmartconfig -lairkiss -ldriver user/.output/eagle/debug/lib/libuser.a -Wl,--end-group -o .output/eagle/debug/image/eagle.app.v6.out 
/home/esp8266/Share/ESP8266_RTOS_SDK/lib/libcirom.a(lib_a-signal.o):(.irom0.literal+0x0): undefined reference to `_getpid_r'
/home/esp8266/Share/ESP8266_RTOS_SDK/lib/libcirom.a(lib_a-signal.o):(.irom0.literal+0x4): undefined reference to `_kill_r'
/home/esp8266/Share/ESP8266_RTOS_SDK/lib/libcirom.a(lib_a-signal.o): In function `_raise_r':
/home/wjg/Repo/esp-open-sdk/crosstool-NG/.build/src/newlib-2.0.0/newlib/libc/signal/signal.c:163: undefined reference to `_getpid_r'
/home/wjg/Repo/esp-open-sdk/crosstool-NG/.build/src/newlib-2.0.0/newlib/libc/signal/signal.c:163: undefined reference to `_kill_r'
collect2: error: ld returned 1 exit status
make: *** [.output/eagle/debug/image/eagle.app.v6.out] Error 1

### Steps to repropduce 
I try some solutions but all of them are not work:
1. Change len parameter in ld/eagle.app.v6.ld to 0xC0000
2. Change SPI_SIZE_MAP when make : 4/6 and 7

Does anyone have the same this issue?
How can I fix it?

donghengqaz, 2018-12-11:
It is the reason than the API "_kill_r" is not supplied, you can call some std C library API like "raise", "getpid" and so on.

It is suggested that you may not use these API, and use APIs of FreeRTOS.

nghiaquy1991, 2018-12-11:
Thank @donghengqaz,

I found the issue because I use assert() macro.

Thank again.

closed 2018-12-11 by nghiaquy1991

donghengqaz, 2018-12-11:
You may try the new branch of v3.0 and later, we are trying to use new framework and components to make it more easier to use.

donghengqaz, 2018-12-11:
And the macro "assert" an be used at these branch.

nghiaquy1991, 2018-12-11:
Is the release/v3.x enough stable at this time? 
I chose release/v2.x.x because I think that It fixed many bugs and stable enough.
I maybe move to v3.x after few next weeks.
 Thanks.