espressif/ESP8266_RTOS_SDK #414: esp-idf风格的FreeRTOS SDK无法连接网络

mars9180, 2018-12-13:
你好：
     我使用最新版本的SDK测试其中的https_wolfssl发现无法连接至路由器，我使用V1.5版本的SDK是可以连接到此路由器的，以下是我的log信息，一直反复在打印scandone，我尝试过3个路由器，都是同样的现象，能帮忙分析以下是什么原因吗？是否是我的配置不对？sdkconfig文件请查看附件，谢谢。
LOG：
 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 7148, room 16 
tail 12
chksum 0xca
ho 0 tail 12 room 4
load 0x3ffe8408, len 24, room 12 
tail 12
chksum 0x73
ho 0 tail 12 room 4
load 0x3ffe8420, len 3508, room 12 
tail 8
chksum 0x27
I (49) boot: ESP-IDF v3.1-dev-376-gfea99ed 2nd stage bootloader
I (49) boot: compile time 11:41:46
I (49) boot: SPI Speed      : 40MHz
I (54) boot: SPI Mode       : QIO
I (58) boot: SPI Flash Size : 1MB
I (62) boot: Partition Table:
I (65) boot: ## Label            Usage          Type ST Offset   Length
I (73) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (80) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (87) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (95) boot:  3 ota_0            OTA app          00 10 00010000 00070000
I (102) boot:  4 ota_1            OTA app          00 11 00080000 00070000
I (110) boot: End of partition table
I (114) boot: No factory image, trying OTA 0
I (119) esp_image: segment 0: paddr=0x00010010 vaddr=0x40210010 size=0x630a8 (405672) map
I (216) esp_image: segment 1: paddr=0x000730c0 vaddr=0x3ffe8000 size=0x0053c (  1340) load
I (217) esp_image: segment 2: paddr=0x00073604 vaddr=0x3ffe853c size=0x0011c (   284) load
I (224) esp_image: segment 3: paddr=0x00073728 vaddr=0x40100000 size=0x06b30 ( 27440) load
I (238) boot: Loaded app from partition at offset 0x10000
I (4) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (13) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
W (20) phy_init: failed to load RF calibration data (0x1102), falling back to full calibration
W (274) phy_init: saving new calibration data because of checksum failure, mode(2)
I (338) example: Setting WiFi configuration SSID Xiaomi_E370...
mode : sta(18:fe:34:f1:65:28)
add if0
scandone
SNTP get time failed, retry after 3000 ms
scandone
scandone
SNTP get time failed, retry after 6000 ms
scandone
scandone
scandone
scandone
SNTP get time failed, retry after 12000 ms
scandone

[sdkconfig.zip](https://github.com/espressif/ESP8266_RTOS_SDK/files/2676043/sdkconfig.zip)

mars9180, 2018-12-14:
> `simple_wifi`这个例子测过吗？

谢谢你的回答，simple_wifi这个例子也测试过，是同样的效果呢。不知道还有其它办法吗？我现在卡在这里了，无法继续下去。

27-301, 2018-12-19:
1、你能在 SYSTEM_EVENT_STA_DISCONNECTED 的 event 中把 disconnect 的原因打印一下吗？
2、你是链接的一个非加密路由？

FayeY, 2018-12-28:
@mars9180 想请问，这个问题有什么进展吗？能否提供更多信息，解答上述疑问，以供分析？

closed 2019-01-09 by FayeY