espressif/ESP8266_RTOS_SDK #417: mbedtls: ssl_tls.c:2479 ssl->f_send() returned -78 (-0x004e)  UNKNOWN ERROR CODE

wanghui0574xs, 2018-12-18:
mbedtls: ssl_tls.c:2479 ssl->f_send() returned 181 (-0xffffff4b)
mbedtls: ssl_tls.c:2473 message length: 309, out_left: 128
mbedtls: ssl_tls.c:2479 ssl->f_send() returned -78 (-0x004e)
mbedtls: ssl_tls.c:2893 mbedtls_ssl_flush_output() returned -78 (-0x004e)
mbedtls: ssl_tls.c:7194 mbedtls_ssl_write_record() returned -78 (-0x004e)
mbedtls: ssl_tls.c:7275 <= write
[inf] _network_ssl_write(499): ssl write fail, code=-78, str=UNKNOWN ERROR CODE (004E)

27-301, 2018-12-27:
能确认下出现该 error 下的剩余内存吗？

Junhao-Espressif, 2019-01-15:
Hi @wanghui0574xs 

Could you give me your example? 
Thanks!

wanghui0574xs, 2019-01-15:
> Hi @wanghui0574xs
> 
> Could you give me your example?
> Thanks!

模块连接的路由器, 不断开模块连接, 而限制模块发送数据. 能复现.

wanghui0574xs, 2019-01-15:
模块连接的路由器, 不断开模块的连接, 通过限制模块发送数据, 能复现这个情况

wanghui0574xs@sina.com
 
From: Junhao-Espressif
Date: 2019-01-15 15:14
To: espressif/ESP8266_RTOS_SDK
CC: wanghui0574xs; Mention
Subject: Re: [espressif/ESP8266_RTOS_SDK] mbedtls: ssl_tls.c:2479 ssl->f_send() returned -78 (-0x004e) UNKNOWN ERROR CODE (#417)
Hi @wanghui0574xs
Could you give me your example?
Thanks!

Junhao-Espressif, 2019-01-15:
Hi @wanghui0574xs 

Please give me your example, or let me know how to disable ESP8266 send data to AP.

Thanks!

Junhao-Espressif, 2019-01-15:
Hi @wanghui0574xs ,

Please give me the SSL/TLS server name that you connect with.

Thank you!

FayeY, 2019-01-23:
请问，能否提供 SSL server 的信息，或者复现代码，以供查证？

FayeY, 2019-01-29:
Closing due to lack of response, please feel free to reopen if this is still an issue and you can supply more details. Thanks.

closed 2019-01-29 by FayeY

Fcheng-1, 2023-11-16:
My client got the same error,when handshake wth server.
`=> handshake
client state: 9
=> flush output
message length: 143, out_left: 143
ssl->f_send() returned -78 (-0x004e)
 <= handshake
mbedtls_ssl_handshake() returned -78 (-0
ssl write fail, code=-78, str=UNKNOWN ERROR CODE (004E)
`
Somebody help? Have a look?

Fcheng-1, 2023-11-16:
My client got the same error,when handshake wth server.

=> handshake 
client state: 9 
=> flush output message length: 143, out_left: 143
 ssl->f_send() returned -78 (-0x004e) 
<= handshake 
mbedtls_ssl_handshake() returned -78 (-0x004e)
ssl write fail, code=-78, str=UNKNOWN ERROR CODE (004E) 

Somebody help? Have a look?