espressif/ESP8266_RTOS_SDK #421: Compare with esp-open-rtos

MalcolmChen, 2018-12-23:
i'm developing a project with esp8266, using esp-open-rtos.
and it worked just well.
and i got another task to get familiar with esp32, witch makes me learn about the esp-idf,and i wanna start it with esp8266 -- witch i have now.
and i wanna know && need to know the difference between esp-idf and esp-free-rtos, cause i got some paper work to do, please help me

27-301, 2018-12-25:
what is `esp-free-rtos`?

MalcolmChen, 2018-12-25:
uh... i'm truely sorry for the typo, i am.
and one more question, the point is the typo ? isn't it? you guys just know how to piss me off. always! ain't you? from technical support and here, i can get nothing but mad.non help at all ! even i wanna pay for it?
so ... what's this repository for?
what's your company for?
well, merry mhristmas...

closed 2018-12-25 by MalcolmChen