espressif/ESP8266_RTOS_SDK #423: 编译后固件输出前面的[0;32mI 字符是怎么产生的,怎样去掉?

1XiaSheng, 2018-12-25:
[0;32mI (38) boot: compile time 15:25:51[0m
[0;32mI (44) qio_mode: Enabling default flash chip QIO[0m
[0;32mI (44) boot: SPI Speed      : 40MHz[0m
[0;32mI (44) boot: SPI Mode       : QIO[0m
[0;32mI (47) boot: SPI Flash Size : 4MB[0m
[0;32mI (49) boot: Partition Table:[0m
[0;32mI (52) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (56) boot:  0 nvs              WiFi data        01 02 00009000 00004000[0m
[0;32mI (61) boot:  1 otadata          OTA data         01 00 0000d000 00002000[0m
[0;32mI (66) boot:  2 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (71) boot:  3 ota_0            OTA app          00 10 00010000 000f0000[0m
[0;32mI (76) boot:  4 ota_1            OTA app          00 11 00110000 000f0000[0m

1XiaSheng, 2018-12-25:
[image]

xiongyumail, 2018-12-25:
你好，这个是终端字符颜色转义序列，你用make monitor就可以看到对应每一行的颜色了。使用普通的串口调试助手没有颜色效果，所以会打印奇怪的字符。

27-301, 2018-12-25:
如果你想关闭，可以把 Component config > Log output > Use ANSI terminal colors in log output 关闭即可，还可以使目标 bin 的大小有所减小。

1XiaSheng, 2018-12-26:
谢谢.你这个解答很nice

closed 2018-12-27 by 27-301