espressif/ESP8266_RTOS_SDK #424: 根据4MB OTA方式编译出来的固件输出信息,从哪里可以确认是用户可用的fileSysytem, nvs中用户可用的地址从哪里开始

1XiaSheng, 2018-12-26:
I (44) qio_mode: Enabling default flash chip QIO
I (44) boot: SPI Speed      : 40MHz
I (44) boot: SPI Mode       : QIO
I (44) boot: SPI Flash Size : 4MB
I (45) boot: Partition Table:
I (47) boot: ## Label            Usage          Type ST Offset   Length
I (51) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (55) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (60) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (64) boot:  3 ota_0            OTA app          00 10 00010000 000f0000
I (68) boot:  4 ota_1            OTA app          00 11 00110000 000f0000
I (72) boot: End of partition table
E (74) boot: ota data partition invalid and no factory, will try all partitions
I (79) esp_image: segment 0: paddr=0x00010010 vaddr=0x40210010 size=0x5cbe8 (379880) map
I (167) esp_image: segment 1: paddr=0x0006cc00 vaddr=0x3ffe8000 size=0x006bc (  1724) load
I (168) esp_image: segment 2: paddr=0x0006d2c4 vaddr=0x3ffe86bc size=0x000f8 (   248) load
I (171) esp_image: segment 3: paddr=0x0006d3c4 vaddr=0x40100000 size=0x06e1c ( 28188) load

请问,ESP-07S模块4MB FLASH,按照OTA配置后,用户自由可用的文件地址从哪里开始?

27-301, 2018-12-27:
当前的 partition 信息为：
I (51) boot: 0 nvs WiFi data 01 02 00009000 00004000
I (55) boot: 1 otadata OTA data 01 00 0000d000 00002000
I (60) boot: 2 phy_init RF data 01 01 0000f000 00001000
I (64) boot: 3 ota_0 OTA app 00 10 00010000 000f0000
I (68) boot: 4 ota_1 OTA app 00 11 00110000 000f0000

当前只用了 2MB，如果你想用还有 2MB 作为文件系统或其他参数区，需要修改 partition table cvs。
具体可以参看：https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/partition-tables.html

1XiaSheng, 2018-12-27:
辛苦解答,最好能有个示例, 看了还要照做才知道.

27-301, 2018-12-28:
```
# Name,   Type, SubType, Offset,   Size, Flags
# Note: if you change the phy_init or app partition offset, make sure to change the offset in Kconfig.projbuild
nvs,      data, nvs,     0x9000,  0x4000
otadata,  data, ota,     0xd000,  0x2000
phy_init, data, phy,     0xf000,  0x1000
ota_0,    0,    ota_0,   0x10000, 0xF0000
ota_1,    0,    ota_1,   0x110000,0xF0000
customer, 0x40, 0, 0x200000, 0x200000
```
你可以把 customer 分区作为你存放私有数据的区域。