espressif/ESP8266_RTOS_SDK #425: 经常crash， 百思不解，帮忙看看

bincodecn, 2018-12-26:
用的sdk是rtos v3.0中的simple_wifi。作为sta时，大概十分钟会crash一次。作为ap时，一连上去就crash。这是什么原因？

Task stack [ppT] stack from [3fff17d8] to [3fff1fd4], total [2048] size

          00000000 00000001 00000002 00000003 00000004 00000005 00000006 00000007 00000008 00000009 0000000a 0000000b 0000000c 0000000d 0000000e 0000000f 

3fff17d8  a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 
3fff1818  a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 
3fff1858  a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 
3fff1898  a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 
3fff18d8  a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 
3fff1918  a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 
3fff1958  a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 
3fff1998  a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 
3fff19d8  a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 
3fff1a18  a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 
3fff1a58  a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 
3fff1a98  a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 
3fff1ad8  a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 
3fff1b18  a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 
3fff1b58  a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 
3fff1b98  a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 
3fff1bd8  a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 
3fff1c18  a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 40107cb0 00000031 3ffef360 4010755c 40248f90 a5a5a5a5 00000000 
3fff1c58  3fff1670 402379da 00000000 00000000 3fff1670 40107cb0 00000006 3ffef360 4010755c 40248f90 00000000 00000000 3fff1670 402379da 00000000 00000000 
3fff1c98  3fff1670 40107cb0 00000029 3ffef360 4010755c 40248f90 00000000 00000000 3fff1670 402379da 00000000 00000000 3fff16c8 40237caa 0000000a 00000000 
3fff1cd8  4010755c 402494e4 0000000a 3fff1d20 4010755c 40248725 40247d20 3ffef360 4010755c 3ffea394 00000000 3fff16c8 40228f24 40228f47 3ffef360 00000004 
3fff1d18  00000000 4010435d 3ffea0d4 3fff4ad8 00000012 4022143d 40226207 00000000 3ffef360 3fff0468 00000000 00000000 3fff2f68 40237caa 40225cec 3ffea248 
3fff1d58  3ffea944 40222744 00000000 3fff1d90 00000030 3fff2ea8 40107d90 4023fda8 3fff2ea8 3fff2ea8 40107d90 4023fda8 3fff2e68 4024004d 3fff2e68 00080282 
3fff1d98  3fff2ea8 402411dc 00000030 00000166 3fff0558 402407bc 00000000 3fff16c8 40228f24 3fff1e50 00000001 40107d90 3fff2ea8 402407fc 3ffea1ac 3fff4808 
3fff1dd8  00000000 4024069f 00000001 40107d90 3fff1e00 4024c449 3ffef6ec 00000002 00000000 4024ce2c 00000000 3ffebdaa 000000ae 4024c408 40107d90 00000000 
3fff1e18  3ffea944 3fff4808 00000001 3ffe9360 3ffe87d0 00000001 000000ae 3ffebdaa 3ffe97d4 401026a8 00003333 ef68fb00 c0d3bc43 4022c849 3ffebdaa 000000ae 
3fff1e58  00000000 00000000 00000000 00000030 00000000 3fff07ba 00000002 3fff4ad8 3fff4ad8 3ffebdaa 00000018 3ffe97d4 3fff4808 3ffebdaa 3fff07a8 3ffe97d4 
3fff1e98  3fff4808 4022b012 40106ca4 40100452 00000030 4010049b 3fff1ef0 00000020 00000000 3ffee84c 3fff2dfc ffffffff 00000002 00000003 00000000 0002a974 
3fff1ed8  3ff20dc4 3fff16c8 00000000 00000001 00000000 0000001e 00000001 00000000 ffffffff 4010049b 00000001 00000000 3fff16c8 40237f2a 00000000 00000e48 
3fff1f18  3ffea228 401045a5 ffffffff 3fff1f80 00000040 00000000 00000002 00000001 3fff0298 00000001 00000000 3ffea390 3ffe87d0 40228ee3 00000000 00000000 
3fff1f58  3ffea944 402217ea a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 3ffea1c4 40221734 00000030 00000000 00000010 00000000 
3fff1f98  00000008 00000004 00000001 ffffffff 00000010 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 a5a5a5a5 

      PC: 0x3ff20c18       PS: 0x00000033       A0: 0x402218e9       A1: 0x3fff1f60 
0x402218e9: ppTask at ??:?

      A2: 0x3ff20c18       A3: 0x2c9f0300       A4: 0x0000000a       A5: 0x3fff1f84 
      A6: 0x3ffea1c4       A7: 0x00000000       A8: 0x00000003       A9: 0x00000000 
     A10: 0x0002a974      A11: 0x3ff20dc4      A12: 0x3ffea944      A13: 0x00000000 
     A14: 0x00000000      A15: 0x00000001      SAR: 0x0000001e EXCCAUSE: 0x00000002

xiongyumail, 2018-12-27:
有修改过例程嘛? 复位后的log提供一下

bincodecn, 2018-12-27:
问题解决，电源问题，用了一个250ma的ldo。现在换了一个大电流的dcdc就可以了。

closed 2018-12-27 by bincodecn