espressif/ESP8266_RTOS_SDK #426: load partition table error!

wanghui0574xs, 2018-12-27:
ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 4908, room 16 
0x40100000: _stext at ??:?

tail 12
chksum 0xe9
ho 0 tail 12 room 4
load 0x3ffe8008, len 4016, room 12 
tail 4
chksum 0xe5
csum 0xe5�[0;32mI (34) boot: ESP-IDF v0.0.0_f5a623d36-49-g66266d5-dirty 2nd stage bootloader
I (34) boot: compile time 02:02:04
D (35) boot: magic e9
D (37) boot: segments 02
D (39) boot: spi_mode 00
D (41) boot: spi_speed 00
D (43) boot: spi_size 08
I (46) boot: SPI Speed      : 40MHz
I (50) boot: SPI Mode       : QIO
I (54) boot: SPI Flash Size : 8MB
D (57) boot: mapped partition table 0x8000 at 0x40208000
E (63) flash_parts: partition 0 invalid magic number 0x0
E (69) boot: Failed to verify partition table
E (74) boot: load partition table error!

27-301, 2018-12-27:
`E (63) flash_parts: partition 0 invalid magic number 0x0`

位于 0x8000 地址的 partition 分区格式不对，感觉烧写就没对。能先确认你的烧写吗？

wanghui0574xs, 2018-12-27:
font{
        line-height: 1.6;
    }
    ul,ol{
        padding-left: 20px;
        list-style-position: inside;
    }

    
    
        
            
        
    
    模组一开始是正常工作的. 后来不知道是不是被静电打到了还是什么, 就出现这种情况.在mac底下烧写, 是正确执行的.也显示烧写完成, 在windows下, 我试着想连接, 没连上.我把flash换到别的模组上, 能正常工作起来.
    
        
            
        
    
    
          
        
            
    
        font{
            line-height: 1.5;
        }
    
----------------------------wanghui0574xs@sina.com

closed 2019-01-02 by 27-301

Tmy-BedKid, 2019-03-05:
我也遇见了上诉问题，我是这样解决的。---Thanks,

E (76) flash_parts: partition 0 invalid magic number 0x0
E (85) boot: Failed to verify partition table
E (93) boot: load partition table error!

mark 下，如果在搭建ESP8266 的Ubuntu 环境的时候，下载遇见上诉错误，make menuconfig -> Serial flasher config 中的配置参数,;
正确的参数是: 
{
2nd boot version : 1.5
  SPI Speed      : 40MHz
  SPI Mode       : DOUT
  SPI Flash Size & Map: 32Mbit(512KB+512KB)
jump to run user1 @ 1000
}

zenz, 2021-07-14:
> 我也遇见了上诉问题，我是这样解决的。---Thanks,
>
> E (76) flash_parts: partition 0 invalid magic number 0x0
> E (85) boot: Failed to verify partition table
> E (93) boot: load partition table error!
>
> mark 下，如果在搭建ESP8266 [...]

useful tip.
the important is the SPI mode should be DOUT for some module.