espressif/ESP8266_RTOS_SDK #428: esp8266 OTA problem

fghjhuang, 2018-12-29:
when i add the OTA function to the esp8266， RTOS SDK V2.0.0，it keep showing an error：
/cygdrive/d/esp8266_new/projects/litehome_gatewayv2/gatewayV2/app/user/user_main.c:329: undefined reference to `system_upgrade_start'

but i had make sure i already include all the file i need. the source code like this :
`
void ota_finished_callback(void *arg) 
	struct upgrade_server_info *server = arg;
	if (server->upgrade_flag == true) {
		printf("OTA  Success ! rebooting!\n");
		system_upgrade_reboot();
	} else {
		printf("OTA failed!\n");
	}
}

/**
 * server_ip: 服务器地址
 * port:服务器端口
 * path:文件路径
 */
void ota_start_Upgrade(void) {
	struct upgrade_server_info *update= NULL;

	const char* file;
	update= (struct upgrade_server_info *)zalloc(sizeof(struct upgrade_server_info)*10);
	//获取系统的目前加载的是哪个bin文件
	uint8 userBin = system_upgrade_userbin_check();

	switch (userBin) {

	
	case UPGRADE_FW_BIN1:
		file = "user2.4096.new.6.bin";
		break;

		
	case UPGRADE_FW_BIN2:
		file = "user1.4096.new.6.bin";
		break;

		
	default:
		printf("Fail read system_upgrade_userbin_check! \n\n");
		return;
	}

	printf(file);
	printf("\r\n");
	//update->pespconn = (struct espconn *) os_zalloc(sizeof(struct espconn));
	//设置服务器地址
	bzero(&update->sockaddrin, sizeof(struct sockaddr_in) );

	update->sockaddrin.sin_family = AF_INET;
	update->sockaddrin.sin_addr.s_addr = inet_addr("192.168.3.12");
	update->sockaddrin.sin_port = htons( 80 );
	update->sockaddrin.sin_len = sizeof( struct sockaddr );

	//memcpy(update->sockaddrin.sin_addr.s_addr, server_ip, 4);
	//设置服务器端口
	//update->port = port;
	update->sockaddrin.sin_port = 80;
	//设置OTA回调函数
	//update->check_cb = ota_finished_callback;
	//设置定时回调时间
	update->check_times = 10000;
	//从 4M *1024 =4096申请内存
	update->url = (uint8 *) zalloc(4096);

	//打印下請求地址
	printf(
			"Http Server Address:%d.%d.%d.%d ,port: %d,filePath: %s,fileName: %s \n",
			(update->sockaddrin.sin_addr.s_addr&0x000000ff),
			(update->sockaddrin.sin_addr.s_addr&0x0000ff00)>>8,
			(update->sockaddrin.sin_addr.s_addr&0x00ff0000)>>16,
			(update->sockaddrin.sin_addr.s_addr&0xff000000)>>24,
			update->sockaddrin.sin_port, "litehonepanel1ch/", file);

	//拼接完整的 URL去请求服务器
	sprintf((char*) update->url, "GET /%s%s HTTP/1.1\r\n"
			"Host: 192.168.3.12:80\r\n"
	"Connection: keep-alive\r\n"
	"\r\n", "litehonepanel1ch/", file);
	printf("\r\n");
printf((char*) update->url);
printf("\r\n");
system_upgrade_start(update);
	if (!1) {
		printf(" Could not start upgrade\n");
	} else {
		printf(" Upgrading...\n");
	}
}
`

ustccw, 2019-01-02:
@fghjhuang Hi, here is a [ota demo](https://github.com/espressif/ESP8266_RTOS_SDK/tree/release/v2.x.x/examples/ota_demo) in RTOS v2.0, please have a look.

fghjhuang, 2019-01-14:
it turn out that this api only can be apply in NONOS version, in RTOS version i had to write my own OTA function.

closed 2019-01-14 by fghjhuang