espressif/ESP8266_RTOS_SDK #429: lwip tcp socket_mt send不会等待数据发送结束

981213, 2018-12-29:
SDK版本为当前master (189f1d067b2fe58db5a608056266b46c0f80f66e)
尝试用socket写HTTP服务器(代码附后)，但是并不能用，curl报Connection reset by peer:
```
*   Trying 192.168.2.100...
* TCP_NODELAY set
* Connected to 192.168.2.100 (192.168.2.100) port 80 (#0)
> GET / HTTP/1.1
> Host: 192.168.2.100
> User-Agent: curl/7.61.1
> Accept: */*
> 
< HTTP/1.1 200 OK
< Content-type: text/html
* no chunk, no close, no size. Assume close to signal end
< 
* Recv failure: Connection reset by peer
* Closing connection 0
curl: (56) Recv failure: Connection reset by peer
```

猜测send不会阻塞到发送结束或者close不会等待发送完成，在close前加个延时可以缓解这个现象（然而只是降低出现频率）
禁用socket_mt后问题消失

```c
/*
 * HTTP server based on LwIP httpserver-netconn example.
 */
#include <lwip/err.h>
#include <lwip/sockets.h>
#include <lwip/sys.h>
#include <lwip/netdb.h>

#include <fcntl.h>

#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_log.h"
#include "esp_system.h"

#include <stdlib.h>

#define HTTP_PORT 80
#define HTTP_LOGTAG "httpd"
#define HTTP_RECV_BUF_LEN 128

static const char http_err_hdr[] = "HTTP/1.1 500 Internal Server Error\r\n\r\n";

static const char http_html_hdr[] =
	"HTTP/1.1 200 OK\r\nContent-type: text/html\r\n\r\n";
static const char http_index_html[] =
	"<html><head><title>Congrats!</title></head><body><h1>Welcome to our lwIP HTTP server!</h1><p>This is a small test page.</body></html>";

static void httpd_cli_handler(int cli_socket)
{
	char *buf;
	int ret;
	buf = (char *)malloc(HTTP_RECV_BUF_LEN);
	if (!buf) {
		ESP_LOGE("HTTP_LOGTAG",
			 "failed to allocate memory for current request.");
		send(cli_socket, http_err_hdr, sizeof(http_err_hdr) - 1, 0);
		return;
	}
	ret = recv(cli_socket, buf, HTTP_RECV_BUF_LEN, 0);
	if (ret > 0) {
		/* Is this an HTTP GET command? (only check the first 5 chars, since
		there are other formats for GET, and we're keeping it very simple )*/
		if (ret >= 5 && buf[0] == 'G' && buf[1] == 'E' &&
		    buf[2] == 'T' && buf[3] == ' ' && buf[4] == '/') {
			send(cli_socket, http_html_hdr,
			      sizeof(http_html_hdr) - 1, 0);
			send(cli_socket, http_index_html,
			      sizeof(http_index_html) - 1, 0);
		}
	}
	
	close(cli_socket);
	free(buf);
}

static void httpd_worker(void)
{
	int server_socket;
	int ret;
	int client_socket;

	struct sockaddr_in server_addr = {
		.sin_addr.s_addr = htonl(INADDR_ANY),
		.sin_family = AF_INET,
		.sin_port = htons(HTTP_PORT),
	};
	struct sockaddr_in client_addr;
	size_t addr_len = sizeof(client_addr);
	server_socket = socket(AF_INET, SOCK_STREAM, IPPROTO_IP);
	if (server_socket < 0) {
		ESP_LOGE(HTTP_LOGTAG, "failed to create socket.");
		return;
	}

	ret = bind(server_socket, (struct sockaddr *)&server_addr,
		   sizeof(server_addr));
	if (ret < 0) {
		ESP_LOGE(HTTP_LOGTAG, "failed to bind socket. errno %d", errno);
		goto ERR0;
	}
	ret = listen(server_socket, 1);
	if (ret < 0) {
		ESP_LOGE(HTTP_LOGTAG, "failed to listen: errno %d", errno);
		goto ERR0;
	}

	while ((client_socket =
			accept(server_socket, (struct sockaddr *)&client_addr,
			       &addr_len)) > 0) {
		httpd_cli_handler(client_socket);
	}

ERR0:
	shutdown(server_socket, 0);
	close(server_socket);
	return;
}

static void httpd_task(void *pvParameters)
{
	while (1) {
		httpd_worker();
		ESP_LOGE(HTTP_LOGTAG, "httpd worker stopped.");
	}
	vTaskDelete(NULL);
}

void httpd_init(void)
{
	xTaskCreate(httpd_task, "httpd_task", 2048, NULL, 13, NULL);
}
```

donghengqaz, 2018-12-31:
socket_mt 出于易用性考虑，创建的 socket 会强制 so_link 使得 close 时强制释放资源，这是个问题，我会添加配置项来屏蔽这个功能。

bincodecn, 2019-01-16:
struct linger linger;

	linger.l_onoff = 0;
	linger.l_linger = 0;

	// If turn it on, there will lost data when close sock.
	setsockopt(sock, SOL_SOCKET, SO_LINGER, &linger, sizeof(linger));

我也遇到这个问题了，关了SO_LINGER就可以了。

981213, 2019-01-16:
呃…现在menuconfig有选项可用了，这issue我关了……

closed 2019-01-16 by 981213