espressif/ESP8266_RTOS_SDK #432: 刚刚更新了SDK,新编译后无法启动

1XiaSheng, 2018-12-29:
I (55) boot: compile time 14:19:14
I (56) boot: SPI Speed      : 40MHz
I (56) boot: SPI Mode       : QIO
I (59) boot: SPI Flash Size : 4MB
I (62) boot: Partition Table:
I (65) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (78) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (84) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (91) boot:  3 ota_0            OTA app          00 10 00010000 000f0000
I (97) boot:  4 ota_1            OTA app          00 11 00110000 000f0000
I (104) boot: End of partition table
E (107) boot: ota data partition invalid and no factory, will try all partitions
I (114) esp_image: segment 0: paddr=0x00010010 vaddr=0x40210010 size=0x5f908 (391432) map
I (299) esp_image: segment 1: paddr=0x0006f920 vaddr=0x3ffe8000 size=0x006d4 (  1748) load
I (300) esp_image: segment 2: paddr=0x0006fffc vaddr=0x3ffe86d4 size=0x000f8 (   248) load
I (304) esp_image: segment 3: paddr=0x000700fc vaddr=0x40100000 size=0x06dec ( 28140) load
I (324) boot: Loaded app from partition at offset 0x10000

No task
      PC: 0x40254101       PS: 0x00000030       A0: 0x40254478       A1: 0x3ffffec0 
      A2: 0x60000208       A3: 0x016aa101       A4: 0x006a8101       A5: 0x401006a3 
      A6: 0x00000000       A7: 0x00000000       A8: 0x00000000       A9: 0x0000ffff 
     A10: 0x00000001      A11: 0x00000002      A12: 0x3fffff10      A13: 0x00000002 
     A14: 0x00000000      A15: 0x00000000      SAR: 0x00000005 EXCCAUSE: 0x00000009 

bootloader.bin 体积变小了

27-301, 2018-12-29:
和 sdkconfig 有关吗？把 sdkconfig 发上来看看。

你拿个自带的 example 试试呢？
把 sdkconfig，build 目录删除，重新编译下呢？

zhubinghan, 2019-01-04:
我今天用了最新的试了下。
log打印串口就输出这些

 ets Jan  8 2013,rst cause:1, boot mode:(3,6)

load 0x40100000, len 4508, room 16 
tail 12
chksum 0x8d
ho 0 tail 12 room 4
load 0x3ffe8408, len 24, room 12 
tail 12
chksum 0x04
ho 0 tail 12 room 4
load 0x3ffe8420, len 1024, room 12 
tail 4
chksum 0x04
csum 0x04?&狿zt吶kN?

也重新编译了。不行。

27-301, 2019-01-04:
使用 commit： 50f39bb1
示例：examples/get-started/project_template

```
rm -rf build sdkconfig sdkconfig.old
make defconfig
make flash monitor
```

log 如下：
```
 ets Jan  8 2013,rst cause:1, boot mode:(3,6)

load 0x40100000, len 7188, room 16 
tail 4
chksum 0x6b
load 0x3ffe8408, len 24, room 4 
tail 4
chksum 0xe2
load 0x3ffe8420, len 3548, room 4 
tail 8
chksum 0x86
I (43) boot: ESP-IDF v3.1-dev-456-g50f39bb1 2nd stage bootloader
I (43) boot: compile time 16:46:57
I (43) qio_mode: Enabling default flash chip QIO
I (52) boot: SPI Speed      : 40MHz
I (58) boot: SPI Mode       : QIO
I (64) boot: SPI Flash Size : 2MB
I (70) boot: Partition Table:
I (76) boot: ## Label            Usage          Type ST Offset   Length
I (87) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (98) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (110) boot:  2 factory          factory app      00 00 00010000 000f0000
I (122) boot: End of partition table
I (128) esp_image: segment 0: paddr=0x00010010 vaddr=0x40210010 size=0x2fb50 (195408) map
I (207) esp_image: segment 1: paddr=0x0003fb68 vaddr=0x3ffe8000 size=0x0053c (  1340) load
I (208) esp_image: segment 2: paddr=0x000400ac vaddr=0x3ffe853c size=0x0011c (   284) load
I (218) esp_image: segment 3: paddr=0x000401d0 vaddr=0x40100000 size=0x05914 ( 22804) load
0x40100000: _stext at ??:?

I (239) boot: Loaded app from partition at offset 0x10000
I (286) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (296) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (506) reset_reason: RTC reset 1 wakeup 0 store 0, reason is 1
SDK version:v3.1-dev-456-g50f39bb1
```
请确认使用的相关硬件。

zhubinghan, 2019-01-04:
[0;32mI (65) boot: compile time 17:13:12[0m
[0;32mI (74) qio_mode: Enabling default flash chip QIO[0m
[0;32mI (74) boot: SPI Speed      : 40MHz[0m
[0;32mI (74) boot: SPI Mode       : QIO[0m
[0;32mI (77) boot: SPI Flash Size : 2MB[0m
[0;32mI (81) boot: Partition Table:[0m
[0;32mI (85) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (92) boot:  0 nvs              WiFi data        01 02 00009000 00006000[0m
[0;32mI (99) boot:  1 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (107) boot:  2 factory          factory app      00 00 00010000 000f0000[0m
[0;32mI (114) boot: End of partition table[0m
[0;32mI (119) esp_image: segment 0: paddr=0x00010010 vaddr=0x40210010 size=0x4fe58 (327256) map[0m
[0;31mE (128) bootloader_flash: bootloader_flash_read dest 0x3ffe8001 not 4-byte aligned[0m
[0;31mE (136) esp_image: bootloader read flash @ 0x00010010 1024 error -1[0m
[0;31mE (143) boot: Factory app partition is not bootable[0m
[0;31mE (148) boot: No bootable app partitions in the partition table[0m
user code done
我自己的程序就这样，用你说的例子里面的是和你一样（除了最后一行没打出：后面的版本号）。
难道哪里配置有问题

ustccw, 2019-01-04:
@zhubinghan 试下 make erase_flash 呢

27-301, 2019-01-04:
这个与你之前报的问题已经不是一个了。

从当前 log 看，现在问题出在：
�[0;31mE (128) bootloader_flash: bootloader_flash_read dest 0x3ffe8001 not 4-byte aligned�[0m

方便的话，请提供可复现问题的工程，（可去除关键代码）。

1XiaSheng, 2019-01-04:
已经回滚了SDK, 赶得急,报告给你们就没有深究了

zhubinghan, 2019-01-04:
现在可以了，估计是你们这个sdkconfig文件不一样，我把新下的例子里面的sdkconfig拷贝到我的项目里，改了下就好了

27-301, 2019-01-04:
@zhubinghan 之前的 sdkconfig 有备份吗？可以再分析一下。

@1XiaSheng 抽空可以试下最新的，看看还有没有问题。

1XiaSheng, 2019-01-04:
刚准备跳过的, 现在又碰到了.
$`?oo$`os徾抩r鞉`r捽n{湟I (66) boot: ESP-IDF v3.1-dev-440-g189f1d0 2nd stage bootloader
I (66) boot: compile time 18:35:29
I (76) qio_mode: Enabling default flash chip QIO
I (76) boot: SPI Speed      : 40MHz
I (76) boot: SPI Mode       : QIO
I (76) boot: SPI Flash Size : 4MB
I (79) boot: Partition Table:
I (81) boot: ## Label            Usage          Type ST Offset   Length
I (88) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (94) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (101) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (107) boot:  3 ota_0            OTA app          00 10 00010000 000f0000
I (114) boot:  4 ota_1            OTA app          00 11 00110000 000f0000
I (120) boot: End of partition table
E (124) boot: ota data partition invalid and no factory, will try all partitions
I (131) esp_image: segment 0: paddr=0x00010010 vaddr=0x40210010 size=0x59ba8 (367528) map
E (139) bootloader_flash: bootloader_flash_read dest 0x3ffe8001 not 4-byte aligned
E (146) esp_image: bootloader read flash @ 0x00010010 1024 error -1
E (152) boot: OTA app partition slot 0 is not bootable
E (157) esp_image: image at 0x110000 has invalid magic byte
W (162) esp_image: image at 0x110000 has invalid SPI mode 255
W (168) esp_image: image at 0x110000 has invalid SPI size 15
E (173) boot: OTA app partition slot 1 is not bootable
E (178) boot: No bootable app partitions in the partition table
user code done

1XiaSheng, 2019-01-04:
[sdkconfig.zip](https://github.com/espressif/ESP8266_RTOS_SDK/files/2726659/sdkconfig.zip)

1XiaSheng, 2019-01-04:
[image]
[image]

1XiaSheng, 2019-01-04:
主要是这两个区别, 改为release  和silent 后不能启动

27-301, 2019-01-04:
> [sdkconfig.zip](https://github.com/espressif/ESP8266_RTOS_SDK/files/2726659/sdkconfig.zip)
> SDK没有更新,刚刚操作了一下 make menuconfig, 把所有的波特率改为115200 ;
> 
> 提个建议啊,能否SDK默认的波特率全部改为115200, 现在的元器件和工艺还用74800 很恼火,每次需要去修改.

如果默认改成 74880 的话，rom 里的打印就会是乱码了。

27-301, 2019-01-04:
@1XiaSheng 我用你使用的 commit 和 sdkconfig，没有遇到问题。log 如下：
```
ESP-IDF v3.1-dev-440-g189f1d06 2nd stage bootloader
I (47) boot: compile time 18:59:00
I (48) qio_mode: Enabling default flash chip QIO
I (50) boot: SPI Speed      : 40MHz
I (53) boot: SPI Mode       : QIO
I (56) boot: SPI Flash Size : 4MB
I (59) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (68) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (74) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (81) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (87) boot:  3 ota_0            OTA app          00 10 00010000 000f0000
I (94) boot:  4 ota_1            OTA app          00 11 00110000 000f0000
I (100) boot: End of partition table
I (104) boot: No factory image, trying OTA 0
I (108) esp_image: segment 0: paddr=0x00010010 vaddr=0x40210010 size=0x2ebf0 (191472) map
I (169) esp_image: segment 1: paddr=0x0003ec08 vaddr=0x3ffe8000 size=0x0052c (  1324) load
I (170) esp_image: segment 2: paddr=0x0003f13c vaddr=0x3ffe852c size=0x000ec (   236) load
I (174) esp_image: segment 3: paddr=0x0003f230 vaddr=0x40100000 size=0x058dc ( 22748) load
0x40100000: _stext at ??:?

I (188) boot: Loaded app from partition at offset 0x10000
I (228) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (228) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (428) reset_reason: RTC reset 1 wakeup 0 store 0, reason is 1
SDK version:v3.1-dev-440-g189f1d06
```
1. 请问你使用的 gcc 是哪个版本的？
2. 你用 examples/get-started/project_template 测试是否是正常的？
3. 能提供你的 elf 文件吗？

27-301, 2019-01-04:
@1XiaSheng @zhubinghan 请临时修改 esp_image_format.c 中 process_segment_data 内：
将 `static char pbuf[SHA_CHUNK];` 修改为 `static uint32_t pbuf[SHA_CHUNK / 4];`

另外，请提供一下 build/bootloader/bootloader.elf 文件，需要分析一下，看看是什么导致 `E (139) bootloader_flash: bootloader_flash_read dest 0x3ffe8001 not 4-byte aligned`。

donghengqaz, 2019-01-07:
最新版本 master 已经做了相关问题的修复。

1XiaSheng, 2019-01-08:
更新到最新测试解决了.

closed 2019-01-08 by 1XiaSheng