espressif/ESP8266_RTOS_SDK #433: OTA: problem/error with esp_ota_begin => dev 1319

THKDev, 2018-12-31:
## Environment

- Development Kit: SonOff Basic (ESP8285)
- IDF version (``git rev-parse --short HEAD`` to get the commit id.): 
    //3d0fa3c5aedd7a275143ae5dcee39b7cad385622
- Development Env:      Make
- Operating System:     Linux
- Power Supply:         external 3.3V

## Problem Description
Sometimes OTA failed after calling function `esp_ota_begin(ota_data.update_partition, OTA_SIZE_UNKNOWN, &ota_data.update_handle);` Function returns ESP_OK but on the serial output displays `dev 1319` . If this output happens, then the OTA update fails.

### Expected Behavior
esp_ota_begin should return an error in case something went wrong.

### Actual Behavior
OTA dies without any error message.

### Steps to repropduce
At the moment i have no simple test program. May be you can find the problem anyway.

## Debug Logs
```
2018-12-31 12:42:36.924:  I (56288) boot: compile time 21:44:55
2018-12-31 12:42:36.928:  I (56289) boot: SPI Speed      : 40MHz
2018-12-31 12:42:36.932:  I (56291) boot: SPI Mode       : DOUT
2018-12-31 12:42:36.937:  I (56296) boot: SPI Flash Size : 1MB
2018-12-31 12:42:36.941:  I (56301) boot: Partition Table:
2018-12-31 12:42:36.945:  I (56305) boot: ## Label            Usage          Type ST Offset   Length
2018-12-31 12:42:36.958:  I (56315) boot:  0 nvs              WiFi data        01 02 00009000 00004000
2018-12-31 12:42:36.969:  I (56325) boot:  1 otadata          OTA data         01 00 0000d000 00002000
2018-12-31 12:42:36.979:  I (56336) boot:  2 phy_init         RF data          01 01 0000f000 00001000
2018-12-31 12:42:36.988:  I (56346) boot:  3 ota_0            OTA app          00 10 00010000 00070000
2018-12-31 12:42:36.997:  I (56356) boot:  4 ota_1            OTA app          00 11 00080000 00070000
2018-12-31 12:42:37.009:  I (56367) boot: End of partition table
2018-12-31 12:42:37.014:  I (56372) esp_image: segment 0: paddr=0x00080010 vaddr=0x40210010 size=0x5dd18 (384280) map
2018-12-31 12:42:37.180:  I (56553) esp_image: segment 1: paddr=0x000ddd30 vaddr=0x3ffe8000 size=0x00640 (  1600) load
2018-12-31 12:42:37.189:  I (56554) esp_image: segment 2: paddr=0x000de378 vaddr=0x3ffe8640 size=0x00104 (   260) load
2018-12-31 12:42:37.201:  I (56561) esp_image: segment 3: paddr=0x000de484 vaddr=0x40100000 size=0x064a4 ( 25764) load
2018-12-31 12:42:37.243:  I (56585) boot: Loaded app from partition at offset 0x80000
2018-12-31 12:42:37.274:  I (56650) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
2018-12-31 12:42:37.287:  I (56655) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
2018-12-31 12:42:37.517:  I (56880) reset_reason: RTC reset 1 wakeup 0 store 3, reason is 3
2018-12-31 12:53:29.239:  I (708615) esp_err_t prepare_ota(): Running partition type 0 subtype 17 (offset 0x00080000)
2018-12-31 12:53:29.248:  I (708620) esp_err_t prepare_ota(): Erasing partition at 0x10000 of size 458752 bytes
2018-12-31 12:53:29.336:  dev 1319
2018-12-31 12:53:56.093:  BK	%u@Y+QuB	-))5PKT\	)C)LZPJJ	RKTZ
2018-12-31 12:53:56.108:  @\%```

27-301, 2019-01-02:
Seems that you don't initial WiFi? Can you provide some reference codes to reproduce this issue?

THKDev, 2019-01-02:
I reproduced the issue with [OTA example](https://github.com/espressif/ESP8266_RTOS_SDK/tree/master/examples/system/ota).
In the attached logs you can see that the firts OTA update is ok. The second one crashes ~25 seconds after printing out **"dev 1319"** .
- [x] sdkconfig
- [x] [log.zip](https://github.com/espressif/ESP8266_RTOS_SDK/files/2721690/log.zip)

27-301, 2019-01-04:
@THKDev We will check about this.

Junhao-Espressif, 2019-01-04:
Hi @THKDev 

Could you capture WiFi packet and send it to me?

Could you provide your router model？

Thanks!

THKDev, 2019-01-05:
With the special OTA handling (for 1MB flash) inside the universal_ota sample all works as expected.

closed 2019-01-05 by THKDev