espressif/ESP8266_RTOS_SDK #435: esp8266作为http服务器，连续请求多次后请求失败。

bincodecn, 2019-01-01:
## Environment

- Development Kit: esp8266-12f
- IDF version: f561b36
- Development Env:     Make
- Operating System:     MacOS
- Power Supply:         Battery

## Problem Description

测试命令：
while true;do curl http://192.168.4.31/sensor.json;echo; done
192.168.4.31是esp8266作为sta的地址。是个小型的http服务器，用的是bsd socket api。
在sendbuf为2920时会在210次左右请求后，8266 send数据就会失败。sendbuf为4096时会在230次左右会失败。这时8266是可以正常recv到数据，但是send一直失败。
没有仿真器调试很麻烦，请帮忙看看啥问题。有必要的话，可以提供固件。

以下时部分log
`
lwip_recvfrom(1, 0x40107f70, 1, 0x0, ..)
lwip_recvfrom: top while sock->lastdata=0x3ffff8c8
lwip_recvfrom: buflen=87 len=1 off=0 sock->lastoffset=85
lwip_recvfrom(1): addr=192.168.4.208 port=60876 len=1
lwip_recvfrom: lastdata now netbuf=0x3ffff8c8
lwip_recvfrom(1, 0x40107f70, 1, 0x0, ..)
lwip_recvfrom: top while sock->lastdata=0x3ffff8c8
lwip_recvfrom: buflen=87 len=1 off=0 sock->lastoffset=86
lwip_recvfrom(1): addr=192.168.4.208 port=60876 len=1
lwip_recvfrom: deleting netbuf=0x3ffff8c8
[DEBUG] Respond
[DEBUG] Send response 
{"temp":10.25,"light":0.0322265625}

lwip_send(1, data=0x40107f70, size=17, flags=0x0)
lwip_send(1) err=-7 written=0
[ERROR] Error occoured where sending response. errno 11
[DEBUG] 500

lwip_send(1, data=0x40235418, size=83, flags=0x0)
0x40235418: chip_v6_unset_chanfreq at ??:?

lwip_send(1) err=-7 written=0
[DEBUG] close client 500
`

bincodecn, 2019-01-02:
初步看，在lwip的mem.c的malloc有很多大小为208的内存没有释放。猜测是pbuf的leak。

donghengqaz, 2019-01-02:
你是 TCP socket 用的 non-blocking 模式？

bincodecn, 2019-01-02:
> 你是 TCP socket 用的 non-blocking 模式？

不是。

经过研究，结论如下：
我的这个项目需要设置esp_wifi_set_ps(WIFI_PS_MAX_MODEM);
这个模式在cpu空闲的情况下，cpu会暂停。这可能会导致lwip线程没有太多时间释放相应资源。
在我没有设置WIFI_PS_MAX_MODEM时，在大量访问的情况下，也会有OOM的情况，但是会有内存释放的动作，会恢复工作。若是设置了WIFI_PS_MAX_MODEM，就会一直拒绝服务。

在我的项目里，做压力测试的话，很快就会死掉。正常使用的话，短的时候只能坚持半天。希望你们改进这个问题。

donghengqaz, 2019-01-04:
返回 -7 是 ERR_WOULDBLOCK，表示使用 non blocking 模式，在这个模式下 socket 的状态同步很慢，就是 send/recv 会很慢。不知道这种问题会不会最终导致你的问题。

bincodecn, 2019-01-05:
非也，我用的是blocking模式。这个模式下，设置了超时的话，也会返回ERR_WOULDBLOCK。在我的例子里，根据我跟踪得知，会出现这个错误是因为出现OOM无法分配recv buf，导致一直重试，直到超时。OOM的原因是如我前面所说，light sleep模式下cpu暂停，大量的pbuf(size 192)没有释放。调试得知，oom的时候，系统只有2.7k的剩余内存了，然后有大量未释放的pbuf。

981213, 2019-01-16:
试试在lwipopts.h中定义TCP_MSL，比如2秒：
```c
/* reduce TCP_MSL to save memory during TCP TIME_WAIT */
#define TCP_MSL				2000UL
```

作为服务端主动关闭连接后tcp连接会在TIME_WAIT状态下等待两倍TCP_MSL的时间，这期间内存不会被释放，默认的TCP_MSL是60000ms，所以内存要等待2分钟后才会被释放，这对拿esp8266当httpd非常不友好。。。

981213, 2019-01-16:
呃……有个workaround:
如果你内存够多的话，可以使用lwip的静态mempool，然后就不存在申请/释放pbuf的问题了，lwip也会自动重用处在TCP TIME_WAIT状态下的内存。（mempool大小应该可以调，我没仔细看过）
如下修改我用在lwip自带的httpd上能正常用而且跑while curl也不会死，我不知道用在socket上是什么效果……
```diff
diff --git a/components/r_lwip/port/esp8266/include/lwipopts.h b/components/r_lwip/port/esp8266/include/lwipopts.h
index d3df94e..cb6529e 100644
--- a/components/r_lwip/port/esp8266/include/lwipopts.h
+++ b/components/r_lwip/port/esp8266/include/lwipopts.h
@@ -260,7 +260,7 @@ size_t memp_malloc_get_size(size_t type);
  * ATTENTION: Currently, this uses the heap for ALL pools (also for private pools,
  * not only for internal pools defined in memp_std.h)!
  */
-#define MEMP_MEM_MALLOC                 1
+#define MEMP_MEM_MALLOC                 0
 
 /**
  * MEM_ALIGNMENT: should be set to the alignment of the CPU
@@ -304,7 +304,7 @@ size_t memp_malloc_get_size(size_t type);
  * (requires the LWIP_TCP option)
  */
 #define MEMP_NUM_TCP_PCB                CONFIG_LWIP_MAX_ACTIVE_TCP
-
+#define MEMP_NUM_PPP_PCB                1
 /**
  * MEMP_NUM_TCP_PCB_LISTEN: the number of listening TCP connections.
  * (requires the LWIP_TCP option)
@@ -526,8 +526,9 @@ size_t memp_malloc_get_size(size_t type);
  * this option does not affect outgoing packet sizes, which can be controlled
  * via IP_FRAG.
  */
-#define IP_REASSEMBLY                   CONFIG_LWIP_IP_REASSEMBLY
-
+#ifdef CONFIG_LWIP_IP_REASSEMBLY
+#define IP_REASSEMBLY                   1
+#endif
 /**
  * IP_FRAG==1: Fragment outgoing IP packets if their size exceeds MTU. Note
  * that this option does not affect incoming packet sizes, which can be
@@ -719,8 +720,9 @@ size_t memp_malloc_get_size(size_t type);
 /**
  * LWIP_AUTOIP==1: Enable AUTOIP module.
  */
-#define LWIP_AUTOIP                     CONFIG_LWIP_AUTOIP
-
+#ifdef CONFIG_LWIP_AUTOIP
+#define LWIP_AUTOIP                     1
+#endif
 /**
  * LWIP_DHCP_AUTOIP_COOP==1: Allow DHCP and AUTOIP to be both enabled on
  * the same interface at the same time.
@@ -1722,7 +1724,9 @@ size_t memp_malloc_get_size(size_t type);
 /**
  * LWIP_IPV6==1: Enable IPv6
  */
-#define LWIP_IPV6                       CONFIG_LWIP_IPV6
+#ifdef CONFIG_LWIP_IPV6
+#define LWIP_IPV6                       1
+#endif
 
 /**
  * LWIP_IPV6_NUM_ADDRESSES: Number of IPv6 addresses per netif.
```

bincodecn, 2019-01-16:
经测试，改用memory poll后确实没有内存被吃光的情况了。不过在3.1-rc1启用memory pool的时候，会爆iram，关LWIP_GLOBAL_DATA_LINK_IRAM就可以了。

closed 2019-01-23 by FayeY

JunJianX, 2020-12-17:
> LINK_IRAM

> 经测试，改用memory poll后确实没有内存被吃光的情况了。不过在3.1-rc1启用memory pool的时候，会爆iram，关LWIP_GLOBAL_DATA_LINK_IRAM就可以了。

> 经测试，改用memory poll后确实没有内存被吃光的情况了。不过在3.1-rc1启用memory pool的时候，会爆iram，关LWIP_GLOBAL_DATA_LINK_IRAM就可以了。

请问爆iram是编译的时候提示这个吗？`/home/xu/esp/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/5.2.0/../../../../xtensa-lx106-elf/bin/ld: /home/xu/esp/ESP_Server/examples/wifi/smart_config/build/smart_config.elf section `.iram0.bss' will not fit in region `iram0_0_seg'
/home/xu/esp/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/5.2.0/../../../../xtensa-lx106-elf/bin/ld: IRAM0 segment data does not fit.
collect2: error: ld returned 1 exit status
/home/xu/esp/ESP_Server//make/project.mk:509: recipe for target '/home/xu/esp/ESP_Server/examples/wifi/smart_config/build/smart_config.elf' failed
make: *** [/home/xu/esp/ESP_Server/examples/wifi/smart_config/build/smart_config.elf] Error 1`
具体应该怎么改啊？