espressif/ESP8266_RTOS_SDK #436: 部分路由器无法用wolfSSL OTA更新

wsgsra, 2019-01-02:
log
E (48016) ota: Running partition type 0 subtype 17 (offset 0x00110000)
E (48016) ota: Setting hostname for TLS session...

E (48516) ota: Init wolfSSL...

E (48516) ota: Set wolfSSL ctx ...

E (48516) ota: Creat socket ...

E (48516) ota: Connecting to host:443...

E (48706) ota: Create wolfSSL...

E (48706) ota: Performing the SSL/TLS handshake...

E (57016) ota: Writing HTTPS request...

E (57036) ota: Writing to partition subtype 16 at offset 0x10000
E (63736) ota: esp_ota_begin succeeded
E (63746) ota: Reading HTTPS response...

E (63756) ota: wolfSSL_read : 275 HTTP/1.1 200 OK

Date: Sun, 30 Dec 2018 13:39:40 GMT

Server: Apache/2.2.15 (CentOS)

Last-Modified: Sat, 29 Dec 2018 09:04:31 GMT

ETag: "480ec9-72f90-57e2577c0634d"

Accept-Ranges: bytes

Content-Length: 470928

Connection: close

Content-Type: application/octet-stream

E (63776) ota: esp_ota_write : 0
E (63776) ota: Total binary data length : 470928

能读取到http头 但是body的数据读取不到，不知道什么原因（手机热点可以正常更新）

27-301, 2019-01-02:
请提供测试路由型号，以及相关测试代码。

另外，请先测试一下仅下载，不做 flash 写入的测试。

wsgsra, 2019-01-02:
#define REQUEST "GET %s HTTP/1.0\r\n" \
    "Host: "WEB_SERVER"\r\n" \
    "User-Agent: esp-idf/1.0 espressif\r\n" \
    "\r\n"

static const char *TAG = "ota";
void ota_task(void *pvParameter)
{
    esp_err_t err;
    /* update handle : set by esp_ota_begin(), must be freed via esp_ota_end() */
    esp_ota_handle_t update_handle = 0 ;
    const esp_partition_t *update_partition = NULL;
	char *url = (char *)pvParameter;

    USER_LOGI(TAG, "Starting OTA url-->%s\n",url);
	vTaskDelay(2000 / portTICK_RATE_MS);
    const esp_partition_t *configured = esp_ota_get_boot_partition();
    const esp_partition_t *running = esp_ota_get_running_partition();

    if (configured != running) {
        ESP_LOGE(TAG, "Configured OTA boot partition at offset 0x%08x, but running from offset 0x%08x",
                 configured->address, running->address);
        ESP_LOGE(TAG, "(This can happen if either the OTA boot data or preferred boot image become corrupted somehow.)");
    }
    USER_LOGI(TAG, "Running partition type %d subtype %d (offset 0x%08x)",
             running->type, running->subtype, running->address);

    /*connect to http server*/
    int32_t ret = 0;
    WOLFSSL_CTX* ctx = NULL;
    WOLFSSL* ssl = NULL;

    int32_t socket = -1;
    struct sockaddr_in sock_addr;
    struct hostent* entry = NULL;
	  
    USER_LOGI(TAG,"Setting hostname for TLS session...\n");      
    do {
        entry = gethostbyname(WEB_SERVER);
        vTaskDelay(500 / portTICK_RATE_MS);
    }while (entry == NULL);

    USER_LOGI(TAG,"Init wolfSSL...\n");
    ret = wolfSSL_Init();

    if (ret != WOLFSSL_SUCCESS) {
        USER_LOGE(TAG,"Init wolfSSL failed:%d...\n", ret);
        goto failed1;
    }

    USER_LOGI(TAG,"Set wolfSSL ctx ...\n");
    ctx = wolfSSL_CTX_new(wolfTLSv1_2_client_method());

    if (!ctx) {
        USER_LOGE(TAG,"Set wolfSSL ctx failed...\n");
        goto failed1;
    }

    USER_LOGI(TAG,"Creat socket ...\n");
    socket = socket(AF_INET, SOCK_STREAM, 0);

    if (socket < 0) {
        USER_LOGE(TAG,"Creat socket failed...\n");
        goto failed2;
    }
    wolfSSL_CTX_set_verify(ctx, WOLFSSL_VERIFY_NONE, NULL);
	
    memset(&sock_addr, 0, sizeof(sock_addr));
    sock_addr.sin_family = AF_INET;
    sock_addr.sin_port = htons(WEB_PORT);
    sock_addr.sin_addr.s_addr = ((struct in_addr*)(entry->h_addr))->s_addr;

    USER_LOGI(TAG,"Connecting to %s:%d...\n", WEB_SERVER, WEB_PORT);
    ret = connect(socket, (struct sockaddr*)&sock_addr, sizeof(sock_addr));

    if (ret) {
        USER_LOGE(TAG,"Connecting to %s:%d failed: %d\n", WEB_SERVER, WEB_PORT, ret);
        goto failed3;
    }

    USER_LOGI(TAG,"Create wolfSSL...\n");
    ssl = wolfSSL_new(ctx);

    if (!ssl) {
        USER_LOGE(TAG,"Create wolfSSL failed...\n");
        goto failed3;
    }

    wolfSSL_set_fd(ssl, socket);

    USER_LOGI(TAG,"Performing the SSL/TLS handshake...\n");
    ret = wolfSSL_connect(ssl);

    if (WOLFSSL_SUCCESS != ret) {
        USER_LOGE(TAG,"Performing the SSL/TLS handshake failed:%d\n", ret);
        goto failed4;
    }

    /*send GET request to http server*/
	USER_LOGI(TAG,"Writing HTTPS request...\n");
	char *request_msg = (char *)malloc(1024);
	memset(request_msg,0,1024);	
	int len = sprintf(request_msg,REQUEST,url);
    ret = wolfSSL_write(ssl, request_msg, len);
	free(request_msg);
	
    if (ret <= 0) {
        USER_LOGE(TAG,"Writing HTTPS request failed:%d\n", ret);
        goto failed5;
    }
	
    update_partition = esp_ota_get_next_update_partition(NULL);
    USER_LOGI(TAG, "Writing to partition subtype %d at offset 0x%x",
             update_partition->subtype, update_partition->address);
    assert(update_partition != NULL);

    err = esp_ota_begin(update_partition, OTA_SIZE_UNKNOWN, &update_handle);
    if (err != ESP_OK) {
        USER_LOGE(TAG, "esp_ota_begin failed, error=%d", err);
        goto failed5;
    }
    USER_LOGE(TAG, "esp_ota_begin succeeded");
	
	USER_LOGI(TAG,"Reading HTTPS response...\n");
	
	char *rev_msg = (char *)malloc(1024);
	memset(request_msg,0,1024);	
	
	len = wolfSSL_read(ssl, rev_msg,1024);
	char *p = strstr(rev_msg,"Content-Length: ");
	if(p==NULL){
		free(rev_msg);
		goto failed5;
	}
	int binary_file_length = strtoul(p+16,NULL,10);
	if(binary_file_length<0)goto failed5;
	ret = len;
	USER_LOGI(TAG, "Total binary data length : %d", binary_file_length);
read_until_data:	
	p = strstr(rev_msg,"\r\n\r\n");
	if(p == NULL){
		len = wolfSSL_read(ssl, rev_msg + ret,1024-ret);
		ret = ret+len;
		if(ret >1024)goto failed5;
		goto read_until_data;
	}
	p = p+4;
	len = ret - (p - rev_msg);
	err = esp_ota_write(update_handle, (const void *)p, len);
	if (err != ESP_OK) {
        USER_LOGE(TAG, "Error: esp_ota_write failed! err=0x%x", err);
        goto failed5;
    }
	binary_file_length = binary_file_length -len;
	while(binary_file_length>0){
		memset(rev_msg, 0, 1024);
		len = wolfSSL_read(ssl, rev_msg,1024);
		if (len <= 0) {
            USER_LOGE(TAG,"\nConnection closed\n");
            break;
		}
		err = esp_ota_write( update_handle, (const void *)rev_msg,len);
        if (err != ESP_OK) {
            USER_LOGE(TAG, "Error: esp_ota_write failed! err=0x%x", err);
            goto failed5;
        }
		binary_file_length -= len;
        USER_LOGI(TAG, "image left length %d", binary_file_length);
	}   
    if (esp_ota_end(update_handle) != ESP_OK) {
        USER_LOGE(TAG, "esp_ota_end failed!");
        goto failed5;
    }
    err = esp_ota_set_boot_partition(update_partition);
    if (err != ESP_OK) {
        USER_LOGE(TAG, "esp_ota_set_boot_partition failed! err=0x%x", err);       
    }
    USER_LOGE(TAG, "Prepare to restart system!");	
failed5:
    wolfSSL_shutdown(ssl);
failed4:
    wolfSSL_free(ssl);
failed3:
    close(socket);
failed2:
    wolfSSL_CTX_free(ctx);
failed1:
    wolfSSL_Cleanup(); 		
	esp_restart();
}

测试的路由器是华为4G路由器2（B311As） 确认手机热点、小米路由器可以启动任务升级成功

wsgsra, 2019-01-02:
用这个路由器 http 也升级不了 主要是读取数据很慢（1分钟只读取到几KB） 然后超时 这个一般是什么原因引起的?

27-301, 2019-01-02:
你把做 esp_ota_begin 那部分的代码移到连接 server 之前去试试。
```c
update_partition = esp_ota_get_next_update_partition(NULL);
USER_LOGI(TAG, "Writing to partition subtype %d at offset 0x%x",
         update_partition->subtype, update_partition->address);
assert(update_partition != NULL);

err = esp_ota_begin(update_partition, OTA_SIZE_UNKNOWN, &update_handle);
if (err != ESP_OK) {
    USER_LOGE(TAG, "esp_ota_begin failed, error=%d", err);
    goto failed5;
}
USER_LOGE(TAG, "esp_ota_begin succeeded");
```

wsgsra, 2019-01-02:
@27-301 这个我也测试过 影响不大  我觉得原因是socket read出了问题 E (41173) ota: write_data length 749
E (41173) ota: Total binary data length : 470928
E (41173) ota: image left length 470179
E (41183) ota: image left length 469791
E (41843) ota: image left length 468767
E (41843) ota: image left length 468379
E (43123) ota: image left length 467355
E (43123) ota: image left length 466967
E (45423) ota: image left length 465943
E (45423) ota: image left length 465555
E (49773) ota: image left length 464531
E (49773) ota: image left length 464143
E (58483) ota: image left length 463119
E (58483) ota: image left length 462731
E (75403) ota: image left length 461707
E (75403) ota: image left length 461319
E (109243) ota: image left length 460295
E (109253) ota: image left length 459907

将近60s才读取到20KB的数据 这种情况一般是什么问题引起的

27-301, 2019-01-02:
所以建议做如下测试：
1. 不使用 wolfSSL 是否是一样的结果；
2. 只做下载，不真正操作 flash，看是否一样的结果；

wsgsra, 2019-01-02:
1、不使用wolfSSL，只打印读取的数据和
2、只做下载，不操作flash，只打印读取的数据和
上述都是一样的结果 无法完全下载文件，只能下载很少一部分数据

Junhao-Espressif, 2019-01-02:
Hi @wsgsra 

能否把路由器改为802.11 b/g模式试一下？

grtstar, 2019-01-02:
请问下为什么设置成 802.11 b/g 就可以, 我们也遇到了类似问题, 在 802.11 n 或者其他模式下,是否有解决方案

27-301, 2019-01-02:
[libnet80211.zip](https://github.com/espressif/ESP8266_RTOS_SDK/files/2720240/libnet80211.zip)

@wsgsra @grtstar 请基于 master，用附件的 libnet802.11 替代 components/esp8266/lib 下的 libnet80211.a 在 bgn 模式下做下测试。期待反馈，谢谢！

27-301, 2019-01-04:
@wsgsra @grtstar 请拉最新代码做下测试，看是否已经解决该问题。

FayeY, 2019-01-16:
由于长期无回复，此问题暂时关闭。如果问题仍然存在，欢迎重新开启。谢谢！

closed 2019-01-16 by FayeY