espressif/ESP8266_RTOS_SDK #440: Latest V3 release - many more asserts than previous

jhinklejr, 2019-01-05:
I just upgraded to the latest V3 version.

I have asserts enabled in my version of FreeRTOS and had a previous post about asserts in your blob code.

Previously, it only happened once/twice during initial boot but NOW -- they occur all the time.

Here is my debug statements that identify were your code is generating asserts.
                                            
RTOS Assert C:/msys32/home/Joe/esp/esp8266/components/freertos/freertos/queue.c 
 Line 736     Calling Task rtT
                                                              
RTOS Assert C:/msys32/home/Joe/esp/esp8266/components/freertos/freertos/queue.c 
 Line 736        Calling Task pmT

What ever code you added / changed to tasks rtT and pmT -- you are not being aware of proper RTOS coding requirements

Here is the code in queue.c that you are violating:

	#if ( ( INCLUDE_xTaskGetSchedulerState == 1 ) || ( configUSE_TIMERS == 1 ) )
	{
		configASSERT( !( ( xTaskGetSchedulerState() == taskSCHEDULER_SUSPENDED ) && ( xTicksToWait != 0 ) ) );	// hinkle - commented out in 10_2018 version
	}

I posted originally in Oct 2018 

https://github.com/espressif/ESP8266_RTOS_SDK/issues/346

You stated at that time - you were not going to investigate it because it was associated with legacy code.

This latest release shows it is not associated with legacy code but bad coding practices.

I found one reason for these asserts - as stated in issue 346 -- that they are generated when a callback function is called (say from an interrupt routine) and in that call a "take mutex" is called with a wait time.  You can't wait for a mutex in an interrupt.

Please review you newly added code and fix these asserts and implement better coding practices.  Just because asserts are turn off in you 8266 IDF version should not mean you don't adhere to proper RTOS coding practices.

I'm off my soapbox now.  Please resolve as my debug output is totally filled with these asserts from your new code.

Thanks in advance for fixing these.

Joe

donghengqaz, 2019-01-08:
You can try the library, and we will release it later after we test it well.

[libpp.zip](https://github.com/espressif/ESP8266_RTOS_SDK/files/2735028/libpp.zip)

.

jhinklejr, 2019-01-08:
Thanks - the new lib got rid of all the asserts - thanks for responding fast on that issue.

The only assert that stills occurs is shown below.  You can see it in relation to some of your blob messages.

scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 4
RTOS Assert C:/msys32/home/Joe/esp/esp8266/components/freertos/freertos/queue.c 
 Line 1401   Calling Task ppT

I show this in case you want to explore addressing this old one also.

Thanks again for the fast response.

Please close this issue.

closed 2019-01-08 by FayeY