espressif/ESP8266_RTOS_SDK #441: cycle failed

eqfox, 2019-01-06:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at bbs.espressif.com instead.

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs on the monitor](http://esp-idf.readthedocs.io/en/latest/get-started/idf-monitor.html#automatically-decoding-addresses).
4. Provide more items under **Other items if possible** can help us better locate your problem.
5. Use markup (buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:
- IDF version (``git rev-parse --short HEAD`` to get the commit id.): 
    //bd6ea4393c7d2f059fc4decc70f1ec3eb3597268
- Development Env:      [Arduino IDE|Make|Eclipse|other]
- Operating System:     [Windows|Ubuntu|MacOS]
- Power Supply:         [USB|external 5V|external 3.3V|Battery]

## Problem Description

//Detailed problem description goes here.

### Expected Behavior

### Actual Behavior

### Steps to repropduce

1. step1
2. ...

// It helps if you attach a picture of your setup/wiring here.

### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"

void app_main()
{
    
}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
Debug log goes here.
Please copy the plain text here for us to search the error log. Or attach the complete logs but leave the main part here if the log is *too* long.
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

aibeb, 2019-01-07:
The latest demo used，Only the following differences
```
CONFIG_MQTT_SUB_TOPIC="/temperature"
CONFIG_MQTT_PUB_TOPIC="/temperature"
```

ustccw, 2019-01-08:
@aibeb please make sure that your MQTT broker is still working.

aibeb, 2019-01-08:
@ustccw  
Cycle print the following logs, mqrr broker is working。
```
I (38331) example: MQTT subscribe to topic /temperature OK
I (38341) example: MQTT published topic /temperature, len:17 heap:59100
I (38351) example: Message arrived[len:17]: message number 15
W (38351) mc: mqtt close session
E (38361) mc: MQTTRun cycle failed
W (42351) mc: mqtt close session
E (42351) example: Return code from MQTT publish is -1
```

aibeb, 2019-01-08:
@ustccw 

You can run mqtt demo to do a test, only pub=sub, I think this problem must be

ustccw, 2019-01-19:
@aibeb 此现象确实存在, 由于 sub topic == pub topic.  
1. 可以将 Pub QoS 设置为 0 即可.  
[image]

2. 通常情况下, 不应该设置 sub topic == pub topic, 因为 MQTT 消息会发给自己, 没有意义.

cipibad, 2019-01-22:
Same issue for me also, pls advise

ustccw, 2019-01-23:
@cipibad This phenomenon does exist because sub topic = pub topic.

1. you could set pub QoS=0 to make it work.
2. Normally, you shouldn't set sub topic = pub topic, because the MQTT message will be sent to you, which is meaningless.

closed 2019-01-23 by aibeb

cipibad, 2019-01-23:
This means there is still bug in mqtt for QoS 1/QoS 2?

ustccw, 2019-01-24:
if you want to set QoS1/QoS2, you should set recv cycle and publish interval in `make menuconfig`.

ustccw, 2019-01-24:
**Normally, you shouldn't set sub topic = pub topic, because the MQTT message will be sent to you, which is meaningless.**

cipibad, 2019-01-24:
I am not using same pub/sub. 
I have an application using mqtt and that was working in earlier versions (https://github.com/espressif/ESP8266_RTOS_SDK/commit/01d32b73d4cc370795c6b23b56841ad1d3984963)
It is using pub/sub on mqtt via SSL.

Currently subscribe looks to work only without SSL:
with ssl:

> connected with SSID, channel 6
> I (12826) event: sta ip: 192.168.201.203, mask: 255.255.255.0, gw: 192.168.201.1
> 
> E (13796) example: Return code from network connect is 1
> I (13796) example: wait wifi connect...
> E (14396) example: Return code from network connect is 1
> I (14396) example: wait wifi connect...
> I (14426) example: MQTT Connecting
> I (14446) example: MQTT Connected
> I (14456) example: Use MQTTStartTask
> I (14466) example: MQTT subscribe to topic /espressif/sub OK
> I (14506) example: MQTT published topic /espressif/pub, len:16 heap:5100
> I (19556) example: MQTT published topic /espressif/pub, len:16 heap:5100
> I (24586) example: MQTT published topic /espressif/pub, len:16 heap:5100

without ssl:
I (9970) event: sta ip: 192.168.201.203, mask: 255.255.255.0, gw: 192.168.201.1
I (11550) example: MQTT Connecting
I (11560) example: MQTT Connected
I (11560) example: Use MQTTStartTask
I (11580) example: MQTT subscribe to topic /espressif/sub OK
I (11630) example: MQTT published topic /espressif/pub, len:16 heap:52588
....
I (96420) example: Message arrived[len:14]: a3169da3.2a93c

Thank you,
Ciprian

ustccw, 2019-01-25:
@cipibad MQTT based on SSL will be in the plan of `release v3.2`