espressif/ESP8266_RTOS_SDK #442: Kind request for a WPA2-Enterprise example

d-a-v, 2019-01-07:
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

- Development Kit: wemos/lolin d1 mini
- IDF version: bf27e59 (current master)
- Development Env:      Make
- Operating System:     Ubuntu
- Power Supply:         USB

## Problem Description

Would you please be able to provide a WPA2-Enterprise example like the one provided in nonos-sdk ?
https://github.com/espressif/ESP8266_NONOS_SDK/tree/master/examples/wpa2_enterprise

ustccw, 2019-01-07:
@d-a-v WPA2-Enterprise feature will be in our SDK in `v3.2`, please wait.

d-a-v, 2019-01-07:
In nonos-sdk, authentication phase2 TTLS-MSCHAPv2 is working, but not TTLS-PAP.
I hope you will include this one too.
Thanks

closed 2019-01-07 by d-a-v

ustccw, 2019-01-07:
maybe `EAP-TTLS` is working, more see: 
[Delivery_hostapd configuration for WPA2 Enterprise.zip](https://github.com/espressif/ESP8266_RTOS_SDK/files/2732687/Delivery_hostapd.configuration.for.WPA2.Enterprise.zip)

d-a-v, 2019-01-07:
Thanks,
In your example, simply adding a new user `userpap` like this may help testing.
```
"user"          MSCHAPV2                        "password"          [2]
"espressif"     MSCHAPV2                        "test11"            [2]
"userpap"       TTLS-PAP                        "password"          [2]
```
I will also try myself when I can build a setup.

That would be great for the wide spread *eduroam* network ([reaching Asia](https://www.eduroam.org/2018/05/02/eduroam-reaching-out-to-asia/), [China](http://eduroam.cstnet.cn/)).

More insight: https://wiki.freeradius.org/guide/eduroam