espressif/ESP8266_RTOS_SDK #443: Task stack [MQTTTask] stack from [3fff4600] to [3fff5500], total [3840] size

aibeb, 2019-01-07:
## Environment

- Development Kit: master mqtt demo
- IDF version (``git rev-parse --short HEAD`` to get the commit id.): 
    //bd6ea4393c7d2f059fc4decc70f1ec3eb3597268
- Development Env:     Make
- Operating System:     MacOS
- Power Supply:       external 5V

## Problem Description
Change the publish task delay time to 8 seconds
```
vTaskDelay(8000 / portTICK_RATE_MS);
```
### Expected Behavior

### Actual Behavior

### Steps to repropduce

1. step1
2. ...

// It helps if you attach a picture of your setup/wiring here.

### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"

void app_main()
{
    
}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
I (1135861) example: MQTT published topic /temperature, len:18 heap:60332
I (1143921) example: MQTT published topic /temperature, len:18 heap:60332
I (1152091) example: MQTT published topic /temperature, len:18 heap:60332
I (1160271) example: MQTT published topic /temperature, len:18 heap:60332
I (1168461) example: MQTT published topic /temperature, len:18 heap:60332
I (1176481) example: MQTT published topic /temperature, len:18 heap:60332
I (1184661) example: MQTT published topic /temperature, len:18 heap:60332

Task watchdog got triggered.

Task stack [MQTTTask] stack from [3fff4600] to [3fff5500], total [3840] size

          00000000 00000004 00000008 0000000c 00000010 00000014 00000018 0000001c 00000020 00000024 00000028 0000002c 00000030 00000034 00000038 0000003c

3fff4600  a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5
3fff4640  a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5
3fff4680  a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

ustccw, 2019-01-08:
@aibeb you could config MQTT recv interval by `make menuconfig` -> `Component config` -> `MQTT(Paho)`  ->  `MQTT recv cycle(ms) `

aibeb, 2019-01-08:
@ustccw 

What does this parameter mean?

ustccw, 2019-01-09:
@aibeb here is a help on your choice right.

aibeb, 2019-01-16:
> Any update about this issue? Is it solved now?

Increase the MQTT_CLIENT_THREAD_STACK_WORDS to reduce the problem!