espressif/ESP8266_RTOS_SDK #447: Can't receive data completely after connecting wifi with newest 3.0 SDK

kmalloc8, 2019-01-08:
I have write a NEC infrared driver in v3.0 SDK, it runs well before wifi connected.
But it often can not receive NEC data all completely, the NEC data 32 bits, for example, sometimes only have 16bits or 31bits or 20bit, etc.  
It is strange that it is OK after disconnected Wi-Fi, and, runs well in v2.0 SDK with wifi connected.
Any ideas ?

kmalloc8, 2019-01-09:
Any ideas?

xiongyumail, 2019-01-10:
Have you used HW_TIMER?

Junhao-Espressif, 2019-01-12:
Hi @kmalloc8，

The power save type default is WIFI_PS_MIN_MODEM, and wifi will not go to sleep if wifi is not connect to AP.

You should call esp_wifi_set_ps(WIFI_PS_NONE) to disable sleep mode.

kmalloc8, 2019-01-12:
Hi @Junhao-Espressif , 
All right. Thank you!!!

closed 2019-01-15 by kmalloc8