espressif/ESP8266_RTOS_SDK #450: Strange "Error 2: 800xxx" log, while ESP8266 is running

nghiaquy1991, 2019-01-10:
## Environment

- Development Kit: ESP8266 12F
- IDF version (``git rev-parse --short HEAD`` to get the commit id.): 
    7d4782fcd34fe55d6b403bdd33e5988741e11239
- Development Env:      [Make]
- Operating System:     [Ubuntu]
- Power Supply:         [USB]

## Problem Description

While my Application is running, I see some strange log in UART debug. It look like:
`åÂr2:800110
åBÿr2:800060
....
åÂÿr2:800120
åBÿr2:800013
åBÿr2:800145
åBÿr2:800059
åBr2:800138
åBÿr2:800019
åÂr2:800086
åÆÿr2:800087
åÆr2:800052
åBÿr2:800085
åBÿr2:800104
åÂÿr2:800040
åBÿr2:800061
åÂÿr2:800125
åBr2:800090
åÆr2:800021
åÆr2:800089
åBÿr2:800156
åÆr2:800040
åBr2:800038`

My UART baudrate is default 74880. I guess "åBÿr2" or "åÆr2" is Error.
But the device will run normally.
But I worry about the potential Error in future. So can you explain to me what type of this Error log?

xiongyumail, 2019-01-10:
At what stage does the scrambling occur? Is it in the startup stage?

nghiaquy1991, 2019-01-10:
It is not in startup state. It is in normal running state

27-301, 2019-01-10:
Can you also check your code? seems that it's not inside the SDK.

nghiaquy1991, 2019-01-11:
I checked my code many times, But I have print any log like that. 
Are you sure that it is not a part of SDK?

27-301, 2019-01-15:
Can you share your *.dump & *.S files?

nghiaquy1991, 2019-01-17:
@27-301 Ofcousre yes, Could you give me your email? or send an email to me at nghiaquy91@gmail.com 
Thanks.

nghiaquy1991, 2019-01-17:
I checked *.S and *.dump file.
I think that I have some clue.
[image]
I guess the log from here. In **g_phyFuns_instance** function.
[image]
Please help me to check it.

Thanks.

FayeY, 2019-07-15:
Hi, could you have a try with our latest RTOS SDK v3.2? 
If the issue persists on the v3.2, could you share a test project so we can reproduce the issue and help debug it?
Thanks.

nghiaquy1991, 2019-07-23:
Hi,
I stopped the project using esp so I can not help you debug this issue.
Thanks

closed 2019-07-23 by nghiaquy1991