espressif/ESP8266_RTOS_SDK #451: ppT task Problem

mnana3a, 2019-01-10:
I keep getting this error, tried to increase the ppT Task stack size but apparently the config value doesn't change anything in the code it stays to its default size. any help is appreciated

`Task stack [ppT] stack from [3fff4914] to [3fff5114], total [2048] size

          00000000 00000004 00000008 0000000c 00000010 00000014 00000018 0000001c 00000020 00000024 00000028 0000002c 00000030 00000034 00000038 0000003c 

3fff4914  a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 
3fff4954  a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 
3fff4994  a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 
3fff49d4  a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 
3fff4a14  a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 
3fff4a54  a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 
3fff4a94  a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 
3fff4ad4  a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 
3fff4b14  a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 
3fff4b54  a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 
3fff4b94  a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 
3fff4bd4  a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 
3fff4c14  a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 
3fff4c54  a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 
3fff4c94  a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 
3fff4cd4  a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 
3fff4d14  a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 
3fff4d54  a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 
3fff4d94  a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 
3fff4dd4  a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 
3fff4e14  a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 
3fff4e54  a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 
3fff4e94  a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 40100415 a5a5a5a5 a5a5a5a5 a5a5a5a5 40100415 3ffe8c0c 
3fff4ed4  00000172 a5a5a5a5 40253d42 3ffe8c0c 0000017f a5a5a5a5 3ffe8c0c 3ffe84d8 0000001c 00000004 3ffe8c0c 3ffe84d8 3fff2f8c 00000004 402427c6 00000000 
3fff4f14  a5a5a5a5 a5a5a5a5 4021d100 00000000 a5a5a5a5 a5a5a5a5 00000000 00000000 3fff4810 402428f8 3fff27b4 00000016 3fff5000 3ffedd30 4024cb4d 3fff2f8c 
3fff4f54  3fff7bc4 3fff26b8 3fff4fc0 00000001 3fff77d4 3fff5000 40247b26 3fff4fc0 00000010 00000001 3fff27b4 3fff5000 3fff293c 3ffedd30 4024a834 73726f68 
3fff4f94  61747365 00656c70 3fff4fc0 00000001 3fff7bc4 3fff2f8c 40248b98 3fff4fc0 a5a5a5a5 a5a5a5a5 40104260 3ffea4bc 3ffeb382 0000002e 00000000 3fff50c0 
3fff4fd4  00000001 00000000 402539b4 3fff28b8 3ffe9408 3ffe9408 401053d0 4010579c 000c3500 1d74fb88 4024b78b 401057c4 4010033e 00000030 4010038b 3fff5050 
3fff5014  00000020 00000000 40106ef8 3fff466c ffffffff 00000002 00632bfc 00232bfc 3fff2bfc 3fff2c10 3fff4810 00000001 3fff50c0 00000000 0000001e 3fff50c0 
3fff5054  00000001 ffffffff 4010038b 00000000 00000001 3fff4810 40251e6a 00000000 0000025d 3ffe9da0 401031dd ffffffff 3fff2d50 3fff2d64 3ffe9da0 00000002 
3fff5094  3ffed814 3ffed824 3ffe9da0 00000000 3ffe84d8 3ffe9f08 402428bb 00000000 00000000 3ffea4bc 402218e4 3ffe9c40 40221630 00000030 00000000 3ffeb2c0 
3fff50d4  00000000 00000000 3ffe9da0 00000004 ffffffff 00000010 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 

      PC: 0x40221770       PS: 0x00000030       A0: 0x402218e4       A1: 0x3fff50c0 
0x40221770: ppTask at ??:?

0x402218e4: ppTask at ??:?

      A2: 0xffffffe5       A3: 0x00000004       A4: 0x08000000       A5: 0x00000010 
      A6: 0x00000001       A7: 0x00000000       A8: 0x00000003       A9: 0x00232bfc 
     A10: 0x3fff2bfc      A11: 0x3fff2c10      A12: 0x00000000      A13: 0x00000000 
     A14: 0x3fff2bb8      A15: 0xffffffff      SAR: 0x0000001e EXCCAUSE: 0x00000009 
`

27-301, 2019-01-10:
how is it happen? can you upload your log file? and your test code?

mnana3a, 2019-01-10:
no idea why it happened but solved by doing a clean , erase flash then upload

closed 2019-01-10 by mnana3a