espressif/ESP8266_RTOS_SDK #455: mbedtls升级固件问题

MinjunHuang, 2019-01-14:
## 环境

- Development Kit:  ESP-LAUNCHER
- IDF version: SDK version:v3.1-dev-478-g91c7c10
- Development Env:      Make
- Operating System:     Ubuntu 14.04
- Power Supply:         USB

## 问题描述
1. 使用mbedtls升级固件失败，握手，发送https请求到服务器都是成功的，在接收服务器的返回数据时候，出现错误，打印的debug日志如下：
..........
[0;32mI (127563) OTA: Send GET request to server succeeded[0m
[0;32mI (127633) OTA: Reading HTTPS response...[0m
mbedtls: ssl_tls.c:6860 => read
mbedtls: ssl_tls.c:3785 => read record
mbedtls: ssl_tls.c:2256 => fetch input
mbedtls: ssl_tls.c:2414 in_left: 0, nb_want: 5
mbedtls: ssl_tls.c:2438 in_left: 0, nb_want: 5
mbedtls: ssl_tls.c:2439 ssl->f_recv(_timeout)() returned 5 (-0xfffffffb)
mbedtls: ssl_tls.c:2451 <= fetch input
mbedtls: ssl_tls.c:3535 dumping 'input record header' (5 bytes)
mbedtls: ssl_tls.c:3535 0000:  17 03 03 40 40                                   ...@@
mbedtls: ssl_tls.c:3544 input record: msgtype = 23, version = [3:3], msglen = 16448
mbedtls: ssl_tls.c:3582 bad message length
mbedtls: ssl_tls.c:3793 mbedtls_ssl_read_record_layer() returned -29184 (-0x7200)
mbedtls: ssl_tls.c:6959 mbedtls_ssl_read_record() returned -29184 (-0x7200)
[0;31mE (128483) esp-tls: read error :-29184:[0m
[0;32mI (128543) OTA: receved text: [0m
[0;31mE (128583) OTA: esp_tls_conn_read  returned -0x7200[0m
[0;31mE (128693) OTA: Exiting task due to fatal error...[0m

2. 经测试使用wolfssl升级固件是可以成功的，不知是否是mbedtls这里的问题？

donghengqaz, 2019-01-15:
```
menuconfig ->
     Component config  --->
         SSL ->
             mbedTLS ->
                 TLS maximum OUTPUT message content length (保持不变)
                 TLS maximum INPUT message content length (配置大一点)
```

试一下以上的配置。

MinjunHuang, 2019-01-15:
我将这个值修改为10240,结果还是一样，信息如下：
mbedtls: ssl_tls.c:2451 <= fetch input
mbedtls: ssl_tls.c:3535 dumping 'input record header' (5 bytes)
mbedtls: ssl_tls.c:3535 0000:  17 03 03 40 40                                   ...@@
mbedtls: ssl_tls.c:3544 input record: msgtype = 23, version = [3:3], msglen = 16448
mbedtls: ssl_tls.c:3582 bad message length
mbedtls: ssl_tls.c:3793 mbedtls_ssl_read_record_layer() returned -29184 (-0x7200)
mbedtls: ssl_tls.c:6959 mbedtls_ssl_read_record() returned -29184 (-0x7200)
E (92373) esp-tls: read error :-29184:
从这里看，可能是从服务器读取的的https返回的content_length太长导致的问题。

我看下了源代码，在ssl_tls.c的代码中，有如下检测： 
/* Check length against the size of our buffer */
    if( ssl->in_msglen > MBEDTLS_SSL_IN_BUFFER_LEN
                         - (size_t)( ssl->in_msg - ssl->in_buf ) )
    {
        MBEDTLS_SSL_DEBUG_MSG( 1, ( "bad message length" ) );
        return( MBEDTLS_ERR_SSL_INVALID_RECORD );
    }
而定义了：MBEDTLS_SSL_MAX_CONTENT_LEN         16384，相应的MBEDTLS_SSL_IN_BUFFER_LEN的值就限定了。

我服务器上的固件大小是497344，但我不知道和msglen = 16448的关系是什么？

MinjunHuang, 2019-01-15:
我使用的是esp_tls_conn_new，esp_tls_conn_write，esp_tls_conn_read系列函数。
我测试了下载固件和通过https上报数据两种情况，上报数据是可以的，但下载固件，即使是50K的文件，都是同样的问题。

Junhao-Espressif, 2019-01-15:
Hi @MinjunHuang ,

Please set  "TLS maximum INPUT message content length"  to 16384 and try again.

Thank you!

closed 2019-01-15 by Junhao-Espressif

lindabang, 2019-07-10:
我也遇到了相同的问题，然后将TLS maximum INPUT message content length 设置为16384，但是数据只接收了 16133字节后就出错了：
D (17675) /home/bond/home/esp8266/IL3/main/ota.c: HTTP_EVENT_ON_DATA, len=512
D (17685) HTTP_CLIENT: is_data_remain=1, is_chunked=0
D (17685) HTTP_CLIENT: need_read=1536, byte_to_read=512, rlen=512, ridx=512
D (17685) HTTP_CLIENT: http_on_body 512
D (17695) /home/bond/home/esp8266/IL3/main/ota.c: HTTP_EVENT_ON_DATA, len=512
D (17695) HTTP_CLIENT: is_data_remain=1, is_chunked=0
D (17705) HTTP_CLIENT: need_read=1024, byte_to_read=512, rlen=512, ridx=1024
D (17705) HTTP_CLIENT: http_on_body 512
D (17705) /home/bond/home/esp8266/IL3/main/ota.c: HTTP_EVENT_ON_DATA, len=512
D (17715) HTTP_CLIENT: is_data_remain=1, is_chunked=0
D (17715) HTTP_CLIENT: need_read=512, byte_to_read=512, rlen=261, ridx=1536
D (17715) HTTP_CLIENT: http_on_body 261
D (17725) /home/bond/home/esp8266/IL3/main/ota.c: HTTP_EVENT_ON_DATA, len=261
D (17725) HTTP_CLIENT: is_data_remain=1, is_chunked=0
E (24705) esp-tls: read error :-76:
E (24705) TRANS_SSL: esp_tls_conn_read error, errno=No more processes
D (24705) HTTP_CLIENT: need_read=251, byte_to_read=251, rlen=-1, ridx=1797
D (24715) esp_https_ota: Written image length 16133
D (24725) esp_https_ota: 99: free=3504
D (24725) HTTP_CLIENT: is_data_remain=1, is_chunked=0
D (29725) HTTP_CLIENT: need_read=2048, byte_to_read=512, rlen=0, ridx=0
I (29725) esp_https_ota: Connection closed,all data received
D (29725) /home/bond/home/esp8266/IL3/main/ota.c: HTTP_EVENT_DISCONNECTED
D (29745) /home/bond/home/esp8266/IL3/main/ota.c: HTTP_EVENT_DISCONNECTED
D (29755) esp_https_ota: Total binary data length writen: 16133
D (29755) esp_image: reading image header @ 0x110000
D (29755) esp_image: image header: 0xe9 0x04 0x03 0x02 4021f998
I (29765) esp_image: segment 0: paddr=0x00110010 vaddr=0x40210010 size=0x726c8 (468680) map
E (29925) esp_image: unaligned segment length 0xffffffff
E (29925) esp_https_ota: Error: esp_ota_end failed! err=0x5379. Image is invalid