espressif/ESP8266_RTOS_SDK #456: setsockopt() SO_RCVTIMEO timeval BUG？

ichenxing, 2019-01-15:
## Environment

- Development Kit: ESP8266_RTOS_SDK-2.1.0 Release
- IDF version (``git rev-parse --short HEAD`` to get the commit id.): 
    //bd6ea4393c7d2f059fc4decc70f1ec3eb3597268
- Development Env:     [Make]
- Operating System:     [Ubuntu]
- Power Supply:            [Battery]

## Problem Description
设置的超时响应时间与预期的不符

### Expected Behavior

### Actual Behavior
实际的情况超时响应时间是随机的，有时会超过5s以上
### Steps to repropduce

### Code to reproduce this issue

```
    struct timeval receiving_timeout;
    receiving_timeout.tv_sec = 1800;	// 1.8s?//设置太小时接收会立即返回？
    receiving_timeout.tv_usec = 0;
    if (setsockopt(sock_fd, SOL_SOCKET, SO_RCVTIMEO, &receiving_timeout,
            sizeof(receiving_timeout)) < 0){
        esp_dbg("failed to set socket receiving timeout.\r\n");
		goto jump_func_end;
    }
    esp_dbg("set socket receiving timeout success.\r\n");
```

## Other items if possible

1XiaSheng, 2019-01-16:
这个我也碰到了

FayeY, 2019-02-21:
请问，以上建议是否解决了此问题？

ichenxing, 2019-02-21:
SDK v2.x.x以下版本的存在这个问题@FayeY