espressif/ESP8266_RTOS_SDK #460: esp8266  aws iot  demo  无法正常运行

myzhang001, 2019-01-16:
ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 7224, room 16 
tail 8
chksum 0x9f
load 0x3ffe8408, len 24, room 0 
tail 8
chksum 0x76
load 0x3ffe8420, len 3552, room 0 
tail 0
chksum 0x34
csum 0x34
[0;32mI (60) boot: ESP-IDF v3.1-dev-489-g74b4a55-dirty 2nd stage bootloader[0m
[0;32mI (60) boot: compile time 10:44:53[0m
[0;32mI (69) qio_mode: Enabling default flash chip QIO[0m
[0;32mI (69) boot: SPI Speed      : 40MHz[0m
[0;32mI (76) boot: SPI Mode       : QIO[0m
[0;32mI (82) boot: SPI Flash Size : 4MB[0m
[0;31mE (88) flash_parts: partition 0 invalid magic number 0x3333[0m
[0;31mE (97) boot: Failed to verify partition table[0m
[0;31mE (105) boot: load partition table error![0m
user code done

27-301, 2019-01-16:
1. 如何烧写各个 bin 的？
2. 在其他模组上是否正常？

myzhang001, 2019-01-16:
我是直接使用make  flash   烧录的

27-301, 2019-01-16:
换块板子可以吗？
把 0x8000 地址的 4K dump 出来看下。

myzhang001, 2019-01-16:
是不是 aws  iot  demo  交叉编译版本 一定要用5.2.0 呀

27-301, 2019-01-16:
这个和 aws iot demo 没有关系，是在 bootloader 阶段 parse partition 分区时有问题，你编译其他 demo 可能也有问题。
```
flash_parts: partition 0 invalid magic number 0x3333
```
这里有问题，partition 分区读出来的 magic number 为 0x3333。
请帮忙做一下测试：
1. 试下换个板子;
2. 试下换个编译器
3. 把 0x8000 地址的 4K dump 出来看下

myzhang001, 2019-01-16:
我编译其他的  demo  是可以正常跑的 ，我先做下测试

27-301, 2019-01-16:
你改成 2MB 的 Flash，然后烧进去跑跑看。
直接把 2MB 都 dump 出来，发上来，我在我这边 2MB 的板子上烧下去跑跑看
python esptool.py --port /dev/tty.SLAB_USBtoUART read_flash 0 2097152 firmware.bin

esptool.py 在 ESP8266_RTOS_SDK/components/esptool_py/esptool/ 下有，串口改成你自己的。

另外，把 sdkconfig 文件也发上来吧。

myzhang001, 2019-01-16:
解决了 我 改成 DOUT      SPI speed  80M  就可以了

现在新的问题是：
[0;31mE (11756) aws_iot: failed! mbedtls_ssl_handshake returned -0x7200[0m
[0;31mE (11786) subpub: Error(-4) connecting to apg8net6ftn73-ats.iot.us-west-2.amazonaws.com:8883[0m

这是什么问题呢

27-301, 2019-01-16:
看来 nodeMCU 里面用的是特殊的 SPI Flash。

close 该 issue。新的问题，请提新的 issue。

closed 2019-01-16 by 27-301