espressif/ESP8266_RTOS_SDK #462: ESP8266_RTOS_SDK HSPI "Core 0 was running in ISR context:"

ichenxing, 2019-01-16:
## Environment

- Development Kit:
- IDF version (``git rev-parse --short HEAD`` to get the commit id.): 
    bcc4445
- Development Env:      [Make]
- Operating System:     [Ubuntu]
- Power Supply:         [Battery]

## Problem Description

esp8266作为spi slave，连接到路由之后，通过io通知master读取状态寄存器，8266出现崩溃

### Expected Behavior

### Actual Behavior

### Steps to repropduce

### Code to reproduce this issue

```
static const char *TAG = "user_spi";

#define SPI_SLAVE_HANDSHARK_GPIO     5
#define SPI_SLAVE_HANDSHARK_SEL      (1ULL<<SPI_SLAVE_HANDSHARK_GPIO)

static SemaphoreHandle_t spi_wr_sta_done_sem = NULL;
static SemaphoreHandle_t spi_rd_buf_done_sem = NULL;
static SemaphoreHandle_t spi_wr_buf_done_sem = NULL;

void IRAM_ATTR spi_event_callback(int event, void *arg)
{
    BaseType_t xHigherPriorityTaskWoken;
    uint32_t status;
    uint32_t trans_done;

    switch (event) {
        case SPI_INIT_EVENT: {
            status = false;
            spi_slave_set_status(HSPI_HOST, &status);
        }
        break;

        case SPI_TRANS_START_EVENT: {

        }
        break;

        case SPI_TRANS_DONE_EVENT: {
            trans_done = *(uint32_t *)arg;

            if (trans_done & SPI_SLV_RD_BUF_DONE) {
                xSemaphoreGiveFromISR(spi_rd_buf_done_sem, &xHigherPriorityTaskWoken);
            }
            if (trans_done & SPI_SLV_WR_BUF_DONE) {
                status = false;
                spi_slave_set_status(HSPI_HOST, &status);
                xSemaphoreGiveFromISR(spi_wr_buf_done_sem, &xHigherPriorityTaskWoken);
            }
            if (trans_done & SPI_SLV_RD_STA_DONE) {
                gpio_set_level(SPI_SLAVE_HANDSHARK_GPIO, 0);
                status = false;
                spi_slave_set_status(HSPI_HOST, &status);
                printf("SPI_SLV_RD_STA_DONE\n");
            }
            if (trans_done & SPI_SLV_WR_STA_DONE) {
                gpio_set_level(SPI_SLAVE_HANDSHARK_GPIO, 0);
                spi_slave_get_status(HSPI_HOST, &status);
                if (status == true) {
[... 171 lines cut ...]
inform spi master
I (4475) main task: 3: hello world, stack: 1316.
I (6475) main task: 4: hello world, stack: 1316.

Core 0 was running in ISR context:

          00000000 00000004 00000008 0000000c 00000010 00000014 00000018 0000001c 00000020 00000024 00000028 0000002c 00000030 00000034 00000038 0000003c

3ffe82e0  00000000 00000000 00000000 00000000 00000000 00000000 00000000 ffffffff 40224e94 40224e7c 00000001 40247bf8 00000000 00000000 00000000 4023c294
3ffe8320  00000000 00000000 00000000 00000000 00000000 00000000 00000000 0000ff00 00ff0000 ff000000 00000000 00000000 00000004 3ffedf50 3fff19bc 40247894
3ffe8360  00000000 00000000 00000000 40247aca 00000000 00000000 00000000 00000000 00000000 00000000 00000000 ffffffff 40224e94 40224e7c 00000001 40247bf8
3ffe83a0  00000000 00000000 00000000 4023c294 00000000 00000000 00000000 00000000 00000000 00000000 00000000 0000ff00 00ff0000 ff000000 00000000 00000000
3ffe83e0  00000004 3ffef83c 3fff1abc 40247894 00000000 00000000 00000000 40247aca 00000000 00000000 00000000 4023bfc6 40246e39 00000000 00000000 00000001
3ffe8420  00000004 3ffee7ac 3ffef7e4 40247bf8 00000004 00000000 3ffef8b4 40246849 40225d58 00000013 4022256e 00000001 3ffe8440 00000002 00000014 4023a713
3ffe8460  3ffe8498 00000004 40225d58 402468b0 3ffe9532 00000004 00000014 40100559 00000400 00000000 3ffe84ec 40102cb9 00000005 00000000 3ffef144 40100612
3ffe84a0  00000014 00000001 08000000 4023bbfc 4023bd1e 40229fa8 3ffef184 4023bd31 3fff07a0 00000001 402222ac 00000000 00000000 00000000 40105304 4010534d
      PC: 0x4023c282       PS: 0x00000031       A0: 0x3ff7ffe0       A1: 0x3ff7ffe0
      A2: 0x00000001       A3: 0x40224e94       A4: 0x40224e7c       A5: 0x0000ff00
      A6: 0x00ff0000       A7: 0xff000000       A8: 0x00454e4f       A9: 0x22800000
     A10: 0x20000000      A11: 0xcfffffff      A12: 0x3fff19bc      A13: 0x3ffedf50
     A14: 0x00000004      A15: 0xffffffff      SAR: 0x0000001b EXCCAUSE: 0x00000003

```

## Other items if possible

xiongyumail, 2019-01-16:
事件回调函数会在spi中断中调用，请不要使用printf,或许你可以使用ets_printf
需要包含#include"esp_libc.h"

ichenxing, 2019-01-16:
@xiongyumail 的确中断中不能使用printf；不过spi传输数据不对，第一帧数据出现问题,，如下：
```
0x00020000 0x00060004 0x000A0008 0x000E000C 0x00120010 0x00160014 0x001A0018 0x001E001C 
0x11001000 0x13001200 0x15001400 0x17001600 0x19001800 0x1B001A00 0x1D001C00 0x1F001E00 
0x21002000 0x23002200 0x25002400 0x27002600 0x29002800 0x2B002A00 0x2D002C00 0x2F002E00
```

spi  master code:
```
	u16 i=0,temp[16];

	for(i=0;i<16;i++)
	{
		temp[i] = i;
	}
	HSPI_Write_Pages(temp, 16);

	for(i=16;i<32;i++)
	{
		temp[i-16] = i;
	}
	HSPI_Write_Pages(temp, 16);

	for(i=32;i<48;i++)
	{
		temp[i-32] = i;
	}
	HSPI_Write_Pages(temp, 16);
```
sdk给的例程中的通信也有点看不懂，如下：
[image]

xiongyumail, 2019-01-16:
[image]

读写8266 slave 需要有cmd, addr
例程中通过status和io4进行流控

ichenxing, 2019-01-16:
@xiongyumail 例程中主机读完状态寄存器后一共发送了1+4+32=37个字节，波形图中一共出现了41个字节，且write_data[0]的值并不等于0；我用的是教程后面的spi单线透传协议。

xiongyumail, 2019-01-16:
嗯嗯，发现了，请以例程为准，波形图将会更新。谢谢!

ichenxing, 2019-01-16:
@xiongyumail **波形图只是硬件在执行程序时的实际体现**；

主机实际发送的数据应该是：
```
0x01000000 0x03000200 0x05000400 0x07000600 0x09000800 0x0B000A00 0x0D000C00 0x0F000E00
0x11001000 0x13001200 0x15001400 0x17001600 0x19001800 0x1B001A00 0x1D001C00 0x1F001E00 
0x21002000 0x23002200 0x25002400 0x27002600 0x29002800 0x2B002A00 0x2D002C00 0x2F002E00
```
esp8266接收的数据确是如下，且**接收到的第一帧数据比较有意思，都是偶数**：
```
0x00020000 0x00060004 0x000A0008 0x000E000C 0x00120010 0x00160014 0x001A0018 0x001E001C 
0x11001000 0x13001200 0x15001400 0x17001600 0x19001800 0x1B001A00 0x1D001C00 0x1F001E00 
0x21002000 0x23002200 0x25002400 0x27002600 0x29002800 0x2B002A00 0x2D002C00 0x2F002E00
```

程序参考：
[image]

xiongyumail, 2019-01-16:
看上去是右移了7bit, example第一帧也有这种现象嘛？

ichenxing, 2019-01-17:
@xiongyumail 例程没有测试，没有两块esp8266进行spi通信，不过有点看不懂例程（主机读取从机数据）的波形图如下，**为什么会在03后出现8个字节 00 00 00 56 00 00 00 AB，且这些字节所对应的值都为FF**？
[image]

我自己的程序主机读取esp8266状态寄存器时，读出来的值一直为0xFF？

xiongyumail, 2019-01-23:
example中的波形图有问题，请以程序为准。
esp8266 spi只能半双工通讯，因此在发送或者接收时，有一方全为0xFF或0x00

ichenxing, 2019-01-23:
@xiongyumail手上缺少逻辑分析仪，没办法从波形图上发现问题，

ichenxing, 2019-01-29:
@xiongyumail 下面是主机发送从机(ESP8266)接收的波形图，一共发送了4帧数据，第一帧和第二帧发送的数据是一样的，但是8266接收到的数据确是不一样的：
esp8266接收到的数据：
[image]

采集到的波形图：
[image]
编号1：
[image]
编号2：
[image]
编号3：
[image]
编号4：
[image]

ichenxing, 2019-01-29:
@xiongyumail 下面是主机读从机(ESP8266)状态寄存器的波形图，读出来的值都是0xFF，如下：
 波形图：
[image]

esp8266打印的结果显示进入了2次读状态寄存器，实际程序只读了一次，波形图上也只反应进入一次读状态寄存器：
[image]

代码如下：
```
static void spi_slave_read_master_task(void *arg)
{
    int x;
    uint16_t cmd;
    uint32_t addr;
    uint32_t read_data[8];
    spi_trans_t trans;
    uint32_t status = true;

    trans.cmd = &cmd;
    trans.addr = &addr;
    trans.mosi = read_data;
    trans.bits.val = 0;
    // In Slave mode, spi cmd must be longer than 3 bits and shorter than 16 bits
    trans.bits.cmd = 8 * 1;
    // In Slave mode, spi addr must be longer than 1 bits and shorter than 32 bits
    trans.bits.addr = 8 * 1;
    trans.bits.mosi = 32 * 8;

    xEventGroupWaitBits(wifi_event_group, connected_rt_bit, false, true, portMAX_DELAY);
    ESP_LOGI(TAG, "Connected to AP");

    status = STA_REG_ROUNTER_CONNECT << 24 | STA_REG_ROUNTER_CONNECT << 16 | STA_REG_ROUNTER_CONNECT << 8 | STA_REG_ROUNTER_CONNECT;
    spi_slave_set_status(HSPI_HOST, &status);
    gpio_set_level(SPI_SLAVE_HANDSHARK_GPIO, 1);
    printf("inform spi master: 0x%08X\n", status);

    while (1) {
        xSemaphoreTake(spi_wr_buf_done_sem, portMAX_DELAY);

        spi_trans(HSPI_HOST, trans);

        if (cmd == SPI_MASTER_WRITE_DATA_TO_SLAVE_CMD) {

            for (x = 0; x < 8; x++) {
                printf("0x%08X ", read_data[x]);
            }
            printf("\r\n");
        }

        gpio_set_level(SPI_SLAVE_HANDSHARK_GPIO, 0);
    }
}
```

ichenxing, 2019-01-29:
@xiongyumail 现在除了主机读从机(ESP8266)状态寄存器有问题，其他三种情况都可以：
[image]

主机读8266状态寄存器的波形图：
[image]

主机读8266数据的波形图：
[image]

closed 2019-01-29 by wujiangang

xiongyumail, 2019-01-29:
slave有没有调用`spi_slave_set_status(HSPI_HOST, &status);`这个接口装载status寄存器?

ichenxing, 2019-01-29:
@xiongyumail 从机里面有设置设个状态寄存器的值的
```
static void spi_slave_read_master_task(void *arg)
{
    int x;
    uint16_t cmd;
    uint32_t addr;
    uint32_t read_data[8];
    spi_trans_t trans;
    uint32_t status = true;

    trans.cmd = &cmd;
    trans.addr = &addr;
    trans.mosi = read_data;
    trans.bits.val = 0;
    // In Slave mode, spi cmd must be longer than 3 bits and shorter than 16 bits
    trans.bits.cmd = 8 * 1;
    // In Slave mode, spi addr must be longer than 1 bits and shorter than 32 bits
    trans.bits.addr = 8 * 1;
    trans.bits.mosi = 32 * 8;

    xEventGroupWaitBits(wifi_event_group, connected_rt_bit, false, true, portMAX_DELAY);
    ESP_LOGI(TAG, "Connected to AP");

    **status = STA_REG_ROUNTER_CONNECT << 24 | STA_REG_ROUNTER_CONNECT << 16 | STA_REG_ROUNTER_CONNECT << 8 | STA_REG_ROUNTER_CONNECT;
    spi_slave_set_status(HSPI_HOST, &status);**
    gpio_set_level(SPI_SLAVE_HANDSHARK_GPIO, 1);
    printf("inform spi master: 0x%08X\n", status);

    while (1) {
        xSemaphoreTake(spi_wr_buf_done_sem, portMAX_DELAY);

        spi_trans(HSPI_HOST, trans);

        if (cmd == SPI_MASTER_WRITE_DATA_TO_SLAVE_CMD) {

            for (x = 0; x < 8; x++) {
                printf("0x%08X ", read_data[x]);
            }
            printf("\r\n");
        }

        gpio_set_level(SPI_SLAVE_HANDSHARK_GPIO, 0);
    }
}
```

xiongyumail, 2019-01-29:
在`esp_err_t spi_set_mode(spi_host_t host, spi_mode_t *mode)`函数中，
`SPI[host]->slave1.status_bitlen = 31;`status为(31+1)bits， 
从你的波形图看只有8Bits. 你可以修改slave1.status_bitlen为7, 或者修改master读status程序。

ichenxing, 2019-01-30:
@xiongyumail 问题已解决，程序的问题

[image]

xiongyumail, 2019-01-30:
@ichenxing 建议使用32bits status, 可以发送更多信息。8bits status或32bits status 几乎消耗的是同样的时间。所以spi固定使用32bits

ichenxing, 2019-01-30:
@xiongyumail 是否可以改为成可配置的？