espressif/ESP8266_RTOS_SDK #465: 烧录成功，但是程序运行不起来

YeFly9, 2019-01-17:
环境：RTOS
DEMO：ESP8266_RTOS_SDK
操作：使用make flash烧录
串口打印数据：
ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 7196, room 16 
tail 12
chksum 0x59
ho 0 tail 12 room 4
load 0x3ffe8408, len 24, room 12 
tail 12
chksum 0xd0
ho 0 tail 12 room 4
load 0x3ffe8420, len 3536, room 12 
tail 4
chksum 0xf9
csum 0xf9
[0;32mI (49) boot: ESP-IDF v3.1-rc1 2nd stage bootloader[0m
[0;32mI (49) boot: compile time 13:47:15[0m
[0;32mI (49) qio_mode: Enabling default flash chip QIO[0m
[0;32mI (56) boot: SPI Speed      : 40MHz[0m
[0;32mI (62) boot: SPI Mode       : QIO[0m
[0;32mI (68) boot: SPI Flash Size : 2MB[0m
[0;31mE (74) flash_parts: partition 0 invalid magic number 0x0[0m
[0;31mE (84) boot: Failed to verify partition table[0m
[0;31mE (91) boot: load partition table error![0m
user code done

ustccw, 2019-01-21:
@YeFly9 和 #460 类似, 可否看一下

YeFly9, 2019-01-21:
你是情况和我一样吗？

ustccw, 2019-01-21:
都是 `partition 0 invalid magic number` 错误

YeFly9, 2019-01-21:
嗯嗯，我已经解决了