espressif/ESP8266_RTOS_SDK #469: problems with esp-01

mpere654, 2019-01-20:
Hi, everybody.
I'm introducing in the world of esp8266-rtos-sdk  and developing apps compatible with iot and aws iot with alexa. 
Days ago I received an esp-01, second version with 1 MB of memory (black board). 
My problem is ... I can't boot a simple hello world app. I can flash but when it runs, in the monitor doesn't show anything. I have another app with shadow with aws and it works with nodemcu ... but with esp-01 doesn't boot.
I'm using ubuntu linux and the message after flashing is this:
**************************************************
esptool.py v2.4.0
Flashing binaries to serial port /dev/ttyUSB0 (app at offset 0x10000)...
esptool.py v2.4.0
Connecting....
Chip is ESP8266EX
Features: WiFi
MAC: 80:7d:3a:75:cb:46
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Compressed 10800 bytes to 7163...
Wrote 10800 bytes (7163 compressed) at 0x00000000 in 0.6 seconds (effective 136.2 kbit/s)...
Hash of data verified.
Compressed 220144 bytes to 151837...
Wrote 220144 bytes (151837 compressed) at 0x00010000 in 17.0 seconds (effective 103.3 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 83...
Wrote 3072 bytes (83 compressed) at 0x00008000 in 0.0 seconds (effective 1886.6 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
MONITOR
--- idf_monitor on /dev/ttyUSB0 74880 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
**************************************************
Has somebody flash and run an app with esp-01? Can provide the configuration used for flashing. these are the steps I did:
make 
make flash monitor
Also I tried to change de bauds from monitor in sdkconfig and it doesn't work, the QIO flash ... etc.
ESP-01 works with arduino IDE so I don't know what happen on esp8266-rtos-sdk and what am I doing wrong.
Thanks in advance

mpere654, 2019-01-23:
I got it, it was a rookie's error ... I had a cable with the dupont connector that it didn't make contact (it was broken) . Now it runs and I can watch de logs and connect to AWS. A piece of advice for rookies, check connection cables before ...  If somebody needs help to connect to AWS, I can help. Thanks for your atention.

closed 2019-01-23 by mpere654

Kalyan96, 2020-09-28:
> I got it, it was a rookie's error ... I had a cable with the dupont connector that it didn't make contact (it was broken) . Now it runs and I can watch de logs and connect to AWS. A piece of advice [...]

Hi mpere654, would like to connect with you regarding the steps followed in flashing the esp01 module with esp8266_rtos_sdk

Yahuma, 2022-12-15:
I'm posting it for those interested. Maybe it will be useful for someone.
My hardware:
- ESP-01s
- 1MB Flash
- Quartz: 26Mhz

I am using Windows10. After installing MSYS2 and toolchain for the ESP8266 (look here: https://docs.espressif.com/projects/esp8266-rtos-sdk/en/latest/get-started/index.html#introduction) it was important to set the appropriate configurations for ESP-01s in "make menuconfig" for "hello_world" example. Changed settings:
serial flasher config:
    - Default serial port (changed to "COM7" for me)
    - Flash SPI Mode: "DOUT".
    - Flash SPI Speed: 26Mhz
    - Flash Size: 1MB

After compiling with the "make" command, this was the log after flashing with "make flash":
```
esptool.py v2.4.0
Connecting.......______...______.....__
Chip is ESP8266EX
Features: WiFi
MAC: xx:xx:xx:xx:xx:xx
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Compressed 11072 bytes to 7426...
Wrote 11072 bytes (7426 compressed) at 0x00000000 in 0.7 seconds (effective 134. 5 kbit/s)...
Hash of data verified.
Compressed 169488 bytes to 108620...
Wrote 169488 bytes (108620 compressed) at 0x00010000 in 9.6 seconds (effective 1 41.4 kbit/s)...
Hash of data verified.
Compressed 3072 bytes is 83...
Wrote 3072 bytes (83 compressed) at 0x00008000 in 0.0 seconds (effective 1616.4 kbit/s)...
Hash of data verified.

leaving...
Hard reset via RTS pin...
```

After turning off the ESP-01s and opening the programming jumper, I turned on Putty on the COM7 port, speed: 74480. After turning on the power for ESP-01s, this was the log:

```
  ets Jan 8 2013,rst cause:1, boot mode:(3,7)

load 0x40100000, len 7536, room 16
tail 0
chsum 0x53
load 0x3ffe8408, len 24, room 8
tail 0
chsum 0xea
load 0x3ffe8420, len 3468, room 8
tail 4
chsum 0xcf
csum 0xcf
I (44) boot: ESP-IDF v3.4-73-gd48c4c17 2nd stage bootloader
I (44) boot: compile time 10:14:40
I (44) boot: SPI Speed: 26.7MHz
I (51) boot: SPI Mode: DOUT
I (57) boot: SPI Flash Size : 1MB
I (63) boot: Partition Table:
I (69) boot: ## Label Usage Type ST Offset Length
I (80) boot: 0 nvs WiFi date 01 02 00009000 00006000
I (91) boot: 1 phy_init RF date 01 01 0000f000 00001000
I (103) boot: 2 factory factory app 00 00 00010000 000f0000
I (114) boot: End of partition table
I (121) esp_image: segment 0: paddr=0x00010010 vaddr=0x40210010 size=0x1cca8 (117928) map
I (198) esp_image: segment 1: paddr=0x0002ccc0 vaddr=0x4022ccb8 size=0x07288 ( 29320) map
I (215) esp_image: segment 2: paddr=0x00033f50 vaddr=0x3ffe8000 size=0x00544 ( 1348) load
I (216) esp_image: segment 3: paddr=0x0003449c vaddr=0x40100000 size=0x00080 ( 128) load
I (226) esp_image: segment 4: paddr=0x00034524 vaddr=0x40100080 size=0x050c0 ( 20672) load
I (250) boot: Loaded app from partition at offset 0x10000
hello world!
This is ESP8266 chip with 1 CPU cores, WiFi, silicon revision 1, 1MB external flash
Restarting in 10 seconds...
Restarting in 9 seconds...
Restarting in 8 seconds...
Restarting in 7 seconds...
Restarting in 6 seconds...
Restarting in 5 seconds...
Restarting in 4 seconds...
Restarting in 3 seconds...
Restarting in 2 seconds...
Restarting in 1 seconds...
Restarting in 0 seconds...
```