espressif/ESP8266_RTOS_SDK #473: aws  证书问题

myzhang001, 2019-01-24:
在连接aws iot  经常出现 下面这个错误

E (33947) aws_iot: failed! mbedtls_ssl_handshake returned -0x4310 E (33977) shadow: aws_iot_shadow_connect returned error -4, aborting...

donghengqaz, 2019-01-25:
这个应该是没内存了。

myzhang001, 2019-01-28:
是ram不够了是吧，我怎么查看我使用的 ram 的大小呢

FayeY, 2019-02-18:
请问，这个问题解决了吗？

myzhang001, 2019-02-18:
没有解决，而且还发现新的 问题 就是  在aws  iot  demo  中加入了自己的 用户代码  就会出现证书校验失败的问题， aws iot  thing_shadow 这个例程 ，只能创建三个 task 超过 三个task 就无法正常运行

FayeY, 2019-02-21:
查询的剩余内存大小是多少？ESP32 比 ESP8266 内存多出很多。

myzhang001, 2019-02-22:
剩余的的空间还是有的，就是证书这里一直一直提示E (33947) aws_iot: failed! mbedtls_ssl_handshake returned -0x4310 E (33977) shadow: aws_iot_shadow_connect returned error -4, aborting...
把多余得任务关掉就可以正常跑，你们可以尝试一下，做一下试验，加上smart_config 代码后 就不能正常跑了

FayeY, 2019-03-12:
查询的剩余内存大小是多少？

myzhang001, 2019-03-12:
1496byte

FayeY, 2019-03-12:
确实是内存不足了，SSL 证书认证至少需要 25KByte，还需要留余地跑其他功能。

myzhang001, 2019-03-12:
也就是说  其实  目前这个RTOS sdk   8266  是不能做 aws iot 接入开发得,用在实际项目里面 还需要加其他代码

myzhang001, 2019-03-12:
RTOS sdk   本身占用的内存是多少呀

FayeY, 2019-03-12:
Hi @supreetd21 ，

Could you help with this issue?  It seems that after adding some application code to the aws_iot, this user only left 1496byte available memory to use, then the SSL will fail due to out of memory. 
How much memory will our aws_iot example take? Is there any chance that we can help with this issue?

Thanks.

TimLai201610, 2019-03-18:
請問一下aws-root-ca.pem, certificate.pem.crt, private.pem.key的大小是不是不固定啊? 有沒有一個範圍啊? 我想要把證書燒到spi flash 裡，而不想用example裡的方式:
extern const uint8_t aws_root_ca_pem_start[] asm("_binary_aws_root_ca_pem_start");
extern const uint8_t aws_root_ca_pem_end[] asm("_binary_aws_root_ca_pem_end");
extern const uint8_t certificate_pem_crt_start[] asm("_binary_certificate_pem_crt_start");
extern const uint8_t certificate_pem_crt_end[] asm("_binary_certificate_pem_crt_end");
extern const uint8_t private_pem_key_start[] asm("_binary_private_pem_key_start");
extern const uint8_t private_pem_key_end[] asm("_binary_private_pem_key_end");

Thanks!

donghengqaz, 2019-03-19:
你预留足够的空间就可以了。这些证书不大，你可以每个证书预留一个 sector.

closed 2019-04-17 by FayeY

JunJianX, 2019-09-19:
同样的问题

sky9971, 2019-11-02:
Sam here, i'm running out of memory while using http server along with ESP MQTT and the mbedtls handshake returns -0x4310 which indicates it is low on memory.