espressif/ESP8266_RTOS_SDK #475: 连接wifi耗时长问题

MinjunHuang, 2019-01-25:
## Environment
- Development Kit: ESP-LAUNCHER
- IDF version:  SDK version:v3.2-dev-9-gfc4e94d
- Development Env:      Make
- Operating System:     Ubuntu
- Power Supply:         USB

## Problem Description
1. 应用启动后，在链接wifi的过程中，总共耗时8s多，这个时间太长了，用户体验比较差，发现时间主要消耗在scan和获取ip地址上了；其中有两次scan，每次2.5s左右，获取IP地址，耗时2s。主要的时间消耗就在这两个操作。
2. 问题：
     1）为什么会有两次scan？应用在完成配网后，是否能够直接连接wifi？
     2）是否有好的做法能够降低上述两个操作（scan和获取IP地址）的耗时？

3.应用日志如下： 
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
2019-01-25 10:18:23.896:  0(.((I (4J}*"2%=mj5Rj%BJ}j
2019-01-25 10:18:24.090:  Jr"j
2019-01-25 10:18:24.123:  21-z*UMmj5Rj%BJ}j
2019-01-25 10:18:24.203:  Jr"j
2019-01-25 10:18:24.237:  21-z*UMmj5
2019-01-25 10:18:24.445:  SDK version:v3.2-dev-9-gfc4e94d
2019-01-25 10:18:24.476:  I (412) uart: queue free spaces: 100

2019-01-25 10:18:24.540:  mode : sta(2c:3a:e8:22:05:8e)
2019-01-25 10:18:24.572:  add if0
2019-01-25 10:18:24.573:  I (422) WIFI: initiate wifi station successful
2019-01-25 10:18:24.636:  
2019-01-25 10:18:24.637:  I (472) WIFI: start to connect AP....
2019-01-25 10:18:24.684:  
2019-01-25 10:18:26.986:  scandone
2019-01-25 10:18:27.002:  state: 0 -> 2 (b0)
2019-01-25 10:18:27.017:  state: 2 -> 2 (8a0)
2019-01-25 10:18:27.992:  state: 2 -> 0 (200)
2019-01-25 10:18:28.009:  I (3952) WIFI: connect to the AP, try: 1
2019-01-25 10:18:30.853:  scandone
2019-01-25 10:18:30.854:  state: 0 -> 2 (b0)
2019-01-25 10:18:30.885:  state: 2 -> 3 (0)
2019-01-25 10:18:30.901:  state: 3 -> 5 (10)
2019-01-25 10:18:30.917:  add 0
2019-01-25 10:18:30.933:  aid 3
2019-01-25 10:18:30.933:  pm open phy_2,type:2 0 0
2019-01-25 10:18:30.965:  cnt 
2019-01-25 10:18:30.965:  
2019-01-25 10:18:30.966:  connected with xxx, channel 1
2019-01-25 10:18:30.998:  I (6872) WIFI: connected to AP...
2019-01-25 10:18:31.045:  
2019-01-25 10:18:32.899:  I (8872) event: sta ip: 172.16.4.216, mask: 255.255.248.0, gw: 172.16.0.1
2019-01-25 10:18:32.996:  I (8872) WIFI: got ip:172.16.4.216

donghengqaz, 2019-01-29:
这个和 Wi-Fi 网络环境有关，如果网络环境好就快一点，如果网络不好就慢一点。

MinjunHuang, 2019-01-31:
1. 可能不一定是wifi网络环境的问题，scan一次信道是合理，但是耗时2s多，确实有点长，发送数据的网络处理都很快。
2. 从处理dhcp client的debug日志，看到经过了多次delay，是不是内部延迟时间过长？

MinjunHuang, 2019-02-19:
1. scan时间过程可以通过保存连接AP的channel，并设置wifi_config.sta.scan_method = WIFI_FAST_SCAN来缩短时间。
2. DHCP client获取IP地址的处理，从文档中没有找到处理办法。

Junhao-Espressif, 2019-04-10:
Hi @MinjunHuang 

Please try again with latest VERSION.

Thank you!

closed 2019-04-17 by FayeY