espressif/ESP8266_RTOS_SDK #476: GPIO output only has effect on pin 2

xyx0826, 2019-01-25:
## Environment
- Development Kit: **NodeMCU**
- IDF version: **e634398**
- Development Env:      **Make**
- Operating System:     **Ubuntu** (WSL Ubuntu 18.04.1 LTS)
- Power Supply:         **USB**

## Problem Description
I am using a modified version of [espressif/esp-idf](https://github.com/espressif/esp-idf/tree/master/examples/get-started/blink)'s `blink` example on ESP8266. It works by changing the pin level between `0` and `1` in a loop to blink a LED.

The code is able to blink the onboard LED at GPIO port `2`, I can also verify level changes on this port with a multimeter.

However, when I specify an alternative port (e.g. `12` or `14`), the specified pin would stay at high level with no changes.

(Similar to #297?)

### Expected Behavior
The level on the specified pin will keep changing between low and high.

### Actual Behavior
The level on the specified pin stays high.

### Steps to repropduce
0. `make menuconfig` and set GPIO pin to `2`, aka onboard LED pin
1. `make -j 8 flash monitor` to flash the code
2. Attach a multimeter to `GND` and `D4`, which is GPIO `2`
3. The pin voltage will change between `0v` and `3.3v` once a second. The LED will also blink.
4. Now, `make menuconfig` and set GPIO pin to `12`
5. Attach a multimeter to `GND` and `D6`, which is GPIO `12`
6. While calls to `gpio_set_level` will be successful, the pin will stay at `3.3v` with no changes.

### Code to reproduce this issue
```cpp
#include <stdio.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "driver/gpio.h"
#include "sdkconfig.h"

#include "esp_log.h"

// Value is set via menuconfig
#define BLINK_GPIO CONFIG_BLINK_GPIO

static const char * TAG = "Blink";

void blink_task(void *pvParameter)
{
    ESP_LOGI(TAG, 
        "Setting PIN direction to output. Result: %d", 
        gpio_set_direction(BLINK_GPIO, GPIO_MODE_OUTPUT));
    
    while(1) {
        /* Blink off (output low) */
        ESP_LOGI(TAG, 
            "Setting GPIO to 0. Result: %d", 
            gpio_set_level(BLINK_GPIO, 0));
        
        // Wait for one second
        vTaskDelay(1000 / portTICK_PERIOD_MS);

        /* Blink on (output high) */
        ESP_LOGI(TAG, 
            "Setting GPIO to 1. Result: %d", 
            gpio_set_level(BLINK_GPIO, 1));
        
        // Wait for one second
        vTaskDelay(1000 / portTICK_PERIOD_MS);
    }
}

void app_main()
{
    xTaskCreate(&blink_task, "blink_task", configMINIMAL_STACK_SIZE, NULL, 5, NULL);
}
```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs
```
--- idf_monitor on /dev/ttyS4 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ap
I rl d| l|$c|;ccg'lngbxl;l{lpndcg|lc'o d$`gdl`or'd gcolgn $p'sco<c'o $l`g$d`gso``'bgdno lxo{#g|x#no $l nl$`'{ocl s'bhj%(50) boot: ESP-IDF v3.2-dev-7-ge634398 2nd stage bootloader
I (50) boot: compile time 14:48:08
I (50) qio_mode: Enabling default flash chip QIO
I (55) boot: SPI Speed      : 40MHz
I (59) boot: SPI Mode       : QIO
I (63) boot: SPI Flash Size : 2MB
I (67) boot: Partition Table:
I (70) boot: ## Label            Usage          Type ST Offset   Length
I (78) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (85) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (92) boot:  2 factory          factory app      00 00 00010000 000f0000
I (100) boot: End of partition table
I (104) esp_image: segment 0: paddr=0x00010010 vaddr=0x40210010 size=0x307a0 (198560) map
I (113) esp_image: segment 1: paddr=0x000407b8 vaddr=0x3ffe8000 size=0x0053c (  1340) load
I (122) esp_image: segment 2: paddr=0x00040cfc vaddr=0x3ffe853c size=0x0011c (   284) load
I (130) esp_image: segment 3: paddr=0x00040e20 vaddr=0x40100000 size=0x05968 ( 22888) load
0x40100000: _stext at ??:?

I (139) boot: Loaded app from partition at offset 0x10000
D (154) nvs: nvs_flash_init_custom partition=nvs start=9 count=6
I (174) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
D (174) nvs: nvs_open_from_partition backup_mac 0
D (174) nvs: nvs_get_str_or_blob backup_mac_data
D (174) nvs: nvs_close 1
I (184) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
D (184) nvs: nvs_open_from_partition backup_mac 0
D (184) nvs: nvs_get_str_or_blob backup_mac_data
D (194) nvs: nvs_close 2
D (194) phy_init: loading PHY init data from application binary
D (194) nvs: nvs_open_from_partition phy 0
D (194) nvs: nvs_get_str_or_blob cal_data
D (204) nvs: nvs_get_str_or_blob dc_table
D (204) nvs: nvs_close 3
I (384) reset_reason: RTC reset 2 wakeup 0 store 0, reason is 2
D (384) wdt: Enable task watch dog panic, panic time parameter is 11
D (384) wdt: task watch dog trigger time parameter is 15
I (394) Blink: Setting PIN direction to output. Result: 0
I (394) Blink: Setting GPIO to 0. Result: 0
I (1404) Blink: Setting GPIO to 1. Result: 0
I (2404) Blink: Setting GPIO to 0. Result: 0
I (3404) Blink: Setting GPIO to 1. Result: 0
I (4404) Blink: Setting GPIO to 0. Result: 0
I (5404) Blink: Setting GPIO to 1. Result: 0
I (6404) Blink: Setting GPIO to 0. Result: 0
I (7404) Blink: Setting GPIO to 1. Result: 0
I (8404) Blink: Setting GPIO to 0. Result: 0
I (9404) Blink: Setting GPIO to 1. Result: 0
I (10404) Blink: Setting GPIO to 0. Result: 0
...
```

## Other items if possible
- [x] [sdkconfig file](https://github.com/espressif/ESP8266_RTOS_SDK/files/2797655/sdkconfig.zip)
- [x] [elf file in the ``build`` folder](https://github.com/espressif/ESP8266_RTOS_SDK/files/2797653/blink.zip)

xiongyumail, 2019-01-25:
Hi @xyx0826 ，

I think you need to init the GPIO before using it, please refer to the GPIO example.
<https://github.com/espressif/ESP8266_RTOS_SDK/blob/master/examples/peripherals/gpio/main/user_main.c>

xyx0826, 2019-01-25:
Thank you very much. I was able to solve this issue with `gpio_config`:

```cpp
// Define the pin mask to be configurated
#define GPIO_OUTPUT_PIN_SEL  (1ULL<<BLINK_GPIO)

void initialize_gpio()
{
    gpio_config_t io_conf;
    //disable interrupt
    io_conf.intr_type = GPIO_INTR_DISABLE;
    //set as output mode
    io_conf.mode = GPIO_MODE_OUTPUT;
    //bit mask of the pins that you want to set,e.g.GPIO15/16
    io_conf.pin_bit_mask = GPIO_OUTPUT_PIN_SEL;
    //disable pull-down mode
    io_conf.pull_down_en = 0;
    //disable pull-up mode
    io_conf.pull_up_en = 0;
    //configure GPIO with the given settings
    gpio_config(&io_conf);
    ESP_LOGI(TAG, "GPIO config updated.");
}
```
And by calling it in `app_main` the GPIO pin now functions as expected.

closed 2019-01-25 by xyx0826