espressif/ESP8266_RTOS_SDK #484: uart_enable_swap() blocks indefinitely

dreske, 2019-02-08:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at bbs.espressif.com instead.

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs on the monitor](http://esp-idf.readthedocs.io/en/latest/get-started/idf-monitor.html#automatically-decoding-addresses).
4. Provide more items under **Other items if possible** can help us better locate your problem.
5. Use markup (buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:
- IDF version (``git rev-parse --short HEAD`` to get the commit id.):  a0a9f22
- Development Env:     eclipse
- Operating System:     Arch Linux
- Power Supply:         external 5V

## Problem Description
When I try to use the secondary gpio's for uart0, the swap function never returns. Issue was introduced in d1ab0c20dda8c1f181d61a45e0ee0f587a115d3a (uart.c)

**Remark**: sdk is configured to use UART1 for console output.

It is locally solved by adding following line to uart.c (line 898 @ master)
´´´
xSemaphoreGive(p_uart_obj[uart_num]->tx_done_sem);
´´´
But I don't know, if this is the correct solution, but it sounds like.

### Expected Behavior
UART0 swaps gpio's

### Actual Behavior
uart_enable_swap() blocks indefinitely. 

### Steps to repropduce

### Code to reproduce this issue
https://gist.github.com/dreske/547efb10769032952de4ac65d90c822f

xiongyumail, 2019-02-21:
Do you have to try to update to the latest SDK?

FayeY, 2019-04-17:
Closing due to lack of response, please feel free to reopen if this is still an issue with the latest SDK.
Thanks.

closed 2019-04-17 by FayeY