espressif/ESP8266_RTOS_SDK #485: undefined reference to `uxTaskGetSystemState'

nopnop2002, 2019-02-10:
TaskStatus_t *pxTaskStatusArray;
   volatile UBaseType_t uxArraySize, x;
   uint32_t ulTotalRunTime;

   uxArraySize = uxTaskGetSystemState( pxTaskStatusArray,
                                 uxArraySize,
                                 &ulTotalRunTime );

	$ make
	LD build/i2c.elf
	/home/nop/ESP8266_RTOS_SDK/rtos/task-test3/build/main/libmain.a(user_main.o):(.literal.app_main+0x1c): undefined reference to `uxTaskGetSystemState'
	/home/nop/ESP8266_RTOS_SDK/rtos/task-test3/build/main/libmain.a(user_main.o): In function `app_main':
	/home/nop/ESP8266_RTOS_SDK/rtos/task-test3/main/user_main.c:47: undefined reference to `uxTaskGetSystemState'
	collect2: error: ld returned 1 exit status
	/home/nop/ESP8266_RTOS_SDK/make/project.mk:405: recipe for target '/home/nop/ESP8266_RTOS_SDK/rtos/task-test3/build/i2c.elf' failed
	make: *** [/home/nop/ESP8266_RTOS_SDK/rtos/task-test3/build/i2c.elf] Error 1

donghengqaz, 2019-02-12:
Set ”#define configUSE_TRACE_FACILITY	0“  to be  "#define configUSE_TRACE_FACILITY	1" in file FreeRTOSConfig.h.

27-301, 2019-02-21:
@donghengqaz Please add a menuconfig to configure trace facility.

nopnop2002, 2019-03-02:
Do you have a plan to include in menuconfig that trace facility becomes effective or not??

nopnop2002, 2019-03-08:
I confirmed that configUSE_TRACE_FACILITY can become effectively using menuconfig.
Please close this issue.

closed 2019-04-17 by FayeY