espressif/ESP8266_RTOS_SDK #486: how go in lightsleep when after using SNTP get time

thisdream, 2019-02-12:
----------------------------- Delete below -----------------------------

how go in lightsleep when after using SNTP get time 

INSTRUCTIONS
============

I use "wifi_fpm_set_sleep_type(LIGHT_SLEEP_T)" make ESP go to lightsleep mode.But after using "sntp_init()",ESP cann't go to lightsleep mode.Although I using "sntp_stop()",ESP cann't go to lightsleep mode

## Debug Logs

```
state: 5 -> 0 (0)
rm 0
pm close 7 0 0/14454336
del if0
usl
sul 0 0
mode : null
wifi opmode is 0force slp enable,type: 1
fpm open,type:1 0
```
The program will continue ,don't sleep

Junhao-Espressif, 2019-02-21:
Hi @thisdream ,

Have you measured the current of ESP8266?

Please give me your source code.

Thank you!

FayeY, 2019-04-17:
May I ask if there is any update about this issue?

closed 2019-04-24 by FayeY