espressif/ESP8266_RTOS_SDK #487: Reveice many wifi disconnect events before wifi connected.

nghiaquy1991, 2019-02-14:
## Environment

- Development Kit: ESP8266 12F
- IDF version (``git rev-parse --short HEAD`` to get the commit id.): 7d4782f
- Development Env:      Make
- Operating System:     Ubuntu
- Power Supply:         external 5V

## Problem Description
I set up my board to connect a Wifi, with SSID and PASS.
It can connect normally. But, sometimes, I saw that it had many Wifi disconnect events (EVENT_STAMODE_DISCONNECTED) send to  wifi_event_handler_cb_t before receiving EVENT_STAMODE_CONNECTED event --> It took a long time to connect the Wifi. I want to optimize this duration.

I tried to debug why I got EVENT_STAMODE_DISCONNECTED event:
1. I checked the connect status by using wifi_station_get_connect_status() functions with 500 ms period.
I saw that it always return STATION_WRONG_PASSWORD when event EVENT_STAMODE_DISCONNECTED  occurred --> I guessed that SSID or PASS maybe wrong!
2. Then I checked Wifi SSID and Pass to make sure that It was correct by using wifi_station_get_config and wifi_station_get_config_default functions --> ??

Could you explain to me why I got EVENT_STAMODE_DISCONNECTED event? 
What can I do to optimize Wifi connection time?

Thanks,

Junhao-Espressif, 2019-02-21:
Hi @nghiaquy1991 ，

Please give me your code and log.

nghiaquy1991, 2019-02-25:
Hi @Junhao-Espressif 

I will re-test again, then send to you a log and code.

FayeY, 2019-04-17:
May I ask if there is any update?

closed 2019-04-17 by nghiaquy1991