espressif/ESP8266_RTOS_SDK #488: Crash when using custom UART ISR

Bennji94, 2019-02-14:
## Environment

- Development Kit: NodeMCU Board with ESP-12E
- IDF version (``git rev-parse --short HEAD`` to get the commit id.): 
    02ffe0e
- Development Env:      [Make]
- Operating System:     [Ubuntu]
- Power Supply:         [USB]

## Problem Description

I registered a custom ISR for UART0, but when connecting the RX pin the ESP8266 crashes running in ISR context. I initialized the UART0 with uart_init() and registered the custom ISR.

### Actual Behavior
ESP8266 crashes when receaving UART messages.

### Code to reproduce this issue

```cpp
#define portYIELD_FROM_ISR() taskYIELD()
#define UART_ENTER_CRITICAL()    portENTER_CRITICAL()
#define UART_EXIT_CRITICAL()     portEXIT_CRITICAL()

#define UART_CHECK(a, str, ret_val) \
    if (!(a)) { \
        ESP_LOGE(TAG,"%s(%d): %s", __FUNCTION__, __LINE__, str); \
        return (ret_val); \
    }

static DRAM_ATTR uart_dev_t *const UART[UART_NUM_MAX] = {&uart0, &uart1};

void uart_init(void) {
	uart_driver_delete(UART_NUM_0);
	uart_config_t uart_config = { .baud_rate = 9600, .data_bits =
			UART_DATA_8_BITS, .parity = UART_PARITY_DISABLE, .stop_bits =
			UART_STOP_BITS_1, .flow_ctrl = UART_HW_FLOWCTRL_DISABLE };
	uart_param_config(UART_NUM_0, &uart_config);

	uart_driver_install(UART_NUM_0, UART_BUF_SIZE * 2, 0, 0, NULL);
	uart_intr_config_t uart_intr = {
		        .intr_enable_mask = UART_FRM_ERR_INT_ENA_M | UART_PARITY_ERR_INT_ENA_M
        		| UART_RXFIFO_OVF_INT_ENA_M | UART_RXFIFO_FULL_INT_ENA_M
				| UART_BRK_DET_INT_ENA,
		        .rxfifo_full_thresh = 1,
		        .rx_timeout_thresh = UART_TOUT_THRESH_DEFAULT,
		        .txfifo_empty_intr_thresh = 0
		    };
		    uart_intr_config(UART_NUM_0, &uart_intr);

if (uart_isr_register(UART_NUM_0, UARTIntReceviceHandler, 0)
				!= ESP_OK) {
			ESP_LOGE(TAG, "Failed to register uart isr");
			return FAILED;
		} else {
			ESP_LOGD(TAG, "SUCCEDED to register uart isr");
		}

}

static void UARTIntReceviceHandler(void *param) {
	  uart_obj_t *p_uart = (uart_obj_t *) param;
	    uint8_t uart_num = p_uart->uart_num;
	    uart_dev_t *uart_reg = UART[uart_num];
	    int rx_fifo_len = uart_reg->status.rxfifo_cnt;
	    uint8_t buf_idx = 0;
	    uint32_t uart_intr_status = UART[uart_num]->int_st.val;
	    uart_event_t uart_event;
	    BaseType_t task_woken = 0;

	    while (uart_intr_status != 0x0) {
[... 164 lines cut ...]
```

## Debug Logs

Core 0 was running in ISR context:

00000000 00000004 00000008 0000000c 00000010 00000014 00000018 0000001c 00000020 00000024 00000028 0000002c 00000030 00000034 00000038 0000003c 

3ffe82e0  00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 
3ffe8320  00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 
3ffe8360  00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 
3ffe83a0  00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 
3ffe83e0  00000000 00000000 00000000 00000000 00000000 00000000 00000000 40100715 00000000 00000000 00000000 40240ba2 4024b5dd 00000000 00000000 00000000 
3ffe8420  00000000 00000000 00000000 40100715 00000004 00000000 40106efc 40240ba2 401066e4 40240540 00000000 40240ba2 4022435c 0000017a 00000000 4023f6ab 
3ffe8460  3ffe8498 000000ff 3fff2878 4023efcd 3ffe84a0 000000ff 3fff3770 4010068b 00000400 00000400 3ffe84e8 4022eedd 00000005 00000000 00000001 00000001 
3ffe84a0  00000020 00000020 00000000 40258049 40100779 4022ef08 3fff0348 4010078d 3fff1a28 00000002 40222300 00000000 00000000 00000000 4010552c 40105574 
      PC: 0x4024c53b       PS: 0x00000031       A0: 0x40258049       A1: 0x3ffe8470 
0x4024c53b: esp_aio_event at /home/may2lol/esp/ESP8266_RTOS_SDK/components/esp8266/source/esp_socket.c:179

      A2: 0x00000000       A3: 0x4024c52c       A4: 0x00003fff       A5: 0x401065e0 
0x4024c52c: esp_aio_sendto at /home/may2lol/esp/ESP8266_RTOS_SDK/components/esp8266/source/esp_socket.c:172

      A6: 0xfc7000f7       A7: 0x3feffe00       A8: 0xfc70ffff       A9: 0x22800000 
     A10: 0x20000000      A11: 0xcfffffff      A12: 0x00000000      A13: 0x00000020 
     A14: 0x00000020      A15: 0x00000001      SAR: 0x0000001b EXCCAUSE: 0x0000001c

xiongyumail, 2019-02-17:
Why do you need to implement the UART interrupt handler function yourself?

donghengqaz, 2019-02-18:
Ｉ seems that you can call "WIFI send" function at ISR, and it may not be handled. Could some fucntions at the ISR may be related to IP stack or WIFI ?

Bennji94, 2019-02-18:
> Why do you need to implement the UART interrupt handler function yourself?

I want to write the UART interrupt handler function myself because i need a faster byte by byte processing. 

> Ｉ seems that you can call "WIFI send" function at ISR, and it may not be handled. Could some fucntions at the ISR may be related to IP stack or WIFI ?

I used the uart_rx_intr_handler_default code from driver/uart.c. I don't know if there are any function related to the IP stack or WIFI.

xiongyumail, 2019-02-21:
@Bennji94 
If you write your own interrupt function, please pay attention to the initialization of some variables.

Bennji94, 2019-02-28:
Thank you. Initializing p_uart solved it.

closed 2019-02-28 by Bennji94