espressif/ESP8266_RTOS_SDK #489: [Important] V3.1 OTA issue

FayeY, 2019-02-15:
**Note**:

Since the ESP8266 RTOS SDK V3.0, we refactored the SDK to be the ESP-IDF style. And provide an FOTA example in V3.1 to support upgrading from an old SDK to the new SDK.
- "old to new" FOTA: upgrade from an old (NonOS, or V2.x and earlier versions RTOS) SDK to a new (v3.x and later versions) RTOS SDK. 
- "new to new" FOTA: upgrade from a new (V3.x) RTOS SDK to a new (V3.x and later versions) RTOS SDK. 

**Current Situation**：

In the V3.1 FOTA example, "old to new" FOTA and "new to new" FOTA are alternative, not compatible.   
-  The firmware for the "old to new" FOTA is a combined bin of all new version binaries, including the new boot.bin, new app.bin, new partition.bin and so on. So the FOTA can make it completely upgrade to a new (ESP-IDF) style firmware.
-  The firmware for the "new to new" FOTA is just a new app.bin, since the other parts are already the new style ones.

So in this case, "old to new" FOTA and "new to new" FOTA are not compatible.  

**Possible Issue**：

- There will be a problem if both old and new version products are using the same url for FOTA. If the old version products downloading a "new to new" firmware, or the new version products downloading an "old to new" firmware, then the system will fail to boot.

**Temporary Workaround**：

- Use different OTA urls for the "old to new" FOTA and "new to new" FOTA respectively. To ensure the old version products upgrade to the "old to new" firmware, and the new version products upgrade to the "new to new" firmware.

**Future Final Solution**：

- We are working on the OTA example that can be compatible with both "old to new" and "new to new" FOTA now, it will be released soon.

So sorry for the inconvenience.

Thanks.

**说明**：

从 ESP8266 RTOS SDK v3.0 起，我们将 RTOS SDK 重构为 ESP-IDF 架构，并在 v3.1 提供了 OTA 示例支持从旧版本升级到新版本:
- 旧版本（NonOS 或 v2.x 及之前的 RTOS）升级到新版本（V3.x RTOS），后文将称为“旧升新”；
- 新架构（V3.x RTOS）升级到新版本（V3.x及之后的 RTOS），后文将称为“新升新”；

**当前现状**：

V3.1 的 OTA 方案不兼容 “旧升新“ 和 “新升新“，两者只能选其一。这是因为：
-  “旧升新” 时，OTA 下载的是合并后的完整新版固件（包含新版 boot.bin, 新版 app.bin, 新版 partition.bin 等），以保障更新后能完整运行新版固件；
-  “新升新” 时，OTA 下载的仅是 app.bin，只更新应用部分，其他部分使用原新版的即可；

**可能导致的问题**：

- 如果客户的旧版本产品和新版本产品，共用同一 OTA 升级下载链接；则会出现 “旧升新“ 和 “新升新“ 固件的混淆，难以对应下载。旧版本产品如果下载了 “新升新“ 固件，或者新版本产品下载了 “旧升新“ 固件， 将无法正常运行。

**临时解决方案**：

- 在开发时，将“旧升新“和“新升新“的 OTA 下载链接进行区分，不要共用；旧产品下载 “旧升新” 固件，新产品下载 “新升新” 固件。

**最终解决办法**：

- 目前我们正在开发兼容“旧升新“和“新升新“的 OTA 方案，将会在近期推出，敬请关注。

给您带来不便，深感抱歉。

kmalloc8, 2019-07-15:
设备已经烧录分区表"Single factory app, no OTA", 还有可能做OTA么?

ustccw, 2019-07-19:
@kmalloc8 抱歉, 一级 `partition table` 写死 `no OTA` 是没法 `OTA` 的

kmalloc8, 2019-07-31:
@ustccw ,您好，在进行OTA测试的时候，比较容易出现重启现象，从这个log可以看出来什么吗？`register map is 3ffed9a0 error`说明什么呢？谢谢。

> I (22868) ota: Connection closed, all packets received
I (22868) ota: Total Write binary data length : 405780
I (25878) esp_image: segment 0: paddr=0x00110010 vaddr=0x40210010 size=0x5c4f0 (378096) map
I (25948) esp_image: segment 1: paddr=0x0016c508 vaddr=0x3ffe8000 size=0x005a0 (  1440) 
I (25958) esp_image: segment 2: paddr=0x0016cab0 vaddr=0x3ffe85a0 size=0x00190 (   400) 
I (25958) esp_image: segment 3: paddr=0x0016cc48 vaddr=0x40100000 size=0x064a0 ( 25760) 
I (26768) main: Free heap size: 25196 Bytes
I (27978) esp_image: segment 0: paddr=0x00110010 vaddr=0x40210010 size=0x5c4f0 (378096) map
I (28048) esp_image: segment 1: paddr=0x0016c508 vaddr=0x3ffe8000 size=0x005a0 (  1440) 
I (28058) esp_image: segment 2: paddr=0x0016cab0 vaddr=0x3ffe85a0 size=0x00190 (   400) 
I (28058) esp_image: segment 3: paddr=0x0016cc48 vaddr=0x40100000 size=0x064a0 ( 25760) 

Task stack [ota_example_tas] stack from [0x3fff8614] to [0x3fffa610], total [8192] size

                   0          4          8          c         10         14         18         1c         20         24         28         2c         30         34         38         3c 

3fffa580  0x00000110 0x60000240 0x00000002 0x40212eeb 0x3ffeea28 0x40107730 0x3ffee449 0x00000019 0x00000102 0x00000002 0x00072114 0x0000f2bc 0x00063114 0x00063114 0x0000f000 0x3ffede6c 
3fffa5c0  0x00000048 0x000000bb 0x40108eec 0x00000001 0x3ffede6c 0x00000002 0x00000001 0x00063114 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 
3fffa600  0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 

        PC: 0x87159c1b        PS: 0x00000030        A0: 0x0168a101        A1: 0x3fffa5a0
        A2: 0x8703a8c0        A3: 0x00000000        A4: 0x00000001        A5: 0x4000422e
        A6: 0x00000102        A7: 0x00000000        A8: 0x00000000        A9: 0x00000110
       A10: 0x60000240       A11: 0x00000002       A12: 0x40212eeb       A13: 0x3ffeea28
       A14: 0x40107730       A15: 0x3ffee449       SAR: 0x00000019  EXCCAUSE: 0x0000001d

Task stack [ota_example_tas] stack from [0x3fff8614] to [0x3fffa610], total [8192] size

register map is 3ffed9a0 error
                   0          4          8          c         10         14         18         1c         20         24         28         2c         30         34         38         3c 

3fff8614  0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 
3fff8654  0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 
3fff8694  0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 
3fff86d4  0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 
3fff8714  0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 
3fff8754  0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5

ustccw, 2019-07-31:
`crash` 信息看, `PC` 和 `A0` 已经紊乱. `EXCCAUSE` 看是非法存储.
似乎在打印 `Total Write binary data length` 前后进行了一点非法操作, 可否提供您的完整测试代码, 给我们复现。

ustccw, 2019-07-31:
@kmalloc8  `crash` 信息看, `PC` 和 `A0` 已经紊乱. `EXCCAUSE` 看是非法存储.
似乎在打印 `Total Write binary data length` 前后进行了一点非法操作, 可否提供您的完整测试代码, 给我们复现。

ustccw, 2019-07-31:
@kmalloc8 是不是 `partition table` 没有选择 `OTA` 方式呢

kmalloc8, 2019-09-10:
@ustccw ，不要意思，是我非法操作导致的问题。。。

JunJianX, 2019-10-09:
@ustccw  OTA的时候，从url升级就会卡住，从自己局域网内的服务器下载就不会，是怎么回事？

ustccw, 2019-10-09:
@JunJianX https://github.com/espressif/ESP8266_RTOS_SDK/issues/727