espressif/ESP8266_RTOS_SDK #490: how to set os_printf func to uart1 in esp8266 3.1 RTOS_SDK ?

1XiaSheng, 2019-02-15:
新的SDK中怎样将os_printf 转移到 uart1 tx 上, 
更改 CONFIG_CONSOLE_UART_NUM = 1 也没效果.
    // Configure parameters of an UART driver,
    // communication pins and install the driver
    uart_config_t uart_config = {
        .baud_rate = 115200,
        .data_bits = UART_DATA_8_BITS,
        .parity    = UART_PARITY_DISABLE,
        .stop_bits = UART_STOP_BITS_1,
        .flow_ctrl = UART_HW_FLOWCTRL_DISABLE
    };
    // cmm
    uart_param_config(UART_NUM_0, &uart_config);
    uart_driver_install(UART_NUM_0, UART_BUF_SIZE+1 , UART_BUF_SIZE+1, UART_BUF_SIZE-1, &uart0_queue);
    // uart 1 debug info(use gpio2)
    PIN_FUNC_SELECT(PERIPHS_IO_MUX_GPIO2_U, FUNC_U1TXD_BK);
    uart_param_config(UART_NUM_1, &uart_config);
    uart_driver_install(UART_NUM_1, 0 , UART_BUF_SIZE+1, UART_BUF_SIZE+1, NULL);

初始化uart1后 使用uart_write_bytes(UART_NUM_1, (const char *) "detected invalid button press;---9", 34);有效,有没有更简单的办法?

donghengqaz, 2019-02-18:
如果你说的新的版本是 v3.1 的话，可以直接在 menuconfig 里面直接配置：
```
Component config  --->
    ESP8266-specific  --->
        UART for console output (Custom)  --->
        UART peripheral to use for console output (0-1) (UART1)  --->
```
在 v3.0 版本配置项可能有些出入，但是大致类似。

1XiaSheng, 2019-02-18:
@donghengqaz  
谢谢,已经在你描述的位置找到配置.

closed 2019-02-18 by 1XiaSheng

xuhongv, 2019-03-12:
@donghengqaz 请问您通过make menuconfig解决了？我下面配置也是不行的。还是会从 uart0打印数据！

----------------------------------------------

[image]