espressif/ESP8266_RTOS_SDK #491: fatal: bad revision 'HEAD'

DCSA35, 2019-02-15:
Hello,

working with RTOS SDK 2.0 on an ESP8266
on 'make' I got always
fatal: bad revision 'HEAD'
In fact it is not fatal, everything works fine.
What is the meaning?
Thank you.

donghengqaz, 2019-02-18:
If you develop with V2.x.x ESP8266 SDK, you may use "gen_misc.sh" under the root directory to build the project instead of command "make".

FayeY, 2019-02-21:
It is suggested to use our latest [ESP8266 RTOS SDK](https://github.com/espressif/ESP8266_RTOS_SDK/releases). Thanks.

closed 2019-03-12 by FayeY

DCSA35, 2019-03-13:
Dear FayeY,

 

thank you for help.

So far I did not find exact reason, maybe it is a PATH problem,

because I use ESP8266 and ESP32 in parallel.

Anyway it is no real problem.

 

Best wishes

 

 

Gesendet: Dienstag, 12. März 2019 um 03:42 Uhr

Closed #491.

FayeY, 2019-03-13:
With V2.x.x ESP8266 RTOS SDK, the command for compilation is "gen_misc.sh".

It is suggested to use the latest V3.x ESP8266 RTOS SDK which is the same as the ESP32-IDF, you can compile it with the command "make".

Please feel free to reopen the issue, if your problem is still unsolved.

Thanks.