espressif/ESP8266_RTOS_SDK #492: Cannot compile with CMake

Deous, 2019-02-17:
Used latest RTOS SDK with CMake - idf.py

IDE - Visual Studio Code
Windows 10

Getting errors:

-- The ASM compiler identification is GNU
-- Found assembler: C:/Users/dev/Documents/Esp8266/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc.exe
-- The C compiler identification is GNU 5.2.0
-- The CXX compiler identification is GNU 5.2.0
-- Check for working C compiler: C:/Users/dev/Documents/Esp8266/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc.exe
-- Check for working C compiler: C:/Users/dev/Documents/Esp8266/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc.exe -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: C:/Users/dev/Documents/Esp8266/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++.exe
-- Check for working CXX compiler: C:/Users/dev/Documents/Esp8266/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++.exe -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Error at C:/Users/dev/Documents/Esp8266/esp-idf/tools/cmake/project.cmake:118 (add_subdirectory):
add_subdirectory given source "cjson" which is not an existing directory.
Call Stack (most recent call first):
CMakeLists.txt:8 (project)

CMake Error at C:/Users/dev/Documents/Esp8266/esp-idf/tools/cmake/project.cmake:118 (add_subdirectory):
add_subdirectory given source "coap" which is not an existing directory.
Call Stack (most recent call first):
CMakeLists.txt:8 (project)

How can I compile it with CMake using latest idf.py esptool.py? 
Please help

donghengqaz, 2019-02-18:
We will make "Cmake" function work well later, please wait.

Deous, 2019-02-18:
How soon you guys can make it?
I need it now ) I can help testing 
Thanks

donghengqaz, 2019-02-19:
Now I am handing the OTA problem, if it goes well, I will start "Cmake" at tomorrow (02.20), otherwise it will start "Cmake" at the next tomorrow(02.21)。

Deous, 2019-02-22:
Thank you! I look forward to any updates

Deous, 2019-02-24:
Question:  Are you using any specific tool to test CMake?

donghengqaz, 2019-02-27:
I am trying to add Cmake, but it depends on some extra tools, because the SDK must be same as "esp-idf".

Deous, 2019-02-28:
What specific tools it depends on? Can you maybe describe problem in more details?
Is it a bootloader issue?

Deous, 2019-04-14:
Any progress on CMake?

donghengqaz, 2019-05-16:
The cmake supporting has been merged into master branch, I am sorry to delay much time, because it is very complex and I should to push the SDK release to be OK.

Deous, 2019-06-17:
Hey - thank you. I also wanted to suggest that you probably should keep IDF_PATH for ESP8266 and ESP32 separately and use two symbols like IDF_PATH_ESP8266 and IDF_PATH_ESP32 and then when all is seamlessly merged and tested use one variable.Right now it is impossible to work on the same OS with both ESP32 and ESP8266 IDEs - non Arduino.Maybe you should write your own IDE for ESP also.Thanks!Temat: Re: [espressif/ESP8266_RTOS_SDK] Cannot compile with CMake (GIT8266O-70) (#492)Data: 2019-05-16 10:56Nadawca: "dongheng" Adresat: "espressif/ESP8266_RTOS_SDK" ; DW: "Deous" ; "Author" ; The cmake supporting has been merged into master branch, I am sorry to delay much time, because it is very complex and I should to push the SDK release to be OK.

FayeY, 2019-07-15:
Hi, cmake is added in the v3.2. But sorry that we may not change the IDF_PATH. Thanks.

closed 2019-07-15 by FayeY