espressif/ESP8266_RTOS_SDK #493: Can't compile SDK v3.1

r44083, 2019-02-18:
## Environment
- Development Kit: ESP8266 dev board
- IDF version: ESP8266_RTOS_SDK v3.1
- Development Env: Make
- Operating System: Windows 10
- Power Supply: USB

I am trying to build simple test project with **ESP8266_RTOS_SDK v3.1**

Compilation is successful, but there are some errors during linking. Undefined reference to the following functions:
```wDev_ProcessFiq```
```NMIIrqIsOn```
```WDEV_INTEREST_EVENT```
```MacIsrSigPostDefHdl```
```pmIdleHook```
```CCOMPARE_AddVal```
```xthal_set_intset```

```esp_random```
```esp_wifi_set_rx_pbuf_mem_type```

```phy_rx_gain_dc_table```
```phy_check_data_table```
```phy_afterwake_set_rfoption```
```phy_get_bb_evm```

```write_data_to_rtc```
```chip_init```
```get_data_from_rtc```

<details>
<summary>Build log</b></summary>

```
make -j 8 --no-print-directory -C src
xtensa-lx106-elf-c++  -I../src -I../src/hal/ESP8266/ESP8266_RTOS_SDK -I../src/hal/ESP8266/ESP8266_RTOS_SDK/components/esp8266/include -I../src/hal/ESP8266/ESP8266_RTOS_SDK/components/freertos/include/freertos -I../src/hal/ESP8266/ESP8266_RTOS_SDK/components/freertos/include/freertos/private -I../src/hal/ESP8266/ESP8266_RTOS_SDK/components/freertos/port/esp8266/include -I../src/hal/ESP8266/ESP8266_RTOS_SDK/components/freertos/port/esp8266/include/freertos -I../src/hal/ESP8266/ESP8266_RTOS_SDK/components/heap/include -I../src/hal/ESP8266/ESP8266_RTOS_SDK/components/heap/port/esp8266/include -O0 -g3 -Wall -ffunction-sections -fdata-sections -mlongcalls -mtext-section-literals -std=c++11 -fno-exceptions -fno-rtti -felide-constructors -fno-threadsafe-statics -fno-use-cxa-atexit -c main.cpp -o ../out/obj/src/main.o
make -j 8 --no-print-directory -C src/common
xtensa-lx106-elf-gcc-5.2.0  -I../../src -I../../src/hal/ESP8266/ESP8266_RTOS_SDK -I../../src/hal/ESP8266/ESP8266_RTOS_SDK/components/esp8266/include -I../../src/hal/ESP8266/ESP8266_RTOS_SDK/components/freertos/include/freertos -I../../src/hal/ESP8266/ESP8266_RTOS_SDK/components/freertos/include/freertos/private -I../../src/hal/ESP8266/ESP8266_RTOS_SDK/components/freertos/port/esp8266/include -I../../src/hal/ESP8266/ESP8266_RTOS_SDK/components/freertos/port/esp8266/include/freertos -I../../src/hal/ESP8266/ESP8266_RTOS_SDK/components/heap/include -I../../src/hal/ESP8266/ESP8266_RTOS_SDK/components/heap/port/esp8266/include -O0 -g3 -Wall -ffunction-sections -fdata-sections -mlongcalls -mtext-section-literals -std=c99 -c assert.c -o ../../out/obj/src/common/assert.o
make -j 8 --no-print-directory -C src/hal/ESP8266
xtensa-lx106-elf-c++  -I../../../src -I../../../src/hal/ESP8266/ESP8266_RTOS_SDK -I../../../src/hal/ESP8266/ESP8266_RTOS_SDK/components/esp8266/include -I../../../src/hal/ESP8266/ESP8266_RTOS_SDK/components/freertos/include/freertos -I../../../src/hal/ESP8266/ESP8266_RTOS_SDK/components/freertos/include/freertos/private -I../../../src/hal/ESP8266/ESP8266_RTOS_SDK/components/freertos/port/esp8266/include -I../../../src/hal/ESP8266/ESP8266_RTOS_SDK/components/freertos/port/esp8266/include/freertos -I../../../src/hal/ESP8266/ESP8266_RTOS_SDK/components/heap/include -I../../../src/hal/ESP8266/ESP8266_RTOS_SDK/components/heap/port/esp8266/include -I. -O0 -g3 -Wall -ffunction-sections -fdata-sections -mlongcalls -mtext-section-literals -std=c++11 -fno-exceptions -fno-rtti -felide-constructors -fno-threadsafe-statics -fno-use-cxa-atexit -c
gpio/gpio.cpp -o ../../../out/obj/src/hal/ESP8266/gpio/gpio.o
make -j 8 --no-print-directory -C src/hal/ESP8266/ESP8266_RTOS_SDK
xtensa-lx106-elf-gcc-5.2.0 -DNDEBUG -DPARTITION_QUEUE_HEADER=\"sys/queue.h\" -DNVS_CRC_HEADER_FILE=\"crc.h\" -D__ets__ -DESPRESSIF_USE -DEMBEDDED_SUPP -DESP_PLATFORM -DIDF_VER=\"3.1\" -DESP8266 -DICACHE_FLASH -D_CLOCKS_PER_SEC_=CONFIG_FREERTOS_HZ -D_POSIX_THREADS=1 -D_UNIX98_THREAD_MUTEX_ATTRIBUTES=1 -I../../../../src -I../../../../src/hal/ESP8266/ESP8266_RTOS_SDK -I../../../../src/hal/ESP8266/ESP8266_RTOS_SDK/components/esp8266/include -I../../../../src/hal/ESP8266/ESP8266_RTOS_SDK/components/freertos/include/freertos -I../../../../src/hal/ESP8266/ESP8266_RTOS_SDK/components/freertos/include/freertos/private -I../../../../src/hal/ESP8266/ESP8266_RTOS_SDK/components/freertos/port/esp8266/include -I../../../../src/hal/ESP8266/ESP8266_RTOS_SDK/components/freertos/port/esp8266/include/freertos -I../../../../src/hal/ESP8266/ESP8266_RTOS_SDK/components/heap/include -I../../../../src/hal/ESP8266/ESP8266_RTOS_SDK/components/heap/port/esp8266/include -I. -Icomponents/nvs_flash/include -Icomponents/tcpip_adapter/include -Icomponents/lwip/lwip/src/include -Icomponents/lwip/port/esp8266/include -Icomponents/lwip/include/lwip/apps -Icomponents/log/include -Icomponents/bootloader_support/include -Icomponents/spi_flash/include -Icomponents/freertos/include -Icomponents/newlib/newlib/port/include -Icomponents/util/include -O0 -g3 -Wall -ffunction-sections -fdata-sections -mlongcalls -mtext-section-literals -Wno-error=unused-function -Wno-error=deprecated-declarations -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -std=c99 -std=gnu99 -fno-aggressive-loop-optimizations -Wno-implicit-function-declaration -c components\freertos\freertos\event_groups.c -o ../../../../out/obj/src/hal/ESP8266/ESP8266_RTOS_SDK/components\freertos\freertos\event_groups.o
xtensa-lx106-elf-gcc-5.2.0 -DNDEBUG -DPARTITION_QUEUE_HEADER=\"sys/queue.h\" -DNVS_CRC_HEADER_FILE=\"crc.h\" -D__ets__ -DESPRESSIF_USE -DEMBEDDED_SUPP -DESP_PLATFORM -DIDF_VER=\"3.1\" -DESP8266 -DICACHE_FLASH -D_CLOCKS_PER_SEC_=CONFIG_FREERTOS_HZ -D_POSIX_THREADS=1 -D_UNIX98_THREAD_MUTEX_ATTRIBUTES=1 -I../../../../src -I../../../../src/hal/ESP8266/ESP8266_RTOS_SDK -I../../../../src/hal/ESP8266/ESP8266_RTOS_SDK/components/esp8266/include -I../../../../src/hal/ESP8266/ESP8266_RTOS_SDK/components/freertos/include/freertos -I../../../../src/hal/ESP8266/ESP8266_RTOS_SDK/components/freertos/include/freertos/private -I../../../../src/hal/ESP8266/ESP8266_RTOS_SDK/components/freertos/port/esp8266/include -I../../../../src/hal/ESP8266/ESP8266_RTOS_SDK/components/freertos/port/esp8266/include/freertos -I../../../../src/hal/ESP8266/ESP8266_RTOS_SDK/components/heap/include -I../../../../src/hal/ESP8266/ESP8266_RTOS_SDK/components/heap/port/esp8266/include -I. -Icomponents/nvs_flash/include -Icomponents/tcpip_adapter/include -Icomponents/lwip/lwip/src/include -Icomponents/lwip/port/esp8266/include -Icomponents/lwip/include/lwip/apps -Icomponents/log/include -Icomponents/bootloader_support/include -Icomponents/spi_flash/include -Icomponents/freertos/include -Icomponents/newlib/newlib/port/include -Icomponents/util/include -O0 -g3 -Wall -ffunction-sections -fdata-sections -mlongcalls -mtext-section-literals -Wno-error=unused-function -Wno-error=deprecated-declarations -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -std=c99 -std=gnu99 -fno-aggressive-loop-optimizations -Wno-implicit-function-declaration -c components\freertos\freertos\list.c -o ../../../../out/obj/src/hal/ESP8266/ESP8266_RTOS_SDK/components\freertos\freertos\list.o
xtensa-lx106-elf-gcc-5.2.0 -DNDEBUG -DPARTITION_QUEUE_HEADER=\"sys/queue.h\" -DNVS_CRC_HEADER_FILE=\"crc.h\" -D__ets__ -DESPRESSIF_USE -DEMBEDDED_SUPP -DESP_PLATFORM -DIDF_VER=\"3.1\" -DESP8266 -DICACHE_FLASH -D_CLOCKS_PER_SEC_=CONFIG_FREERTOS_HZ -D_POSIX_THREADS=1 -D_UNIX98_THREAD_MUTEX_ATTRIBUTES=1 -I../../../../src -I../../../../src/hal/ESP8266/ESP8266_RTOS_SDK -I../../../../src/hal/ESP8266/ESP8266_RTOS_SDK/components/esp8266/include -I../../../../src/hal/ESP8266/ESP8266_RTOS_SDK/components/freertos/include/freertos -I../../../../src/hal/ESP8266/ESP8266_RTOS_SDK/components/freertos/include/freertos/private -I../../../../src/hal/ESP8266/ESP8266_RTOS_SDK/components/freertos/port/esp8266/include -I../../../../src/hal/ESP8266/ESP8266_RTOS_SDK/components/freertos/port/esp8266/include/freertos -I../../../../src/hal/ESP8266/ESP8266_RTOS_SDK/components/heap/include -I../../../../src/hal/ESP8266/ESP8266_RTOS_SDK/components/heap/port/esp8266/include -I. -Icomponents/nvs_flash/include -Icomponents/tcpip_adapter/include -Icomponents/lwip/lwip/src/include -Icomponents/lwip/port/esp8266/include -Icomponents/lwip/include/lwip/apps -Icomponents/log/include -Icomponents/bootloader_support/include -Icomponents/spi_flash/include -Icomponents/freertos/include -Icomponents/newlib/newlib/port/include -Icomponents/util/include -O0 -g3 -Wall -ffunction-sections -fdata-sections -mlongcalls -mtext-section-literals -Wno-error=unused-function -Wno-error=deprecated-declaratio [... cut ...]
D:\dev\sw\embedded\proj\omef-esp8266\src\hal\ESP8266\ESP8266_RTOS_SDK/components\esp8266\source/startup.c:57: undefined reference to `esp_wifi_set_rx_pbuf_mem_type'
collect2.exe: error: ld returned 1 exit status
make[1]: *** [makefile:112: out/bin/omef-esp8266.elf] Error 1
make: *** [makefile:38: all] Error 2
```
</details>

I noticed that some of mentioned symbols are defined for ESP32 in rom_32.ld. But I didn't find such definitions for ESP8266 in rom_8266.ld.

## Could you please help me to solve this?
## Where is the definition of these symbols?

27-301, 2019-02-20:
seems that you use your own makefile? and only use some source code, header file and library of SDK?

r44083, 2019-02-20:
SDK doesn’t contain definitions for such symbols. You can try to search mentioned names. SDK libs also don’t have it.

27-301, 2019-02-21:
These functions are really included in the libs, such as wDev_ProcessFiq is in libpp.a. Can you check your build system, or provide a simple project for us to reproduce? thanks.

r44083, 2019-02-21:
Hi, after I add ```-Wl,--gc-sections```, some errors have been fixed.

So, the actual list of undefined symbols is:
```wDev_ProcessFiq```
```NMIIrqIsOn```
```WDEV_INTEREST_EVENT```
```MacIsrSigPostDefHdl```
```pmIdleHook```
```CCOMPARE_AddVal```
```xthal_set_intset```
```esp_random```

Here is the link to the simple project:
https://drive.google.com/file/d/1v7ZFYC5Qmnorsm82iMYByNkhows-_UYE/view?usp=sharing

It is already builded, to rebuild it just run in its root directory:
```make clean```
```make```

<details>
<summary>New build log</b></summary>

```
make -j 8 --no-print-directory -C src
xtensa-lx106-elf-c++  -I../src -I../src/hal/ESP8266 -I../src/hal/ESP8266/ESP8266_RTOS_SDK -I../src/hal/ESP8266/ESP8266_RTOS_SDK/components/esp8266/include -I../src/hal/ESP8266/ESP8266_RTOS_SDK/components/freertos/include/freertos -I../src/hal/ESP8266/ESP8266_RTOS_SDK/components/freertos/include/freertos/private -I../src/hal/ESP8266/ESP8266_RTOS_SDK/components/freertos/port/esp8266/include -I../src/hal/ESP8266/ESP8266_RTOS_SDK/components/freertos/port/esp8266/include/freertos -I../src/hal/ESP8266/ESP8266_RTOS_SDK/components/heap/include -I../src/hal/ESP8266/ESP8266_RTOS_SDK/components/heap/port/esp8266/include -O0 -g3 -Wall -ffunction-sections -fdata-sections -mlongcalls -mtext-section-literals -std=c++11 -fno-exceptions -fno-rtti -felide-constructors -fno-threadsafe-statics -fno-use-cxa-atexit -c main.cpp -o ../out/obj/src/main.o
make -j 8 --no-print-directory -C src/common
xtensa-lx106-elf-gcc-5.2.0  -I../../src -I../../src/hal/ESP8266 -I../../src/hal/ESP8266/ESP8266_RTOS_SDK -I../../src/hal/ESP8266/ESP8266_RTOS_SDK/components/esp8266/include -I../../src/hal/ESP8266/ESP8266_RTOS_SDK/components/freertos/include/freertos -I../../src/hal/ESP8266/ESP8266_RTOS_SDK/components/freertos/include/freertos/private -I../../src/hal/ESP8266/ESP8266_RTOS_SDK/components/freertos/port/esp8266/include -I../../src/hal/ESP8266/ESP8266_RTOS_SDK/components/freertos/port/esp8266/include/freertos -I../../src/hal/ESP8266/ESP8266_RTOS_SDK/components/heap/include -I../../src/hal/ESP8266/ESP8266_RTOS_SDK/components/heap/port/esp8266/include -O0 -g3 -Wall -ffunction-sections -fdata-sections -mlongcalls -mtext-section-literals -std=c99 -c assert.c -o ../../out/obj/src/common/assert.o
make -j 8 --no-print-directory -C src/hal/ESP8266
xtensa-lx106-elf-c++  -I../../../src -I../../../src/hal/ESP8266 -I../../../src/hal/ESP8266/ESP8266_RTOS_SDK -I../../../src/hal/ESP8266/ESP8266_RTOS_SDK/components/esp8266/include -I../../../src/hal/ESP8266/ESP8266_RTOS_SDK/components/freertos/include/freertos -I../../../src/hal/ESP8266/ESP8266_RTOS_SDK/components/freertos/include/freertos/private -I../../../src/hal/ESP8266/ESP8266_RTOS_SDK/components/freertos/port/esp8266/include -I../../../src/hal/ESP8266/ESP8266_RTOS_SDK/components/freertos/port/esp8266/include/freertos -I../../../src/hal/ESP8266/ESP8266_RTOS_SDK/components/heap/include -I../../../src/hal/ESP8266/ESP8266_RTOS_SDK/components/heap/port/esp8266/include -I. -O0 -g3 -Wall -ffunction-sections -fdata-sections -mlongcalls -mtext-section-literals -std=c++11 -fno-exceptions -fno-rtti -felide-constructors -fno-threadsafe-statics -fno-use-cxa-atexit -c gpio/gpio.cpp -o ../../../out/obj/src/hal/ESP8266/gpio/gpio.o
make -j 8 --no-print-directory -C src/hal/ESP8266/ESP8266_RTOS_SDK
xtensa-lx106-elf-gcc-5.2.0 -DNDEBUG -DPARTITION_QUEUE_HEADER=\"sys/queue.h\" -DNVS_CRC_HEADER_FILE=\"crc.h\" -DESP_PLATFORM -DIDF_VER=\"3.1\" -DESP8266 -DICACHE_FLASH -I../../../../src -I../../../../src/hal/ESP8266 -I../../../../src/hal/ESP8266/ESP8266_RTOS_SDK -I../../../../src/hal/ESP8266/ESP8266_RTOS_SDK/components/esp8266/include -I../../../../src/hal/ESP8266/ESP8266_RTOS_SDK/components/freertos/include/freertos -I../../../../src/hal/ESP8266/ESP8266_RTOS_SDK/components/freertos/include/freertos/private -I../../../../src/hal/ESP8266/ESP8266_RTOS_SDK/components/freertos/port/esp8266/include -I../../../../src/hal/ESP8266/ESP8266_RTOS_SDK/components/freertos/port/esp8266/include/freertos -I../../../../src/hal/ESP8266/ESP8266_RTOS_SDK/components/heap/include -I../../../../src/hal/ESP8266/ESP8266_RTOS_SDK/components/heap/port/esp8266/include -I. -Icomponents/nvs_flash/include -Icomponents/tcpip_adapter/include -Icomponents/lwip/lwip/src/include -Icomponents/lwip/port/esp8266/include -Icomponents/lwip/include/lwip/apps -Icomponents/log/include -Icomponents/bootloader_support/include -Icomponents/spi_flash/include -Icomponents/freertos/include -Icomponents/newlib/newlib/port/include -Icomponents/util/include -O0 -g3 -Wall -ffunction-sections -fdata-sections -mlongcalls -mtext-section-literals -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wno-unused-parameter -Wno-sign-compare -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -mtext-section-literals -mlongcalls -nostdlib -std=c99 -std=gnu99 -c components\freertos\freertos\event_groups.c -o ../../../../out/obj/src/hal/ESP8266/ESP8266_RTOS_SDK/components\freertos\freertos\event_groups.o
xtensa-lx106-elf-gcc-5.2.0 -DNDEBUG -DPARTITION_QUEUE_HEADER=\"sys/queue.h\" -DNVS_CRC_HEADER_FILE=\"crc.h\" -DESP_PLATFORM -DIDF_VER=\"3.1\" -DESP8266 -DICACHE_FLASH -I../../../../src -I../../../../src/hal/ESP8266 -I../../../../src/hal/ESP8266/ESP8266_RTOS_SDK -I../../../../src/hal/ESP8266/ESP8266_RTOS_SDK/components/esp8266/include -I../../../../src/hal/ESP8266/ESP8266_RTOS_SDK/components/freertos/include/free [... cut ...]
D:\dev\sw\embedded\proj\omef-esp8266\src\hal\ESP8266\ESP8266_RTOS_SDK/components\esp8266\source/esp_wifi_os_adapter.c:326: undefined reference to `esp_random'
out/obj/src/hal/ESP8266/ESP8266_RTOS_SDK/components\esp8266\source\esp_wifi_os_adapter.o: In function `rand_wrapper':
D:\dev\sw\embedded\proj\omef-esp8266\src\hal\ESP8266\ESP8266_RTOS_SDK/components\esp8266\source/esp_wifi_os_adapter.c:335: undefined reference to `esp_random'
collect2.exe: error: ld returned 1 exit status
make[1]: *** [makefile:112: out/bin/omef-esp8266.elf] Error 1
make: *** [makefile:38: all] Error 2
```

</details>

27-301, 2019-02-22:
something wrong with your make system, seems that the libs are not included correctly, so the link cannot link these symbols. if `LIB:= `, the error is same.

can you follow up the SDK framework to build your application?

r44083, 2019-02-22:
Thank you Gargamel, it is really strange that linker doesn't use libs which I pass to it.
I will try to solve this.

Unfortunately I can't use default SDK build system on regular basis since I need to integrate ESP8266_RTOS_SDK into the existing infrastructure. I will try to use default build system of SDK only for solving issue during SDK integration.

If you have such possibility, could you please attach build and linkage log from original build system of ESP8266_RTOS_SDK?

r44083, 2019-02-25:
Hi Gargamel,
The problem was that I passed the libs to the linker before objects files. I have fixed this, so build is almost successful.

The last issue:
```
out/obj/src/hal/ESP8266/ESP8266_RTOS_SDK/components\freertos\port\esp8266\xtensa_vectors.o: In function `_panic_dump_reg':
(.text+0x487): dangerous relocation: call0: call target out of range: Cache_Read_Enable_New
```

Build is successful, If I comment the following line 1481 in file "components\freertos\port\esp8266\xtensa_vectors.S"
```
    call0   Cache_Read_Enable_New
```
_(-m longcalls doesn't help)_
Do you have some hints how to solve this?

r44083, 2019-04-14:
Fixed using patch below:
```diff
diff --git a/components/esp8266/source/startup.c b/components/esp8266/source/startup.c
index 3770eb4d..7f70d040 100644
--- a/components/esp8266/source/startup.c
+++ b/components/esp8266/source/startup.c
@@ -73,7 +73,7 @@ static void user_init_entry(void *param)
 
     phy_get_bb_evm();
 
-    assert(nvs_flash_init() == 0);
+    //assert(nvs_flash_init() == 0);
     assert(wifi_nvs_init() == 0);
     assert(rtc_init() == 0);
     assert(mac_init() == 0);
@@ -104,8 +104,11 @@ static void user_init_entry(void *param)
     vTaskDelete(NULL);
 }
 
-void call_user_start(size_t start_addr)
+__attribute__((section(".iram1"))) void call_user_start(size_t start_addr)
 {
+    Cache_Read_Enable(0, 0, 0);
+    start_addr = 0;
+    
     int i;
     int *p;
 
diff --git a/components/spi_flash/src/spi_flash_raw.c b/components/spi_flash/src/spi_flash_raw.c
index 9fc3afca..df8b6557 100644
--- a/components/spi_flash/src/spi_flash_raw.c
+++ b/components/spi_flash/src/spi_flash_raw.c
@@ -21,21 +21,21 @@
 #include "esp8266/pin_mux_register.h"
 #include "esp8266/spi_register.h"
 
-void Cache_Read_Disable_2(void)
+__attribute__((section(".iram1"))) void Cache_Read_Disable_2(void)
 {
     CLEAR_PERI_REG_MASK(CACHE_FLASH_CTRL_REG,CACHE_READ_EN_BIT);
     while(REG_READ(SPI_EXT2(0)) != 0) { }
     CLEAR_PERI_REG_MASK(PERIPHS_SPI_FLASH_CTRL,SPI_ENABLE_AHB);
 }
 
-void Cache_Read_Enable_2()
+__attribute__((section(".iram1"))) void Cache_Read_Enable_2()
 {
     SET_PERI_REG_MASK(PERIPHS_SPI_FLASH_CTRL,SPI_ENABLE_AHB);
     SET_PERI_REG_MASK(CACHE_FLASH_CTRL_REG,CACHE_READ_EN_BIT);
 }
 void Cache_Read_Enable_New(void) __attribute__((alias("Cache_Read_Enable_2")));
 
-uint32_t spi_flash_get_id_raw(esp_spi_flash_chip_t *chip)
+__attribute__((section(".iram1"))) uint32_t spi_flash_get_id_raw(esp_spi_flash_chip_t *chip)
 {
     uint32_t rdid = 0;
 
@@ -105,7 +105,7 @@ esp_err_t spi_flash_erase_sector_raw(esp_spi_flash_chip_t *chip, size_t sec, siz
     return ret;
 }
 
-esp_err_t spi_flash_enable_qmode_raw(esp_spi_flash_chip_t *chip)
+__attribute__((section(".iram1"))) esp_err_t spi_flash_enable_qmode_raw(esp_spi_flash_chip_t *chip)
 {
     esp_err_t ret;
 

```

closed 2019-04-14 by r44083