espressif/ESP8266_RTOS_SDK #494: Strange pulse on GPIO0 when booting device.

nghiaquy1991, 2019-02-23:
## Environment

- Development Kit: ESP8266 12F
- IDF version (``git rev-parse --short HEAD`` to get the commit id.): 
    7d4782f
- Development Env:     Make
- Operating System:    Ubuntu
- Power Supply:         external 5V

## Problem Description
I make an application which connected GPIO0 with a led. I use deep sleep function.
I saw that after wakeup from the deep sleep state, it always have a strange pulse on GPIO 0 so the led was blinking.  
This is my init code:
``` 
   #define LED_PIN                       (0)
   #define LED_PIN_BIT                 GPIO_Pin_0
   #define LED_MUX                     PERIPHS_IO_MUX_GPIO0_U
   #define LED_FUNC                    FUNC_GPIO0
   
    GPIO_ConfigTypeDef io_conf;
    // Init Led pin control
    // PIN_FUNC_SELECT(LED_MUX, LED_FUNC);
    io_conf.GPIO_IntrType = GPIO_PIN_INTR_DISABLE;
    io_conf.GPIO_Mode = GPIO_Mode_Output;
    io_conf.GPIO_Pin = LED_PIN_BIT;
    io_conf.GPIO_Pullup = GPIO_PullUp_DIS;
    gpio_config(&io_conf);
    GPIO_OUTPUT_SET(GPIO_ID_PIN(LED_PIN), 1);
```
I am using boot_v1.7
[boot_v1.7_low_power.zip](https://github.com/espressif/ESP8266_RTOS_SDK/files/2896611/boot_v1.7_low_power.zip)

### Actual Behavior
[image]
[image]

This issue decreases the lifetime of the device using a battery. So could you help me to fix it?

nghiaquy1991, 2019-02-25:
@xiongyumail Yes, It has pulses when powering up.

xiongyumail, 2019-03-12:
```
#define PERIPHS_IO_MUX_GPIO0_U          (PERIPHS_IO_MUX + 0x34)
#define FUNC_GPIO0                          0
#define FUNC_SPICS2                         1
#define FUNC_CLK_OUT                        4
```
IOMUX of IO0 defaults to clock output mode when ESP8266 is powered on. Please do not use this IO to connect to LED.

closed 2019-03-12 by nghiaquy1991