espressif/ESP8266_RTOS_SDK #498: DHCP server problem

filzek, 2019-02-26:
Latest master branch as of  02/26/2019

The DHCP current drop the lease and stop to provide IPs to the client devices.
Tested with android and iphone, same problem. 

If IP is granted it takes 10 seconds and ips is removed by a udp clean.

included files:
#include <ctype.h>
#include <fcntl.h>
#include <limits.h>
#include <netdb.h>
#include <stdio.h>
#include <string.h>
#include <stdlib.h>
#include <time.h>
#include <unistd.h>

#include <sys/types.h>
#include <sys/stat.h>
#include <sys/param.h>
#include <sys/socket.h>

#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/queue.h"
#include "freertos/event_groups.h"
#include "freertos/semphr.h"

#include "driver/gpio.h"
#include "driver/hw_timer.h"
#include "rom/ets_sys.h"

#include "esp_wifi.h"
#include "esp_event_loop.h"
#include "esp_log.h"
#include "esp_ota_ops.h"
#include "esp_spiffs.h"
#include "esp_spi_flash.h"
#include "esp_ssc.h"
#include "esp_system.h"
#include "esp_wifi.h"

#include "esp8266/efuse_register.h"
#include "esp8266/eagle_soc.h"

#include "spiffs.h"
#include "spiffs_nucleus.h"

#include "nvs.h"
#include "nvs_flash.h"
#include "tcpip_adapter.h"

#include "jsmn.h"

//****** MBED *******
#include "mbedtls/platform.h"
#include "mbedtls/net_sockets.h"
#include "mbedtls/esp_debug.h"
#include "mbedtls/ssl.h"
#include "mbedtls/entropy.h"
#include "mbedtls/ctr_drbg.h"
#include "mbedtls/error.h"
#include "mbedtls/certs.h"

//****** LWIP *******
#include "lwip/err.h"
#include "lwip/netdb.h"
#include "lwip/sockets.h"
#include "lwip/sys.h"

//******** server httpd ***********
#include <esp_http_server.h>

#include "lwip/apps/sntp.h"

LOG OUTPUT:

2019-02-26 20:04:27.379:  add 1
2019-02-26 20:04:27.381:  aid 1
2019-02-26 20:04:27.383:  station: 00:f4:6f:cc:fe:e9 join, AID = 1
2019-02-26 20:04:27.386:  D (41623) event: SYSTEM_EVENT_AP_STACONNECTED, mac:00:f4:6f:cc:fe:e9, aid:1
2019-02-26 20:04:27.392:  I (41633) Testing: station:00:f4:6f:cc:fe:e9 join, AID=1
2019-02-26 20:04:29.010:  add 2
2019-02-26 20:04:29.012:  aid 2
2019-02-26 20:04:29.014:  station: 0c:30:21:0b:0a:de join, AID = 2
2019-02-26 20:04:29.020:  D (43253) event: SYSTEM_EVENT_AP_STACONNECTED, mac:0c:30:21:0b:0a:de, aid:2
2019-02-26 20:04:29.027:  I (43263) Testing: station:0c:30:21:0b:0a:de join, AID=2
2019-02-26 20:04:58.224:  station: 00:f4:6f:cc:fe:e9 leave, AID = 1
2019-02-26 20:04:58.228:  rm 1
2019-02-26 20:04:58.230:  D (72443) event: SYSTEM_EVENT_AP_STADISCONNECTED, mac:00:f4:6f:cc:fe:e9, aid
:1
2019-02-26 20:04:58.237:  I (72453) Testing: station:00:f4:6f:cc:fe:e9leave, AID=1

And the device keep awaiting forever for the IP, sometimes it does receive an IP for 10 seconds and the IP is revoked.

PROBLEM ARE RELATED TO esp_http_server.h, without it the problem never show.

filzek, 2019-02-26:
This keep going forever, same behavior with the new provisioning example.

 station:00:f4:6f:cc:fe:e9leave, AID=1
I (80492) tcpip_adapter: softAP assign IP to station,IP is: 192.168.4.3
I (81492) tcpip_adapter: softAP assign IP to station,IP is: 192.168.4.3
I (83502) tcpip_adapter: softAP assign IP to station,IP is: 192.168.4.3
I (87682) tcpip_adapter: softAP assign IP to station,IP is: 192.168.4.3
I (96542) tcpip_adapter: softAP assign IP to station,IP is: 192.168.4.3
I (97522) tcpip_adapter: softAP assign IP to station,IP is: 192.168.4.3
I (99732) tcpip_adapter: softAP assign IP to station,IP is: 192.168.4.3
I (103542) tcpip_adapter: softAP assign IP to station,IP is: 192.168.4.3
I (112592) tcpip_adapter: softAP assign IP to station,IP is: 192.168.4.3
I (113592) tcpip_adapter: softAP assign IP to station,IP is: 192.168.4.3
I (115602) tcpip_adapter: softAP assign IP to station,IP is: 192.168.4.3
I (119572) tcpip_adapter: softAP assign IP to station,IP is: 192.168.4.3

FayeY, 2019-04-17:
@Junhao-Espressif  Any comments?

closed 2019-05-22 by filzek

umitech, 2020-01-18:
Exactly the same symptom here. Any solution for this problem? I tested with the latest master branch and also the 3.3 release.

umitech, 2020-01-21:
@filzek As you closed this issue, did you find any solution for this problem?

umitech, 2020-02-08:
We tested another dhcp server code on this SDK but the problem exists. May be the problem comes from the lwip udp. Any idea?