espressif/ESP8266_RTOS_SDK #499: call_user_start placed into irom instead of iram

r44083, 2019-02-28:
## Environment
- Development Kit: ESP8266 dev board
- IDF version: ESP8266_RTOS_SDK v3.1
- Development Env: Make
- Operating System: Windows 10
- Power Supply: USB

I have noticed that entry point functions is placed to irom instead of iram.

My app is crashed because ESP8266 can't load ```call_user_start``` from irom on boot. If I add ```IRAM_ATTR``` before ```call_user_start``` definition, app doesn't crash anymore, ```call_user_start``` is executed, but app doesn't start properly.
I don't use 2nd bootloader, so my app starts from 0x40200000.

**Why ```call_user_start``` is placed to the irom?**

donghengqaz, 2019-03-06:
call_user_start shoud unpacks the IRAM functions. So if you link call_user_start to IRAM, no one can unpack the IRAM function.

r44083, 2019-03-06:
But you can not call funtions from irom before SPI Flash initialization.
My app starts from call_user_star() which placed to irom. How can you call call_user_star() on boot from SPI Flash?

donghengqaz, 2019-03-13:
Bootloader will initialize the SPI flash before jumping to user APP.

r44083, 2019-03-13:
User app has the following config in menuconfig:
```CONFIG_BOOTLOADER_INIT_SPI_FLASH```
I set it to 0. So App should init SPI flash itself.

r44083, 2019-04-14:
Fix is:
1. Call ```Cache_Read_Enable(0, 0, 0);``` at the beginning of call_user_start()
2. Move call_user_start() to iram by adding ```__attribute__((section(".iram1")))``` to ```call_user_start()```

closed 2019-04-14 by r44083