espressif/ESP8266_RTOS_SDK #500: esp_register_freertos_idle_hook & esp_register_freertos_tick_hook don't work.

nopnop2002, 2019-03-01:
esp_register_freertos_idle_hook & esp_register_freertos_tick_hook don't work.

	#include "freertos/FreeRTOS.h"
	#include "task.h"
	#include "esp_log.h"
	#include "esp_freertos_hooks.h"

	long idle_count = 0;
	long tick_count = 0;

	void ConsumptionTick(int delay) {
	    TickType_t startTick;
	    TickType_t endTick;
	    TickType_t nowTick;
	    startTick = xTaskGetTickCount();
	    endTick = startTick + delay;
	    //ESP_LOGI(pcTaskGetName(0),"startTick=%d endTick=%d\n",startTick,endTick);
	    while(1) {
	      nowTick = xTaskGetTickCount();
	      if (nowTick > endTick) break;
	    }
	}

	bool ApplicationIdleHook(void)
	{
	    idle_count++;
	    return true;
	}

	void ApplicationTickHook(void)
	{
	    tick_count++;
	}

	void clearHook() {
	    idle_count = 0;
	    tick_count = 0;
	}

	void printHook(char * title) {
	    ESP_LOGI(pcTaskGetName(0),"%s",title);
	    ESP_LOGI(pcTaskGetName(0),"idle_count=%ld tick_count=%ld",idle_count,tick_count);
	}

	void app_main(void)
	{
	    ESP_LOGI(pcTaskGetName(0),"start");

	    esp_register_freertos_idle_hook(&ApplicationIdleHook);
	    esp_register_freertos_tick_hook(&ApplicationTickHook);

	    clearHook();
	    ConsumptionTick(200);
	    printHook("Busy 200");

	    clearHook();
	    vTaskDelay(200);
	    printHook("Delay 200");
	}

27-301, 2019-03-05:
we are checking about this.

closed 2019-03-08 by wujiangang

27-301, 2019-03-08:
@nopnop2002 please pull the master, use menuconfig Component config -> FreeRTOS -> Use FreeRTOS extened hooks to enable the hook firstly, then retry. thanks