espressif/ESP8266_RTOS_SDK #501: espos_timer_start() 导致系统反复重启

huozhe0930, 2019-03-03:
涉及的源代码（节选）如下：

static const char *timer_name = "sensor";
espos_timer_t *timer;

void app_main(void)
{
	
    printf("SDK version:%s\n", esp_get_idf_version());
	
    esp_reset_reason_t reset_reason;
	reset_reason = esp_reset_reason();
	
	i2c_master_tpm112_init(I2C_MASTER_NUM);

	espos_timer_create(timer, timer_name, sensor_task, NULL, 100, ESPOS_TIMER_AUTO_RUN);
	espos_timer_start(*timer);

	//start i2c sensor task
    //xTaskCreate(sensor_task, "sensor_task", 2048, NULL, 10, NULL);
}

描述：
在改成定时任务前，sensor_task可以正常运行，改为定时任务后，去掉了sensor_task的死循环
vTaskDelay(1000 / portTICK_RATE_MS); 其他不变！
然后运行到espos_timer_start(*timer)，就死掉了，等待一会儿就重启了。

串口监控输出如下：
MONITOR
--- idf_monitor on COM3 74880 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
2019-03-03 01:06:33.570:
2019-03-03 01:06:33.578:   ets Jan  8 2013,rst cause:2, boot mode:(3,7)
2019-03-03 01:06:33.604:
2019-03-03 01:06:33.620:  load 0x40100000, len 7188, room 16
0x40100000: _stext at ??:?

2019-03-03 01:06:33.751:  tail 4
2019-03-03 01:06:33.769:  chksum 0x0b
2019-03-03 01:06:33.784:  load 0x3ffe8408, len 24, room 4
2019-03-03 01:06:33.811:  tail 4
2019-03-03 01:06:33.821:  chksum 0xe2
2019-03-03 01:06:33.841:  load 0x3ffe8420, len 3548, room 4
2019-03-03 01:06:33.870:  tail 8
2019-03-03 01:06:33.880:  chksum 0xa6
I (43) boot: ESP-IDF v3.2-dev-106-g4372b84 2nd stage bootloader
2019-03-03 01:06:33.935:  I (43) boot: compile time 22:32:32
2019-03-03 01:06:33.961:  I (43) boot: SPI Speed      : 40MHz
2019-03-03 01:06:33.984:  I (50) boot: SPI Mode       : DIO
2019-03-03 01:06:34.009:  I (56) boot: SPI Flash Size : 4MB
2019-03-03 01:06:34.046:  I (62) boot: Partition Table:
2019-03-03 01:06:34.073:  I (68) boot: ## Label            Usage          Type ST Offset   Length
2019-03-03 01:06:34.125:  I (79) boot:  0 nvs              WiFi data        01 02 00009000 00006000
2019-03-03 01:06:34.171:  I (90) boot:  1 phy_init         RF data          01 01 0000f000 00001000
2019-03-03 01:06:34.209:  I (102) boot:  2 factory          factory app      00 00 00010000 000f0000
2019-03-03 01:06:34.258:  I (113) boot: End of partition table
2019-03-03 01:06:34.289:  I (120) esp_image: segment 0: paddr=0x00010010 vaddr=0x40210010 size=0x32140 (205120) map
2019-03-03 01:06:34.456:  I (219) esp_image: segment 1: paddr=0x00042158 vaddr=0x3ffe8000 size=0x0053c (  1340) load
2019-03-03 01:06:34.526:  I (220) esp_image: segment 2: paddr=0x0004269c vaddr=0x3ffe853c size=0x0011c (   284) load
2019-03-03 01:06:34.586:  I (230) esp_image: segment 3: paddr=0x000427c0 vaddr=0x40100000 size=0x059e0 ( 23008) load
0x40100000: _stext at ??:?

2019-03-03 01:06:34.757:  I (252) boot: Loaded app from partition at offset 0x10000
2019-03-03 01:06:34.800:  I (298) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
2019-03-03 01:06:34.871:  I (308) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
2019-03-03 01:06:34.951:  I (508) reset_reason: RTC reset 2 wakeup 0 store 0, reason is 2
2019-03-03 01:06:35.014:  SDK version:v3.2-dev-106-g4372b84
2019-03-03 01:06:35.040:  I (618) gpio: GPIO[14]| InputEn: 0| OutputEn: 1| OpenDrain: 1| Pullup: 1| Pulldown: 0| Intr:0
2019-03-03 01:06:35.110:  I (618) gpio: GPIO[2]| InputEn: 0| OutputEn: 1| OpenDrain: 1| Pullup: 1| Pulldown: 0| Intr:0
2019-03-03 01:06:35.173:
[... 132 lines cut ...]
  28         2c         30         34         38         3c
2019-03-03 01:07:07.889:
2019-03-03 01:07:07.976:  3ffef180  0x4023852c 0x00000020 0x3ffe8650 0x00000000 0x00000000 0x00000000 0x3ffee3bc 0x40236e4d 0x00000000 0x00000000 0x3ffe
8650 0x40226fd8 0x00000000 0x00000000 0x00000000 0x00000000
0x4023852c: __swrite at /home/dongheng/workspace/espressif/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-lx106-elf/newlib/libc/stdio/../../../.././
newlib/libc/stdio/stdio.c:81

0x40236e4d: app_main at D:/msys32/home/huoda/project/i2c/main/user_main.c:300

0x40226fd8: user_init_entry at D:/msys32/home/huoda/ESP8266_RTOS_SDK/components/esp8266/source/startup.c:86

2019-03-03 01:07:09.880:  3ffef1c0  0x00000000 0x00000000
2019-03-03 01:07:10.061:
2019-03-03 01:07:10.179:
2019-03-03 01:07:10.267:          PC: 0x40236e4f        PS: 0x00000030        A0: 0x40236e4d        A1: 0x3ffef1a0
0x40236e4f: app_main at D:/msys32/home/huoda/project/i2c/main/user_main.c:300

0x40236e4d: app_main at D:/msys32/home/huoda/project/i2c/main/user_main.c:300

2019-03-03 01:07:11.062:          A2: 0x00000000        A3: 0x00000001        A4: 0x00000000        A5: 0x00000000
2019-03-03 01:07:11.447:          A6: 0x00000064        A7: 0x00000001        A8: 0x3ffef170        A9: 0x00000090
2019-03-03 01:07:11.825:         A10: 0x402262e2       A11: 0x00000020       A12: 0x3ffee3bc       A13: 0x00000000
0x402262e2: chip_v6_unset_chanfreq at ??:?

2019-03-03 01:07:12.361:         A14: 0x00000000       A15: 0x00000000       SAR: 0x0000001e  EXCCAUSE: 0x0000001c

huozhe0930, 2019-03-03:
改用freertos的 xTimerCreate(),工作的很好！

donghengqaz, 2019-03-06:
用法不对：

static const char *timer_name = "sensor";
espos_timer_t timer;

void app_main(void)
{

printf("SDK version:%s\n", esp_get_idf_version());

esp_reset_reason_t reset_reason;
reset_reason = esp_reset_reason();

i2c_master_tpm112_init(I2C_MASTER_NUM);

espos_timer_create(&timer, timer_name, sensor_task, NULL, 100, ESPOS_TIMER_AUTO_RUN);
espos_timer_start(timer);

//start i2c sensor task
//xTaskCreate(sensor_task, "sensor_task", 2048, NULL, 10, NULL);

}

huozhe0930, 2019-03-09:
谢谢，我再测试一下。

donghengqaz, 2019-03-13:
可以了吗？

closed 2019-03-18 by huozhe0930