espressif/ESP8266_RTOS_SDK #502: NVS Read nvs_get_blob always returns ESP_ERR_NVS_NOT_FOUND

xyx0826, 2019-03-03:
## Environment
- Development Kit: **NodeMCU**
- IDF version: **4372b84**
- Development Env:      **Make**
- Operating System:     **Ubuntu** (WSL Ubuntu 18.04.1 LTS)
- Power Supply:         **USB**

## Problem Description
When reading a key to a file from NVS within a valid namespace, NVS will return "file not found" instead of the corresponding value blob.

### Expected Behavior
NVS will retrieve the blob value to the specified key and return it as a string.

### Actual Behavior
`nvs_get_blob()` returns `ESP_ERR_NVS_NOT_FOUND`.

### Steps to repropduce
0. `make -j8 erase_flash flash` to wipe flash and flash the code
1. `python ~/esp/ESP8266_RTOS_SDK/components/nvs_flash/nvs_partition_generator/nvs_partition_gen.py --input ./nvs.csv --output nvs.bin --size 0x6000` to generate NVS partition blob
2. `python ~/esp/ESP8266_RTOS_SDK/components/esptool_py/esptool/esptool.py --chip esp8266 --port /dev/ttyS4 --baud 115200 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size 2MB 0x9000 /home/xyx0826/esp/projects/lamp/nvs.bin` to flash NVS partition blob
3. For whatever file requested, the function below will return empty, meaning that the specified key does not exist.

### Code to reproduce this issue

`nvs.csv`:
```
key,type,encoding,value
lamp,namespace,,
index.html,file,string,/path/to/index.html
index.css,file,string,/path/to/index.css
index.js,file,string,/path/to/index.js
response_ok.json,file,string,/path/to/response_ok.json
response_cooldown.json,file,string,/path/to/response_cooldown.json
```
`partitions.csv`:
```
# Espressif ESP32 Partition Table
# Name,   Type, SubType, Offset,  Size
nvs,      data, nvs,     0x9000,  0x6000
phy_init, data, phy,     0xf000,  0x1000
factory,  app,  factory, 0x10000, 1M
```
[file.c](https://gist.github.com/xyx0826/1cf8ab32682d13d1d33abc83f8136013)
## Debug Logs
```
I (3127) File: Requested to read file index.html.
D (3127) nvs: nvs_open_from_partition lamp 0
D (3137) nvs: nvs_get_str_or_blob index.html
```

## Other items if possible
- [x] [sdkconfig file](https://github.com/espressif/ESP8266_RTOS_SDK/files/2922677/sdkconfig.zip)
- [x] [elf file in the ``build`` folder](https://github.com/espressif/ESP8266_RTOS_SDK/files/2922678/lamp.zip)

supreetd21, 2019-03-04:
Hello @xyx0826 ,
Thank you for the very well-documented query.

In the nvs.csv file, It can be seen that the encoding type you have used is the _string_ type.
In the file.c, the NVS API used is _nvs_get_blob()_ which is used to retrieve values of the _binary_ encoding in the csv file. 
Assuming you have called _nvs_init()_ before the NVS calls, could you try changing the encoding type in the nvs.csv to _binary_ or keep the encoding type as _string_ and change the NVS API to _nvs_get_str()_.

xyx0826, 2019-03-04:
@supreetd21, Thank you for your reply.

I've tried replacing all `string` encoding to `blob` in `nvs.csv`, but the partition builder throw an `blob: Unsupported encoding` error.

I'll try using `nvs_get_str()` in my code and report back.

supreetd21, 2019-03-05:
Hello @xyx0826 , 
The encoding type that you should try using is _binary_. Blob isn't a supported encoding type.
Could you try replacing _string_  by _binary_ and then retrieve the data using _nvs_get_blob()_.

Also, do report if you have issues in retrieving through _nvs_get_str()_ when the encoding type is set to _string_.

xyx0826, 2019-03-05:
Hi @supreetd21:

The `binary` encoding worked. I can confirm `nvs_get_str()` works with `string` encoding type, so does `nvs_get_blob()` with `binary`. Closing this issue, thank you very much for your help.

Also: the partition script in this repo does not support multipage blobs, while the one from ESP-IDF does. Is ESP8266 SDK compatible with multipage blob NVS partitions?

closed 2019-03-05 by xyx0826

supreetd21, 2019-03-06:
Hello @xyx0826 ,
Glad that it solved the issue!

You're right, ESP-IDF supports multipage blobs but ESP8266_RTOS_SDK does not.
Thus, the partition generator utility in ESP8266_RTOS_SDK doesn't have it's support.