espressif/ESP8266_RTOS_SDK #506: wifi smartconfig 故障,在进行smartconfig第一次正确配置并连接之后,修改路由器密码,第二次进入smartconfig,不配置wifi参数,使模块超时退出,模块栈溢出

1XiaSheng, 2019-03-11:
SYNC: 0_66_66_66
SYNC: 66_66_66_66
SYNC: 66_66_66_54
SYNC: 0_0_66_66
[3561][app_uart.c][ 357]uart rcv:33,[+HRMADC:1:18;2:1;3:1023;4:0;5:0
]
SYNC: 0_0_66_66
SYNC: 0_0_66_66
[3564][app_uart.c][ 357]uart rcv:32,[+HRMADC:1:0;2:1;3:1023;4:0;5:0
]
[3564][app_network.c][ 120]net state: NET_STATE_CONFIG
del if0
usl
sul 0 0
[3564][app_network.c][ 157]time out,stop smartconfig
[3564][app_network.c][  95]net state: NET_INIT
[3564][app_wifi.c][  40]ssid(89BC9E)and pswd(12345678) already set;
mode : sta(a0:20:a6:15:13:6e)
add if0

Task stack [esp_event_loop_] stack from [3fff7568] to [3fff7d64], total [2048] size

                   0          4          8          c         10         14         18         1c         20         24         28         2c         30         34         38         3c 

3fff7c00  0x4022466c 0x0000007e 0x00000030 0x4010071d 0x3fff7c50 0x3ffee6fc 0x00000000 0x40108068 0x00000000 0x40105e80 0x3ffee884 0x4023671c 0x00000000 0x3fff7c50 0x0000000c 0x4024210c 
3fff7c40  0x00008900 0x00000000 0x40108068 0x40238f00 0x3ffee600 0x00000002 0x0000c10f 0x4023685e 0x3ffee600 0x00000002 0x0000c10f 0x40236c48 0x00000000 0x00000000 0x40108068 0x4024c169 
3fff7c80  0x3fff7ce0 0x3fff7cec 0x00000000 0x00000000 0x3ffe84f0 0x0000c10f 0x3ffed670 0x402391b8 0x3ffe84f0 0x3ffee600 0x00000020 0x40230200 0x00000000 0x00000000 0x3fff7d20 0x402670c8 
3fff7cc0  0x00000000 0x00000000 0x00000000 0x4026dab2 0x00000000 0x00057059 0x40108068 0x4024c250 0x00000000 0x00000000 0x00000000 0x15a620a0 0xdf036e13 0x00000000 0x3fff7d20 0x00000000 
3fff7d00  0x00000000 0x00000000 0x00000000 0x4026d308 0x00000000 0x00000000 0x00000000 0x4026d34a 0x00000002 0x00000000 0x00000000 0x00000020 0x3fff629c 0x3fff62e1 0x00000004 0x4023426e 
3fff7d40  0x00000000 0x00000dec 0x40229dfe 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 

        PC: 0x40238762        PS: 0x00000033        A0: 0x402387f4        A1: 0x3fff7c10
        A2: 0x3ffee6fc        A3: 0x00000000        A4: 0x00000168        A5: 0x40236750
        A6: 0x4023686c        A7: 0x3ffee6fc        A8: 0x3ffed5e8        A9: 0x3ffee6fc
       A10: 0x00000000       A11: 0x0000503a       A12: 0x3ffed5e8       A13: 0x3ffee6fc
       A14: 0x0000000a       A15: 0x00000023       SAR: 0x0000001f  EXCCAUSE: 0x0000001c

 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x40100000, len 7264, room 16 
tail 0
chksum 0xaf
load 0x3ffe8408, len 24, room 8 
tail 0
chksum 0x46
load 0x3ffe8420, len 3508, room 8 
tail 12
chksum 0x40
csum 0x40?ZTB┑Z昒?*PU猆UUT礥エ丛u磚ZQTPZTB﹗?X暘i猆PカUQ?┅e猽寓J(U猨*U昛?UXオUEQ,U?b(　旦UT獷┅サZUTU+UP　?オUT?サZUT?Z匲U︰姩P睻∈-誚┄T?T?サZ?
㏕獏(　　—睻P　　?猆狹UZJPVエ?┇Uó　?　　　—Rⅰ　ˉQ?Q礠QEEUB┑Z?
]Z敳+P　　]?ⅰ　
QQEQEEQ??,债摘　　U)ⅰ　　
QQUQQ?T*U*サZ++　　?E,P　　Q(QQQQEQQTPJUU╒++?
　　　アE,P　　Q?*QQUQQQTPJU╒?叓P睰V匷TT狤┆UB┑Z瓍UR隯U猇J～赫VM?QQTPJUUㄕ猇玍U礪Q睌T玅(Q奣Q禂T玅TU奣Q礥獘\?暴U︰?X?┱UB猉V玍U礪?睌T玅(Q昒Q禂T玅?Q礥獘\QUXQ(珺獣TQTP*U-甎昖玍U礪U睌T玅(Q暔Q禂T玅臮浓汗QU匱??T?┱獴猉V玍U礪U睌T玅(Q?サ擳玅TQ～邯Q〞㏄BU*TQTP狫Uó?TJPXUkT磿ù]UZJP箠QQ麵 (350) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (350) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (540) reset_reason: RTC reset 4 wakeup 0 store 0, reason is 7
I (540) gpio: GPIO[15]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (550) gpio: GPIO[12]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (560) gpio: GPIO[14]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (570) gpio: GPIO[16]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (580) gpio: GPIO[13]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (590) gpio: GPIO[0]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (600) gpio: GPIO[4]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (600) gpio: GPIO[5]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (610) uart: queue free spaces: 255
[image]
[image]

sdk:v3.2-dev-133-g5f2fd09

FireWinter, 2019-03-11:
EXCCAUSE: 0x0000001c 属于操作非法指针，反汇编看一下PC 和 A0指向的是哪个地址？

1XiaSheng, 2019-03-12:
@FireWinter      还没操作过反汇编,怎样反汇编?有没有方法?,谢谢

myzhang001, 2019-03-12:
我发现还有一个问题及时  超时后 调用 esp_stop   然后调用esp_connected  不会出发连接动作，必须要复位wifi  才会连接路由器

FayeY, 2019-04-01:
想请问，这个问题有进展吗？

1XiaSheng, 2019-04-01:
不是经常出现, 偶发性的,后来就没去纠结了

FayeY, 2019-04-01:
这个问题先行关闭，如果有任何更新，欢迎 reopen。

closed 2019-04-01 by FayeY